kizniche/Mycodo #582: https://learn.adafruit.com/adafruits-raspberry-pi-lesson-11-ds18b20-temperature-sensing

electricspaceman, 2018-12-19:
## Mycodo Issue Report:
  - Specific Mycodo Version: **6.4.7**
 
#### Problem Description
Please list:

Goal:
  - Connect temperature DS18B20 sensors

Using:
- RaspberryPi3
- [RPI3v2](http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=22&products_id=67) http://www.sheepwalkelectronics.co.uk/RPI3_information.php
- [SWE2+SWE0 Package](http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=23&products_id=57)

  Error: 

Sensors aren't being detected. I think it's because the adapter I'm using (RPI3v2) is using i2c as well as 1WIRE. The maker has helped me to get addresses and temperature readouts using OWFS, is Mycodo compatitble? The hardware is great because it facilitates scaling up a sensor network, just can't seem to get at it. Anything I can do/that can be done?

[
[image]
](url)
 
### Additional Notes

electricspaceman, 2018-12-20:
Some addtional information- from the maker at sheepwalk

> Starting at the start, does the adapter show up when you use the "i2cdetect" command? Running "sudo i2cdetect -y 1" should show two devices on your i2c bus, the 1-Wire master (at address 1c) and the RTC IC (at address 68).

Maybe that will mean more to you than it does to me.

Just had another play about in terminal. 

pi@raspberrypi:/sys/bus/w1/devices $ owread /FF4238711604/temperature;echo 19.125
ServerRead: Data error on /FF4238711604/temperature
19.125
pi@raspberrypi:/sys/bus/w1/devices $ owread /FF4238711604/temperature
ServerRead: Data error on /FF4238711604/temperature
pi@raspberrypi:/sys/bus/w1/devices $ cd /sys/bus/w1/devices
pi@raspberrypi:/sys/bus/w1/devices $ ls
00-180000000000  00-980000000000  w1_bus_master1
pi@raspberrypi:/sys/bus/w1/devices $ pi@raspi:~$ owread /28.FC6711040000/temperature; echo
-bash: pi@raspi:~$: command not found

pi@raspberrypi:/sys/bus/w1/devices $ 19.125
-bash: 19.125: command not found
pi@raspberrypi:/sys/bus/w1/devices $ pi@raspi:~$ owread /26.57144F010000/humidity; echo
-bash: pi@raspi:~$: command not found

pi@raspberrypi:/sys/bus/w1/devices $ 59.7425
-bash: 59.7425: command not found
pi@raspberrypi:/sys/bus/w1/devices $ owread /28.FF5930711604/temperature
     16.9375pi@raspberrypi:/sys/bus/w1/devices $ owserver
pi@raspberrypi:/sys/bus/w1/devices $ owdir
/10.67C6697351FF
/05.4AEC29CDBAAB
/28.FF4238711604
/28.FF1265711604
/28.FF322C711603
/28.FF5930711604
/bus.9
/bus.8
/bus.7
/bus.6
/bus.5
/bus.4
/bus.3
/bus.2
/bus.16
/bus.15
/bus.14
/bus.13
/bus.12
/bus.11
/bus.10
/bus.1
/bus.0
/uncached
/settings
/system
/statistics
/structure
pi@raspberrypi:/sys/bus/w1/devices $ ls
00-060000000000  w1_bus_master1
00-fa0000000000
pi@raspberrypi:/sys/bus/w1/devices $ cd 00-060000000000
pi@raspberrypi:/sys/bus/w1/devices/00-060000000000 $ ls
driver  id  name  power  rw  subsystem  uevent
pi@raspberrypi:/sys/bus/w1/devices/00-060000000000 $ cd ..
pi@raspberrypi:/sys/bus/w1/devices $ cat /sys/bus/w1/devices/
00-060000000000/ 00-860000000000/
00-460000000000/ w1_bus_master1/
pi@raspberrypi:/sys/bus/w1/devices $ cat /sys/bus/w1/devices/00-060000000000/w1_slave
cat: /sys/bus/w1/devices/00-060000000000/w1_slave: No such file or directory
pi@raspberrypi:/sys/bus/w1/devices $ sudo modprobe w1-gpio
pi@raspberrypi:/sys/bus/w1/devices $ sudo modprobe w1-therm
pi@raspberrypi:/sys/bus/w1/devices $ cd /sys/bus/w1/devices
pi@raspberrypi:/sys/bus/w1/devices $ ls
**00-260000000000  00-c60000000000
00-460000000000  w1_bus_master**1
pi@raspberrypi:/sys/bus/w1/devices $ cd 28.FF5930711604
-bash: cd: 28.FF5930711604: No such file or directory
pi@raspberrypi:/sys/bus/w1/devices $ cd 00-260000000000
pi@raspberrypi:/sys/bus/w1/devices/00-260000000000 $ ls
pi@raspberrypi:/sys/bus/w1/devices/00-260000000000 $ ls-l
-bash: ls-l: command not found
pi@raspberrypi:/sys/bus/w1/devices/00-260000000000 $ ls -l
total 0
lrwxrwxrwx 1 root root    0 Dec 20 00:08 driver -> ../../../bus/w1/drivers/w1_slave_driver
-r--r--r-- 1 root root 4096 Dec 20 00:08 id
-r--r--r-- 1 root root 4096 Dec 20 00:08 name
drwxr-xr-x 2 root root    0 Dec 20 00:08 power
-rw-r--r-- 1 root root 4096 Dec 20 00:08 rw
lrwxrwxrwx 1 root root    0 Dec 20 00:08 **subsystem -> ../../../bus/w1**
-rw-r--r-- 1 root root 4096 Dec 20 00:07 uevent
pi@raspberrypi:/sys/bus/w1/devices/00-260000000000 $ 

I see there's a subsystem file there, are the sensors in there? How do I access that?

[image]

kizniche, 2018-12-20:
I'm still not sure what this is. Can you provide some documentation for the device so I can read how it operates? Is there a code example for how to acquire a measurement?

From your log output, it doesn't appear you are acquiring a measurement. All you are doing is listing directories.

electricspaceman, 2018-12-23:
Hello,

As far as I can tell it enables massively extensible 1Wire sensor networks using i2c.

It's RaspberryPi + [adapter](http://www.sheepwalkelectronics.co.uk/RPI2v2.php) + [passive adapter](http://www.sheepwalkelectronics.co.uk/SWE2b.php) + [sensors](http://www.sheepwalkelectronics.co.uk/SWE0a.php). Uses standard ethernet cables to connect it all together.

> From your log output, it doesn't appear you are acquiring a measurement. All you are doing is listing directories.

Yes, you are correct. Sorry that screenshot isn't that useful. I just wanted to show that the hardware that I am using works as intended.

Might be useful to visit these pages as well

http://www.sheepwalkelectronics.co.uk/1wire_basics.php

http://www.sheepwalkelectronics.co.uk/1w_software.php

http://www.sheepwalkelectronics.co.uk/1w_raspberrypi.php

Do you think it would be possible to add compatibility with Mycodo?

Thank you for you interest and help with this.

electricspaceman, 2018-12-23:
Amazing! Thank you Kyle!!

Is there anything I can do to help? I'd be happy to post you some of this hardware.

kizniche, 2018-12-24:
I just pushed a potential fix. Can you test by [upgrading the master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch), then seeing if there are devices in your Input dropdown menu?

electricspaceman, 2018-12-24:
Thanks Kyle! Will do this when back at base and report back. Should be in either four or eight days.

A happy Christmas to you.

kizniche, 2018-12-25:
Thanks. Merry Christmas!

pmunz75, 2018-12-27:
https://github.com/kizniche/Mycodo/issues/586

electricspaceman, 2019-01-03:
Hello again, 

I hope you had a good holiday!

Currently that hardware using the i2c bridging from sheepwalkelectronics isn't working yet. 

- I've updated to the master branch of Mycodo
- Altered the boot config file #586 
```
`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
#dtoverlay=w1-gpio
dtoverlay=w1-gpio,gpiopin=4,pullup=on
`
```

- Made sure that I am able to get readings via OWFS 
[image]
[image]
- Tried to add a DS18B20 sensor
[image]
- Getting odd device ID
[image]
- Getting system messages when activating
[image]
- Getting a 0C reading on live page
[image]

Any help would be appreciated.

kizniche, 2019-01-04:
If your 1-wire input configuration still show a device ID with a dash (```-```) then you are not using the latest code from [master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch). 7.0.4 has not been released, so if you want to use the pre-release fix, you need to upgrade to master.

kizniche, 2019-01-04:
7.0.4 has not been released, so it is not possible to upgrade to 7.0.4. The master branch is not 7.0.4. To determine if you are upgraded to the master branch, all 1-wire device IDs will lack a dash (```-```) in the ID. Do your 1-wire device IDs have a dash after upgrading to master?

electricspaceman, 2019-01-05:
Hi Kyle,

Sorry!! I was being a dummy, still very new to all this. I successfully upgraded to the master without realising properly. Thanks for your patience. 

Checking the 1-wire interface IDs, there are no longer any dashes but I'm not sure the device ID's are correct.  As far as my understanding goes they should all read {28-}**FFXXXXXXXXXX** but the device ID's showing Mycodo are XXXX00000000, I could be wrong about that though.  On activating the sensors Mycodo shows a warning that it could not activate the sensors with the given (XXXX0000000) addresses.

When I go to the live page to see the temperature readouts they are reading 0C and then when I go to the Inputs page I noticed device IDs (XXXX00000000) are different.

[image]
[image]
[image]

Could it because these sensors I'm trying to use are located on the i2c? Isn't that how OWFS locates and reads data from them?

kizniche, 2019-01-05:
This should get the device IDs detected, but I'm pretty sure the current Input modules (which use [w1thermsensor](https://github.com/timofurrer/w1thermsensor)) won't be able to acquire a measurement. I may have to build another module for each 1-wire device that executes ```owread```.

electricspaceman, 2019-01-05:
OK! Gosh. Sorry to have caused this extra work. I appreciate your time on this. Excited to be able to use it! 
I've attempted the python commands given above.

[image]

Not sure what to make of it. Let me know what you think.

Also, randomly, come back to a 502: Bad Gateway Error, so it would appear that I need to reinstall something. Where's the best way to proceed curing that? I'm willing to do a complete reinstall, just don't know the commands to do it off hand.

[image]

kizniche, 2019-01-05:
Okay, a complete reinstall can be done with the following commands:

First, reboot.

```sudo shutdown now -r```

After the reboot:

```bash
sudo ~/Mycodo/mycodo/scripts/upgrade_commands.sh backup-create
sudo mv ~/Mycodo ~/Mycodo-backup-01
cd ~
git clone https://github.com/kizniche/Mycodo
sudo /bin/bash ~/Mycodo/install/setup.sh
```

electricspaceman, 2019-01-05:
Thanks! Just doing this now.

electricspaceman, 2019-01-05:
OK so reinstalled, upgraded to master and then performed those python commands, with the following result.

[image]

kizniche, 2019-01-05:
from a fresh install, you needed to have installed the w1thermsensor dependency before trying to import it.

```bash
sudo ~/Mycodo/env/bin/pip install w1thermsensor
```

```python
~/Mycodo/env/bin/python
>>> from w1thermsensor import W1ThermSensor
>>> W1ThermSensor.get_available_sensors()
```

electricspaceman, 2019-01-05:
OK - installed the module and performed the commands above..

[image]

electricspaceman, 2019-01-05:
OK So can see the new ID's! Just can't read temperature. See below:

[image]
[image]
[image]

kizniche, 2019-01-05:
Okay. as suspected, the w1thermsensor module can't detect the IDs. Can you check if it can return a measurement?

```python
~/Mycodo/env/bin/python
>>> from w1thermsensor import W1ThermSensor
>>> W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18S20, 'FF1234567890').get_temperature()
```

kizniche, 2019-01-05:
I've been working on pushing a modified ds18b20.py input module, so I just wanted to verify what I've suspected, that the w1thermsensor doesn't work with owfs. I'll have the module ready to test shortly- I'm running a few more tests on it.

electricspaceman, 2019-01-05:
Thank you. 

Here's what I get when trying to get a measurement:

[image]

kizniche, 2019-01-05:
I just pushed a commit that adds OWFS support for 1-wire devices (only DS18B20 is enabled right now). Go ahead and upgrade to master, set the Library option of the DS18B20 Input to "ow-shell", then save. You should then see the IDs appear. Select an ID, save, then activate the Input.

electricspaceman, 2019-01-05:
Upgraded to the master, but when I navigate to the Data/Inputs page I'm getting Error 500: Server Error. Should I try a fresh reinstall again?

[image]

kizniche, 2019-01-05:
Don't reinstall. I just pushed a fix. upgrade to master and try again.

electricspaceman, 2019-01-05:
Hey it's worked! WOO!

kizniche, 2019-01-05:
You're welcome. Can you do me a favor and upgrade to master again and test the different resolutions (9, 10, 11, and 12-bit)? I just pushed a commit that enabled the resolution setting for the ow-shell library for the DS18B20.

electricspaceman, 2019-01-07:
Hi Kyle, 

My apologies for my late response. Due to my natural stupidity I snapped my microSD into two. Back up and running with a new card now. Is it too late to perform your requested testing? I see you've posted 7.0.4 and included OWFS, amazing!

Out of interest where can I learn moroe about the significance or practical implications between 9, 10, 11, and 12-bit resolution?

Also, minor, but how do you say Mycodo? Myco-do, My-co-do, perhaps another way? I'm at odds with a mate and us Brits tend to say things in our own way most of the time, for better or worse. As **the Maker** you get to decide!

electricspaceman, 2019-01-08:
Hit a brick wall so to speak  - sensors aren't showings up. Pretty sure it's naught to do with Mycodo however.I followed the instructions on teh sheepwalk electronics software configuration page, and when  I submit the `owdir` command all I get is:

`pi@raspberrypi:~ $ owdir

/10.67C6697351FF
/05.4AEC29CDBAAB
/bus.0
/uncached
/settings
/system
/statistics
/structure
/simultaneous
/alarm
pi@raspberrypi:~ $ 
`

I know OW deivces begining with 28. refer to the DS18B20s, so who knows what 10. and 05. are. I'm raising this here on the off chance you know what's going on, otherwise I don't think it's anything to do with your work.

kizniche, 2019-01-08:
> Out of interest where can I learn moroe about the significance or practical implications between 9, 10, 11, and 12-bit resolution?

From the [datasheet](https://cdn.sparkfun.com/datasheets/Sensors/Temp/DS18B20.pdf), "The  resolution of the DS18B20 is configurable (9, 10, 11, or 12 bits), with 12-bit readings the factory default state. This equates to a temperature resolution of 0.5°C, 0.25°C, 0.125°C, or 0.0625°C. " (page 5), with the time it takes to acquire measurements for each resolution in Table 3 on page 9.

> how do you say Mycodo?

https://www.howtopronounce.com/mycodo/ does a fairly good job at pronouncing it.

> Hit a brick wall so to speak - sensors aren't showings up.

I can't help with this as I don't have the hardware to do so. You'll have to get things set up to where the IDs are detected as they were before.

closed 2019-01-23 by kizniche

electricspaceman, 2020-11-19:
hey kyle! me again. 

I'm having a little issue adding my DS18B20's as inputs, I think it might be related to action at #746.

I go to add a DS18b20 sensor (via ow-shell) but nothing's showing up in the list.

It's not a hardware issue because I can pick and read from the sensors using OWFS commands.

any help getting this working would be well appreciated!

MILO-GM, 2020-11-20:
Hi
I had a similar problem a few weeks ago installing the dependencies of a DS18B20.
I wanted to report it to Kyle as soon as I had the time, but at this point I write it here, hoping it will help you to solve the problem, but I think mine was just a "lucky" case.

- Raspberrry pi4 (tested in two different boards a few weeks ago and today, with the same results)
-Mycodo 8.8.8

Below I list what happened:

-Installed “onewire” dependencies, tried all pins, never shows probe id. 

[image]

-Installed the dependencies “w1thermsensor”, tried all the pins( for example gpio27) shows this strange id
[image]
- in live-data show 0 degree
[image]

-Tried to connect to GPIO4 show the correct id
[image]

- in live-data show the correct temperature
[image]

The strange thing is that the probe run only if connected to GPIO4. It has been running for two weeks now without any problems ,when compared to a certified professional thermometer, I could see that it is also quite accurate.
[image]

Hope I was helpful in some way, many thanks to Kyle for Mycodo !!!

electricspaceman, 2020-11-20:
Hi @MILO-GM 

Thank you for your post. 

I would prefer to use the ow-shell implementation for the DS18B20 input as I'm using a bridging circuit with my raspberry pi which breaks out the I2C buses on to RJ45 sockets and the vendor ([sheepwalk electronics](https://www.sheepwalkelectronics.co.uk/product_info.php?products_id=67)) vouches for OWFS. 

[image]

If I use owdir I get a response

[image]

and if I poll the sensor for a temperature I get a reading (yeah it's pretty chilly where I am right now) 

[image]

Following your experience, I have also been tried the w1thermsensor input just in case. 

I had a look at the dependencies list and it shows that the dependencies for DS18B20 and DS18B20_OWS are present.
I'm at the point you were at where something appears but it does not make sense (they don't appear with the correct device ID/ device addresses).

[image]

[image]

[image]

I currently have my config.txt file end set with `dtoverlay=w1-gpio` so I think my GPIO pin is also set to 4.  Is there anything that I must do to set this to GPIO PIN 4? The sheepwalk electronics bridging device [is wired to this pin.](https://www.sheepwalkelectronics.co.uk/RPI3_information.php)

> The RJ45 sockets along the left side of the PCB (above the power connector and SD card) have one bus on each (busses 0 and 1) and are wired to the same standard as the RPI2 and all our other modules to allow easy connection via Cat5 cable. Four pins are connected, pins 1 and 2 (GND and +5V), and **pins 4** and 5 (1-Wire Data and Ground).

I'm at a total loss what to do next. Any views from anyone would be massively appreciated.

MILO-GM, 2020-11-20:
Hi Electricspaceman, 

I  send you some photos where you can see how I connected the probe.
Only if I connect it to gpio4 can I see the id of the probe.
If I connect it to any other gpio I don't see any id.
Unfortunately my technical preparation is not programming and I don't know how to help you more than that.
Hope it can help you or help Kyle figure out what the problem is .....
[image]
[image]
[image]