kizniche/Mycodo #583: Dashboard TH16 measurement units display wrong.

SAM26K, 2018-12-23:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.0.2
 
#### Problem Description
Measurement display on dashboard for TH16/10 not right when default not used.
Affects deg F and KpA.  See screen shot.
[image]

Attached is a working python module for Sonoff TH16/10.   Only ip address needs to be changed for specific installed device.
[TH16D.zip](https://github.com/kizniche/Mycodo/files/2705784/TH16D.zip)

SAM26K, 2018-12-23:
Also  affects dew point conversion to deg F.

kizniche, 2018-12-23:
Just added. Thanks for the testing. This will be live in v7.0.3 with the option to set the IP address.

kizniche, 2018-12-23:
...This also happened to be the 4000th commit for Mycodo.

SAM26K, 2018-12-24:
Great job and thanks very much!

SAM26K, 2018-12-24:
I'm using a linux command output control to switch the relay on and off as follows:
curl http://192.168.1.34/cm?cmnd=Power%20on
curl http://192.168.1.34/cm?cmnd=Power%20off
Would be nice if you  could include that with the new Sonoff device module when you get time. Only ip address would be unique.
Thanks Again.

closed 2018-12-24 by kizniche