kizniche/Mycodo #584: Device: 128x32_pioled (PiOLED 128x32)

pmunz75, 2018-12-23:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.0.2
 
#### Problem Description
Please list:
cant install driver for : Device: 128x32_pioled (PiOLED 128x32)
 
### Errors
  Device: 128x32_pioled (PiOLED 128x32)

The install has been initiated for the following dependencies:

    PIL

The Dependency Install Log below will refresh every 5 seconds. Wait until all dependencies have been installed to attempt to add the associated device again. You will know the entire install process have completed by "#### All Dependencies have been installed." appearing at the end of the log.

If there's an issue with the install and you need to reset the install state, delete ~/Mycodo/.dependency to restore dependency install functionality.

[2018-12-23 19:35:36]       File "/usr/lib/python3.5/distutils/command/install.py", line 583, in run
[2018-12-23 19:35:36]         self.run_command('build')
[2018-12-23 19:35:36]       File "/usr/lib/python3.5/distutils/cmd.py", line 313, in run_command
[2018-12-23 19:35:36]         self.distribution.run_command(command)
[2018-12-23 19:35:36]       File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
[2018-12-23 19:35:36]         cmd_obj.run()
[2018-12-23 19:35:36]       File "/usr/lib/python3.5/distutils/command/build.py", line 135, in run
[2018-12-23 19:35:36]         self.run_command(cmd_name)
[2018-12-23 19:35:36]       File "/usr/lib/python3.5/distutils/cmd.py", line 313, in run_command
[2018-12-23 19:35:36]         self.distribution.run_command(command)
[2018-12-23 19:35:36]       File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
[2018-12-23 19:35:36]         cmd_obj.run()
[2018-12-23 19:35:36]       File "/usr/lib/python3.5/distutils/command/build_ext.py", line 339, in run
[2018-12-23 19:35:36]         self.build_extensions()
[2018-12-23 19:35:36]       File "/tmp/pip-install-mtw1b_99/Pillow/setup.py", line 589, in build_extensions
[2018-12-23 19:35:36]         raise RequiredDependencyException(f)
[2018-12-23 19:35:36]     __main__.RequiredDependencyException: jpeg
[2018-12-23 19:35:36]     
[2018-12-23 19:35:36]     During handling of the above exception, another exception occurred:
[2018-12-23 19:35:36]     
[2018-12-23 19:35:36]     Traceback (most recent call last):
[2018-12-23 19:35:36]       File "", line 1, in 
[2018-12-23 19:35:36]       File "/tmp/pip-install-mtw1b_99/Pillow/setup.py", line 814, in 
[2018-12-23 19:35:36]         raise RequiredDependencyException(msg)
[2018-12-23 19:35:36]     __main__.RequiredDependencyException:
[2018-12-23 19:35:36]     
[2018-12-23 19:35:36]     The headers or library files could not be found for jpeg,
[2018-12-23 19:35:36]     a required dependency when compiling Pillow from source.
[2018-12-23 19:35:36]     
[2018-12-23 19:35:36]     Please see the install instructions at:
[2018-12-23 19:35:36]        https://pillow.readthedocs.io/en/latest/installation.html
[2018-12-23 19:35:36]     
[2018-12-23 19:35:36]     
[2018-12-23 19:35:36]     
[2018-12-23 19:35:36]     ----------------------------------------

[2018-12-23 19:35:37] End install of PIL

[2018-12-23 19:35:37] 
[2018-12-23 19:35:37] #### Setting permissions

pmunz75, 2018-12-23:
this help me to fix the error : 

So it looks like you're trying to install PIL without the proper packages installed beforehand. Try installing libjpeg.

sudo apt-get install libjpeg-dev
pip install pillow
sudo reboot

kizniche, 2018-12-24:
Thanks. I'll add libjpeg-dev to the module dependency list.

closed 2018-12-24 by kizniche