kizniche/Mycodo #586: Cannot activate DS18B20 sensor

Magnum-Pl, 2018-12-27:
Hi Kyle, 

I'm unable to activate a DS18B20 sensor. After installing the dependency, I'm able to add the sensor and can find the sensor ID in the drop list, but I get an error when trying to activate or deactivate it. If I unplug the sensor, I don't see the ID in the list. I've tried using different sensors and different pins with the same results. 

The error is get on the data page says this:

> Error: Could not activate Input controller with ID f388140d-b194-479c-b1b8-b4e0e8ebb387: No DS18B20 temperature sensor with id '28-0517005b0aff' found

The log say this:

> 2018-12-26 16:12:33,062 - mycodo.daemon - ERROR - Could not activate Input controller with ID f388140d-b194-479c-b1b8-b4e0e8ebb387: No DS18B20 temperature sensor with id '28-0517005b0aff' found
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 539, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 174, in __init__
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20.py", line 63, in __init__
    self.location)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/w1thermsensor/core.py", line 154, in __init__
    raise NoSensorFoundError(self.type_name, self.id)
w1thermsensor.errors.NoSensorFoundError: No DS18B20 temperature sensor with id '28-0517005b0aff' found

Magnum-Pl, 2018-12-27:
I was just able to get both of the sensors I've been testing to activate and display data but only if I have one connected at a time and I have to leave the ID field blank. Once its working, if I choose the sensor ID, I get the same error and I have to delete the sensor and add it again without choosing the ID.

marnixongena, 2018-12-27:
Hi There,
I have the same problem with a DS18S20P

pmunz75, 2018-12-27:
Hello,
Please read the link.
https://de.pinout.xyz/pinout/1_wire#
or edit your /boot/config.txt file and add

dtoverlay=w1-gpio,gpiopin=10

or whatever gpio example is 10 add your GPIO Pin Connectet DS18xxxx

Pmunz

marnixongena, 2018-12-28:
> Hello,
> Please read the link.
> https://de.pinout.xyz/pinout/1_wire#
> or edit your /boot/config.txt file and add
> 
> dtoverlay=w1-gpio,gpiopin=10
> 
> or whatever gpio example is 10 add your GPIO Pin Connectet DS18xxxx
> 
> Pmunz

I have edited the config file , can see the sensor in the directory but still the same error

Error: Could not activate Input controller with ID dba88056-55f6-4858-a0ad-cd8b7cbf0ff8: No DS18S20 temperature sensor with id '10-0008010fd95a' found

kizniche, 2018-12-29:
I just pushed a potential fix to master. If you would like to test, [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch), then resave the input settings and activate the sensor.

marnixongena, 2018-12-29:
Yes I use some home control system Domotiz who also run on raspberry pi

kizniche, 2018-12-29:
See https://github.com/timofurrer/w1thermsensor/issues/46

kizniche, 2018-12-29:
https://translate.googleusercontent.com/translate_c?depth=1&nv=1&rurl=translate.google.com&sl=pl&sp=nmt4&tl=en&u=https://www.elektroda.pl/rtvforum/topic1757750.html&xid=17259,15700021,15700043,15700122,15700124,15700186,15700190,15700201,15700237,15700242,15700248&usg=ALkJrhgBoE65ltcwi1DcD3puDeXg7YRU3Q indicates it may be a timing issue (if not a wiring issue).

kizniche, 2018-12-29:
http://www.protonbasic.co.uk/showthread.php/59004-DS18S20-16F628A-85°-Read-problem/page4 suggests a potential explanation for this issue.

kizniche, 2018-12-29:
Are you using a 4.7kOhm pullup resistor?

marnixongena, 2018-12-29:
> Are you using a 4.7kOhm pullup resistor?

Yes i do

marnixongena, 2018-12-30:
Solved !
upgrading to master and 
adding
 dtoverlay=w1-gpio,gpiopin=4,pullup=on
to boot/config.txt
did the trick
Happy New Year

pmunz75, 2019-01-01:
Hello ,
Sorry i have the same problems with version 7.0.3
the trick not work for me ( dtoverlay=w1-gpio,gpiopin=4,pullup=on )
ERROR:
Error: Could not activate Input controller with ID a6e1babc-cfce-4587-9a37-ce087e163144: No DS18B20 temperature sensor with id '28-01143275be92' found

in version 6.x.x ran the sensor ( DS18B20 ) with 4,7k pullup perfekt.

Thanks

kizniche, 2019-01-01:
You have to [follow these instructions] to use the latest fix, until 7.0.4 is released that incorporates this fix.

pmunz75, 2019-01-01:
Perfect , thank you !

closed 2019-01-08 by kizniche