kizniche/Mycodo #587: Using Influxdb util

not5, 2018-12-31:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.0.3
 
#### Problem Description
Please list: I'm having trouble using the influxdb read_last_influxdb() function.

Temperature Sensor (MAX31865) unique_ID:	82394f8b-fef6-4e2c-a843-7898a15268b2
Measurement in Fahrenheit unique_ID:	aafd828a-c668-4b77-89ad-01b8b8026900

My read_last_influxdb() function's parameters are as follows: 
unique_id='aafd828a-c668-4b77-89ad-01b8b8026900'
unit='F'
measurement='temperature'
channel=1
duration_sec=None

According to the function query_string(), it's creating the following query which doesn't look right... "SELECT LAST(value) FROM F WHERE device_id='aafd828a-c668-4b77-89ad-01b8b8026900' AND channel='1' AND measure='temperature'"

I don't think this is the intended structure as this would mean there's a table created for each unit which can't be right.
 
### Errors

Function is returning a None data type.

kizniche, 2019-01-03:
Try with channel 0

not5, 2019-01-04:
channel 0 returns None as well.

kizniche, 2019-01-04:
Also, the measurement is stored with the unique_id of the input ,not the measurement. I may have incorrectly told you this in the past.
Try channel = 0 and unique_id = 82394f8b-fef6-4e2c-a843-7898a15268b2

kizniche, 2019-01-04:
Also, if there is a conversion, then measurement is no longer stored. If you specify a measurement, then you will not have a returned value. It is best to just not specify a measurement unless it isn't converted (or just never use the measurement parameter unless you have a purpose to do so).

not5, 2019-01-04:
I'm a little confused now. In mycodo, I have the measurement converted from C to F. I believe the measurement parameter is required for this function, so I don't think I can just not declare it (unless I pass None?).

kizniche, 2019-01-04:
Yes, set it to None.

not5, 2019-01-04:
Ok that worked. Great, thanks!

For the others, here's the function with parameters passed:

read_last_influxdb(input_unique_id, str_unit, None, int_channel).

Note: Channel indexing begins at 0.

closed 2019-01-04 by not5

kizniche, 2019-01-04:
I add 1 to each channel for presenting them to the user, as starting at 1 makes more sense to the end user, and the end user really has no need to query the database itself. Digging in the input modules you can see the exact channels each measurement uses.