kizniche/Mycodo #588: SHT 31 Smart Gadget

RynFlutsch, 2019-01-01:
Version 7.0.0 was working
Since 7.0.1 i got following error, if i try to add the gadget.
[image]

I reinstalled mycodo with 7.0.3, now i can add the gadgets and establish a BT connection but i receive no measurements.

jacopienaar, 2019-01-03:
I have the same issue. I recently installed a vanilla Raspbian with Mycodo 7.03 and do not receive any measurements from my instrument. The LCD also doesn't work.
If I insert another SD card with Mycodo 5.3.3 on it, then everything works just fine.

kizniche, 2019-01-03:
16 seconds is a very small duration. Try extending your duration to something larger like 60 seconds in order to test. Are there any errors in the daemon log?

kizniche, 2019-01-03:
Also, please screenshot your Input configuration.

RynFlutsch, 2019-01-03:
019-01-03 21:35:04,478 - mycodo.daemon - INFO - 38.17 MB RAM in use
2019-01-03 21:35:17,803 - mycodo.sht31_smart_gadget_c71bd692 - INFO - Name: Smart Humigadget, Log Interval: 600.0 sec
2019-01-03 21:40:30,110 - mycodo.sht31_smart_gadget_9eddf4fb - INFO - Name: Smart Humigadget, Log Interval: 600.0 sec
2019-01-03 21:40:45,403 - mycodo.sht31_smart_gadget_9eddf4fb - ERROR - InputModule raised an exception when taking a reading: unpack requires a bytes object of length 4
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 367, in get_measurement
    return_dict[0]['value'] = self.gadget.readTemperature()
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 138, in readTemperature
    return struct.unpack('f', self.characteristics['Temperature'].read())[0]
struct.error: unpack requires a bytes object of length 4

[image]

kizniche, 2019-01-04:
Looking at your log again, it appears you have at least two smart gadgets activated. It appears device c71bd692 is working correctly (returns logging interval close to when the daemon started), however device 9eddf4fb took 5 minutes to return the logging interval. Can you help me understand your configuration and make sense of this log?

RynFlutsch, 2019-01-04:
> I need some more information to know what's going on.
> 
> You mention it doesn't work, but does it work up until the point it doesn't work?

yes, these bug appears since the upgrade from 7.00 to 7.0.1

RynFlutsch, 2019-01-04:
> If your period is 20 seconds, you have 5 minutes from `21:35:04` when the daemon has started until `21:40:45` when the error occurs. This allows for many 20 second periods in between that.

i think i activated the sensor manually , now i restarted with both gadgets activated:
`2019-01-04 08:40:34,797 - mycodo.input_9eddf4fb - INFO - Activated in 641.2 ms
2019-01-04 08:40:35,145 - mycodo.input_c71bd692 - INFO - Activated in 245.2 ms
2019-01-04 08:40:35,512 - mycodo.input_131da12d - INFO - Activated in 266.1 ms
2019-01-04 08:40:35,518 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-04 08:40:35,525 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-04 08:40:35,530 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-04 08:40:35,536 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-04 08:40:36,043 - mycodo.daemon - INFO - Mycodo daemon started in 3.708 seconds
2019-01-04 08:40:36,050 - mycodo.daemon - INFO - 38.27 MB RAM in use
2019-01-04 08:40:54,149 - mycodo.sht31_smart_gadget_c71bd692 - INFO - Name: Smart Humigadget, Log Interval: 600.0 sec`

i changed nothing of your code
the seconds i received no data changes with the period-seconds:
[image]

kizniche, 2019-01-04:
> yes, these bug appears since the upgrade from 7.00 to 7.0.1

Are you suggesting you're not running the latest version? If so, upgrade to the latest version.

RynFlutsch, 2019-01-04:
> Have you taken my recommendation and changed your period to 60 seconds?

yes for sensor 1 
sensor 2 is still 20 sec

> Are you suggesting you're not running the latest version? If so, upgrade to the latest version.

no, i am on 7.0.3

on 7.0.0 the gadget worked properly =)

kizniche, 2019-01-04:
Have you tried pulling the battery of your device and reinserting it?

I will attempt a fresh install of Mycodo and see if I have any issues.

kizniche, 2019-01-04:
Can you install 7.0.1 and see if the device works again? (7.0.0 didn't have the device installed).

You can go back to test an older version with these commands:

```
sudo service mycodo stop
sudo service mycodoflask stop
cd ~
sudo mv ~/Mycodo ~/Mycodo-backup

wget https://github.com/kizniche/Mycodo/archive/v7.0.1.tar.gz
mkdir ~/Mycodo
tar xzfv ./v7.0.1.tar.gz -C ~/Mycodo --strip-components=1
sudo /bin/bash ~/Mycodo/setup/install.sh
```

This will move your old install and install 7.0.1. You can now test if the device works. When done testing, use these commands to get your current version back:

```
sudo service mycodo stop
sudo service mycodoflask stop
cd ~
sudo mv ~/Mycodo ~/Mycodo-backup-7.0.1
sudo mv ~/Mycodo-backup ~Mycodo
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh
```

RynFlutsch, 2019-01-05:
> sudo /bin/bash ~/Mycodo/setup/install.sh

is not working, after unzipping 7.0.1 there is no setup folder:
[image]

RynFlutsch, 2019-01-05:
i have found setup.sh in the install folder =D

kizniche, 2019-01-05:
Sounds like a browser issue

RynFlutsch, 2019-01-05:
Yes.

Daemon.log:
`2019-01-05 17:30:51,378 - mycodo.daemon - INFO - A new version of Mycodo is available. Upgrade through the web interface under Config -> Upgrade. This message will repeat every 10 days unless Mycodo is upgraded or upgrade checks are disabled.
2019-01-05 17:51:49,174 - mycodo.input_1b3a7c8e - INFO - Activated in 297.0 ms
2019-01-05 17:51:51,668 - mycodo.sensor_gadget_sht31_1b3a7c8e - ERROR - Error: Failed to connect to peripheral C9:8E:20:E7:7D:0A, addr type: random`

RynFlutsch, 2019-01-05:
Ok thats weird, the other Sensor is working now.
Newer log:
2019-01-05 17:51:49,174 - mycodo.input_1b3a7c8e - INFO - Activated in 297.0 ms
2019-01-05 17:51:51,668 - mycodo.sensor_gadget_sht31_1b3a7c8e - ERROR - Error: Failed to connect to peripheral C9:8E:20:E7:7D:0A, addr type: random
2019-01-05 17:55:56,444 - mycodo.sensor_gadget_sht31_1b3a7c8e - ERROR - Error: Device disconnected
2019-01-05 17:56:11,100 - mycodo.sensor_gadget_sht31_1b3a7c8e - ERROR - Error: Failed to connect to peripheral C9:8E:20:E7:7D:0A, addr type: random
2019-01-05 17:56:11,116 - mycodo.sensor_gadget_sht31_1b3a7c8e - ERROR - Error: 'NoneType' object has no attribute 'poll'

kizniche, 2019-01-05:
Try activating only one Input at any time.

kizniche, 2019-01-05:
Do you have the sensor very close to the Pi for testing?

RynFlutsch, 2019-01-05:
So it looks like the Gadget and Raspberry don't connect properly. i have to pair the device with bluetoothctl first ?

RynFlutsch, 2019-01-05:
> Do you have the sensor very close to the Pi for testing?

Yes very close

kizniche, 2019-01-05:
You shouldn't need to do anything except allow Mycodo to do the connecting. Do you happen to have another device connecting to the sensor? You should see the Bluetooth logo blinking if it's unpaired.

kizniche, 2019-01-05:
Is this a Pi3 with integrated Bluetooth or are you using a BT dongle?

RynFlutsch, 2019-01-05:
Yes both Gadgets blinking. I added now both.
In 7.0.3 it was not blinking but received no data.
now it receive sometimes data but is not paired properly

Yes Pi 3 no other Master device

kizniche, 2019-01-05:
in 7.0.3, Mycodo should connect, download data, then disconnect, so it should only stop blinking for the period it's connected.

RynFlutsch, 2019-01-05:
Oh ok, i upgrade to 7.0.3 and test again ?

kizniche, 2019-01-05:
I would recommend going back to your 7.0.3 version now that we know Mycodo can connect to at least one of them and download data.

```
sudo service mycodo stop
sudo service mycodoflask stop
cd ~
sudo mv ~/Mycodo ~/Mycodo-backup-7.0.1
sudo mv ~/Mycodo-backup ~Mycodo
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh
```

kizniche, 2019-01-05:
I would also set your Period to something high, like 120 seconds.

RynFlutsch, 2019-01-05:
ok, i do

RynFlutsch, 2019-01-06:
On 7.0.3 again:
` File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 148, in setTemperatureNotification
    int(0).to_bytes(1, byteorder='little'))
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 510, in writeCharacteristic
    return self._getResp('wr')
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 377, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 334, in _waitResp
    raise BTLEException(BTLEException.DISCONNECTED, "Device disconnected")
bluepy.btle.BTLEException: Device disconnected
2019-01-06 01:25:10,209 - mycodo.input_afae92fb - INFO - Deactivated in 1191231.3 ms
2019-01-06 01:26:33,124 - mycodo.sht31_smart_gadget_98735021 - ERROR - InputModule raised an exception when taking a reading: Device disconnected
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 362, in get_measurement
    return_dict[2]['value'] = self.gadget.readBattery()
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 164, in readBattery
    return int.from_bytes(self.characteristics['Battery'].read(), byteorder='little')
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 166, in read
    return self.peripheral.readCharacteristic(self.valHandle)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 497, in readCharacteristic
    resp = self._getResp('rd')
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 377, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 334, in _waitResp
    raise BTLEException(BTLEException.DISCONNECTED, "Device disconnected")
bluepy.btle.BTLEException: Device disconnected
2019-01-06 01:26:33,226 - mycodo.input_98735021 - INFO - Deactivated in 1165713.3 ms
2019-01-06 01:29:28,545 - mycodo.input_afae92fb - INFO - Activated in 226.0 ms
2019-01-06 01:29:31,444 - mycodo.input_98735021 - INFO - Activated in 215.5 ms`

kizniche, 2019-01-06:
One of my other Pi3s couldn't connect to my device until I either upgraded the linix kernel or installed ```pi-bluetooth```. Go ahead and execute these commands and see if ti can acquire measuremnts:

```bash
sudo apt-get update
sudo apt-get upgrade
sudo shutdown now -r
```

Let the pi reboot.

```bash
sudo apt-get install pi-bluetooth
sudo service mycodo restart
```

RynFlutsch, 2019-01-06:
Hey,
pi-bluetooth was already installed. Now on 7.0.3 the BT Icon at the gadget is blinking non stop.
Sometimes the daemon.log iswithout errors. Now i got some errors:
`Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 353, in get_measurement
    self.download_data()
  File "/root/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 200, in download_data
    if (not self.gadget.waitForNotifications(5) or
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 213, in waitForNotifications
    return self.peripheral.waitForNotifications(timeout)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 524, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 377, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 334, in _waitResp
    raise BTLEException(BTLEException.DISCONNECTED, "Device disconnected")
bluepy.btle.BTLEException: Device disconnected
2019-01-06 11:11:51,976 - mycodo.sht31_smart_gadget_afae92fb - ERROR - InputModule raised an exception when taking a reading: Helper not started (did you call connect()?)
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 353, in get_measurement
    self.download_data()
  File "/root/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 197, in download_data
    start_ms=self.gadget.newestTimeStampMs)
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 197, in readLoggedDataInterval
    self.setSyncTimeMs()
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 169, in setSyncTimeMs
    timestampMs.to_bytes(8, byteorder='little'))
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 169, in write
    return self.peripheral.writeCharacteristic(self.valHandle, val, withResponse)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 509, in writeCharacteristic
    self._writeCmd("%s %X %s\n" % (cmd, handle, binascii.b2a_hex(val).decode('utf-8')))
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 272, in _writeCmd
    "Helper not started (did you call connect()?)")
bluepy.btle.BTLEException: Helper not started (did you call connect()?)
2019-01-06 11:13:51,977 - mycodo.sht31_smart_gadget_afae92fb - ERROR - InputModule raised an exception when taking a reading: Helper not started (did you call connect()?)
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 353, in get_measurement
    self.download_data()
  File "/root/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 197, in download_data
    start_ms=self.gadget.newestTimeStampMs)
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 197, in readLoggedDataInterval
    self.setSyncTimeMs()
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 169, in setSyncTimeMs
    timestampMs.to_bytes(8, byteorder='little'))
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 169, in write
    return self.peripheral.writeCharacteristic(self.valHandle, val, withResponse)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 509, in writeCharacteristic
    self._writeCmd("%s %X %s\n" % (cmd, handle, binascii.b2a_hex(val).decode('utf-8')))
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 272, in _writeCmd
    "Helper not started (did you call connect()?)")
bluepy.btle.BTLEException: Helper not started (did you call connect()?)
2019-01-06 11:13:51,979 - mycodo.input_afae92fb - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.`

kizniche, 2019-01-06:
You didn't mention the status of your kernel... This is the more important of the two issues I mentioned.

kizniche, 2019-01-06:
I can now see one of my systems bluetooth not working (that was yesterday during testing). This is the output from checking the bluetooth version:

```
pi@raspberry1:~ $ sudo hciconfig
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:1A:A3:48  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:731 acl:0 sco:0 events:44 errors:0
	TX bytes:1755 acl:0 sco:0 commands:44 errors:0

pi@raspberry1:~ $ sudo hciconfig hci0 version
Can't read version info hci0: Connection timed out (110)
```

And here is what is expected from a working system:

```
pi@raspberry2:~ $ sudo hciconfig
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:4A:E3:E7  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:162264 acl:4949 sco:0 events:922 errors:0
	TX bytes:24884 acl:1334 sco:0 commands:142 errors:0

pi@raspberry2:~ $ sudo hciconfig hci0 version
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:4A:E3:E7  ACL MTU: 1021:8  SCO MTU: 64:1
	HCI Version: 4.1 (0x7)  Revision: 0x168
	LMP Version: 4.1 (0x7)  Subversion: 0x2209
	Manufacturer: Broadcom Corporation (15)
```

kizniche, 2019-01-08:
I finally got ahold of an extra micro SD card so I can test with a fresh install of raspbian and Mycodo and see what exactly needs to be done to get bluetooth working properly.

RynFlutsch, 2019-01-08:
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Linux BastelPi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

RynFlutsch, 2019-01-08:
sudo hciconfig
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:8F:95:29  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:6079 acl:0 sco:0 events:808 errors:0
	TX bytes:8249 acl:0 sco:0 commands:808 errors:0
sudo hciconfig hci0 version
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:8F:95:29  ACL MTU: 1021:8  SCO MTU: 64:1
	HCI Version: 4.1 (0x7)  Revision: 0x168
	LMP Version: 4.1 (0x7)  Subversion: 0x2209
	Manufacturer: Broadcom Corporation (15)

RynFlutsch, 2019-01-12:
I tried only 1 sensor.
Once mycodo received measurements.
Sometimes the log is filled with the errors i posted earlier.

If i activate or deactivate the Gadgets i got following information:

Error: Error: SQL: 'deactivated'

Maybe like issue #604

RynFlutsch, 2019-02-23:
Working now on 7.3.0.
Thank You !

closed 2019-02-23 by RynFlutsch

RynFlutsch, 2019-03-01:
You have some advice to improve the battery life ?

kizniche, 2019-03-01:
Make fewer measurements and fewer transmissions. The device can store a lot of measurements, so you can reduce the transmission rate.

RynFlutsch, 2019-03-04:
Yes that was my idea, too. But how to set this up, properly?
Period time is setup how often the logged data is collected? 
And Set logging Interval how long the sensor is keeping the readings?
Ok and how to setup the reading intervals for the Sensor to store these measurements?

Or is period the reading interval and set logging interval the download interval for stored data? After downloading the measurements the storage is wiped?

kizniche, 2019-03-05:
*Period* is the duration between storing measurements on the device and *Download Interval* is the duration between downloading all the stored data from the device. The standard CR2032 3-volt battery is around ~220 mAh. You can modify the battery compartment by soldering wires onto the positive and negative leads and run them to a larger capacity 3-volt battery. [Here is a list of non-rechargeable 3-volt batteries offered by Digi-Key](https://www.digikey.com/products/en/battery-products/batteries-non-rechargeable-primary/90?FV=840068%2C840078%2C84000e%2C840096%2C840099%2C840012%2C8400b4%2C8400c4%2C840018%2C8400f6%2C840021%2C840171%2C840025%2C8401f2%2C8401f3%2C8401f5%2C8401f7%2C8401fa%2C8401fb%2C840201%2C840205%2C84020a%2C84004d%2C840313%2Cmu3V%7C2079%2Cffe0005a&quantity=0&ColumnSort=-33&page=1&stock=1&k=battery&pageSize=25&pkeyword=battery), sorted by highest mAh rating. You can also find appropriately-sized battery holders on Digi-Key that will provide leads to solder to (if soldering directly to the battery, be careful and wear personal protective equipment [goggles, fire-retardant clothing covering all your skin, etc.] in case it explodes). The largest of these batteries has a capacity of 3 mAh, which is approximately 13 times larger than the stock battery.

kizniche, 2019-03-05:
[The LR14](https://www.digikey.com/product-detail/en/fdk-america-inc-a-member-of-fujitsu-group/LR14-F-BF2/SY631-F021-ND/6098558) particular 3-volt battery has a capacity of 7.55 Ah, which is 34x larger than the CR2032, and is more cost-effective. At ~$0.30, the CR2032 costs $1.36/Ah, where the LR14, at ~$3.32 costs $0.44/Ah.

RynFlutsch, 2019-03-06:
Wow great, thank you !
I setup 30 seconds Period and 10800 sec for logging interval.
You know how many measurements can be stored on the device?

kizniche, 2019-03-06:
I've built in a check to make sure the user can't set a Period/Download Interval that would surpass the logger storage, but if you want to read more about this, check out [this issue](https://github.com/Sensirion/SmartGadget-Firmware/issues/3).

Short answer:

> You get the size of the buffer the device is able to provide for data logging (4 Bytes per Value so 8 Bytes for temperature and humidity).
> So it should be for NEED_CONSOLE_OUTPUT = 0
> (256-127) * 1024 / (4 * 2) = 16512
>
> for NEED_CONSOLE_OUTPUT != 0
> (256-145) * 1024 / (4 * 2) = 14208

RynFlutsch, 2019-03-07:
I really want to use these Bluetooth sensors for PID but its difficult if i can't build on on them ;-(
The Dashboard is showing me a measurement.
Live Page show me no data.
Why its so volatile?
Log:
`2019-03-07 12:42:38,241 - mycodo.input_5f13e0de - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-03-07 12:43:08,242 - mycodo.sht31_smart_gadget_5f13e0de - ERROR - Could not connect: Failed to connect to peripheral FD:E1:1B:A9:D4:8E, addr type: random
2019-03-07 12:44:53,891 - mycodo.sht31_smart_gadget_5f13e0de - ERROR - InputModule raised an exception when taking a reading: 'SHT31' object has no attribute '_SHT31__characteristics'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 362, in get_measurement
    self.connect()
  File "/root/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 176, in connect
    addr=self.location, iface=self.bt_adapter)
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 103, in __init__
    self.__prepareGadget()
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 108, in __prepareGadget
    self.__peripheral.getCharacteristics(uuid=UUID(0x2A23))[0],
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 508, in getCharacteristics
    rsp = self._getResp('find')
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 416, in _getResp
    self.delegate.handleNotification(hnd, data)
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 82, in handleNotification
    self.parent.setHumidityNotification(False)
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 189, in setHumidityNotification
    self.__setHumidityNotification(tmp)
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 193, in __setHumidityNotification
    self.__characteristics['Humidity'].valHandle + 2,
AttributeError: 'SHT31' object has no attribute '_SHT31__characteristics'
`