kizniche/Mycodo #592: Conditional function using feedback from TH16 never changes state.

SAM26K, 2019-01-04:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.03
 
#### Problem Description
Conditional function using feedback from TH16 never changes state.  Is the output a string or number?
See screenshots of data input and one of the conditional functions.  Tried with and without the float cast. Neither work.
[image]
[image]

kizniche, 2019-01-04:
Thanks for the bug report. I'll try to replicate your setup and test.

SAM26K, 2019-01-04:
Did some experimenting and if I change the action outputs from the TH16 to a remote pi with relays controlled by GPIO pins it does toggle when it should.  The output controls for the TH16 does change the TH16's relay if done manually.  So it seems to be something between the conditional:condition function and the TH16 output control.  Also when using the TH16  output control it never even sends the email whereas it does send an email if the GPIO relays are used.
Let me know if you want me to try something or if you need to logon to the system.
Thanks Again!

kizniche, 2019-01-04:
I can't replicate the issue. Are there any errors in the daemon log? If not, I'll need SSH and Mycodo UI access to figure out what's wrong.

SAM26K, 2019-01-05:
I changed the temp output of the TH16 data control to deg C (no conversion) and it works (toggles when its supposed to).  Will repeat test with deg F just to make sure.  Daemon does have some errors but computer went into sleep mode overnight and had to restart things to get it working again so I think their unrelated:

2019-01-04 17:14:01,470 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 938, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-01-04 17:14:01,640 - mycodo.daemon - ERROR - Could not query output state: 'NoneType' object has no attribute 'output_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 938, in output_state
    return self.controller['Output'].output_state(output_id)
AttributeError: 'NoneType' object has no attribute 'output_state'
2019-01-04 17:14:07,932 - mycodo.output - INFO - Output controller activated in 19382.2 ms
2019-01-04 17:14:08,618 - mycodo.conditional_36aedeb1 - INFO - Activated in 81.8 ms
2019-01-04 17:14:08,984 - mycodo.conditional_f200d074 - INFO - Activated in 275.3 ms
2019-01-04 17:14:09,116 - mycodo.conditional_9acfa451 - INFO - Activated in 81.5 ms
2019-01-04 17:14:09,246 - mycodo.conditional_5f998c4a - INFO - Activated in 79.8 ms
2019-01-04 17:14:09,247 - mycodo.daemon - INFO - All activated Conditional controllers started
2019-01-04 17:14:09,247 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-01-04 17:14:09,934 - mycodo.input_64a5f8ec - INFO - Activated in 603.6 ms
2019-01-04 17:14:10,253 - mycodo.input_65491931 - INFO - Activated in 233.4 ms
2019-01-04 17:14:11,471 - mycodo.input_1ab8f7c4 - INFO - Activated in 1078.8 ms
2019-01-04 17:14:12,924 - mycodo.input_dc446b55 - INFO - Activated in 1278.8 ms
2019-01-04 17:14:12,949 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-04 17:14:12,953 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-04 17:14:12,954 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-04 17:14:12,954 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-04 17:14:13,463 - mycodo.daemon - INFO - Mycodo daemon started in 25.602 seconds
2019-01-04 17:14:13,528 - mycodo.daemon - INFO - 51.09 MB RAM in use
2019-01-04 17:20:34,742 - mycodo.conditional_69a7d632 - INFO - Activated in 81.8 ms
2019-01-04 17:51:44,630 - mycodo.conditional_a6f92a78 - INFO - Activated in 81.5 ms
2019-01-04 17:58:21,585 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response',))

SAM26K, 2019-01-05:
Confirmed that it works if deg C is used.  deg F doesn't ever toggle the conditionals.

kizniche, 2019-01-05:
Thanks, I'll look into it.

All measurement conversions are handled by a conversion framework I built for Mycodo. It can handle any unit being returned from the sensor. Celsius is what I use for all temperature sensors by default, as a standard convention, before the user chooses to convert to a different unit.

SAM26K, 2019-01-05:
Latest from daemon log with temps set to deg F:
2019-01-04 19:58:13,659 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 19:58:45,397 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 19:59:13,592 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 19:59:45,181 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 19:59:53,836 - mycodo.conditional_69a7d632 - INFO - Deactivated in 142.1 ms
2019-01-04 20:00:06,669 - mycodo.conditional_69a7d632 - INFO - Activated in 185.9 ms
2019-01-04 20:00:17,218 - mycodo.input_1ab8f7c4 - INFO - Deactivated in 45.6 ms
2019-01-04 20:00:33,071 - mycodo.input_1ab8f7c4 - INFO - Activated in 285.4 ms
2019-01-04 20:00:45,316 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:01:07,000 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:01:45,454 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:02:06,999 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:02:45,309 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:03:07,069 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:03:45,368 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:04:06,967 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:04:45,368 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:05:07,078 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:05:45,203 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:06:06,885 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:06:45,275 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:07:06,972 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:07:45,244 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:08:06,806 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:08:45,326 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:09:06,877 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:09:45,127 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:10:06,940 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:10:45,183 - mycodo.conditional_a6f92a78 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
2019-01-04 20:11:06,990 - mycodo.conditional_69a7d632 - ERROR - Could not determine measurement from measurement ID: 7da12220-e154-426d-a15e-5563497b418f
Thanks Again!

kizniche, 2019-01-05:
I found the issue and pushed a fix. You can upgrade to master or modify your controller_conditional.py to use the fix or wait for the 7.0.4 release.

closed 2019-01-05 by kizniche

SAM26K, 2019-01-05:
Updated to the master and changed to "convert to F" but now the function controls have problems configuring.  Existing functions that handle TH16's won't deactivate so no re-editing. New conditional:conditonal controls won't let me config measurement readings.  Keeps asking me to add a device.

kizniche, 2019-01-05:
Can you paste the entirety of the errors you're receiving?

SAM26K, 2019-01-05:
It doesn't generate any error messages to the user but here is daemon log:

2019-01-05 11:15:22,840 - mycodo.input_1ab8f7c4 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-05 11:15:37,879 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x65caf7d0>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:15:52,839 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x714f0770>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:16:07,800 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7447e730>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:16:07,800 - mycodo.input_1ab8f7c4 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-05 11:16:22,920 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x70ce2970>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:16:37,879 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7449b8d0>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:16:52,840 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x70ce83d0>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:16:52,841 - mycodo.input_1ab8f7c4 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-05 11:17:07,879 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x65cf4fd0>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:17:22,840 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x714b1ff0>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:17:37,880 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x714b1a30>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:17:37,881 - mycodo.input_1ab8f7c4 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-05 11:17:52,839 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7444cc70>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:18:07,800 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x716a9c70>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:18:22,840 - mycodo.th16_1ab8f7c4 - ERROR - InputModule.get_measurement() method raised IOError: HTTPConnectionPool(host='192.168.1.37', port=80): Max retries exceeded with url: /cm?cmnd=status%2010 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x65ca79b0>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:18:22,841 - mycodo.input_1ab8f7c4 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
[... cut ...] tion.HTTPConnection object at 0x74d33ef0>: Failed to establish a new connection: [Errno 113] No route to host',))
2019-01-05 11:33:15,678 - mycodo.input_65491931 - INFO - Deactivated in 96.9 ms
2019-01-05 11:34:05,328 - mycodo.input_65491931 - INFO - Activated in 594.1 ms
2019-01-05 13:22:32,649 - mycodo.trigger_function_actions_a6f92a78 - ERROR - Wait 296 seconds to email again.
2019-01-05 13:25:10,462 - mycodo.trigger_function_actions_36aedeb1 - ERROR - Wait 138 seconds to email again.

Thanks

SAM26K, 2019-01-06:
> I'm not sure what this means.

I just noticed that none of the existing functions will deactivate with the
master update.    Clicking on deactive does nothing.

kizniche, 2019-01-06:
What functions are referring to? You mentioned an error requesting to add a device. I need more I formation in order to understand what you are doing and what errors are being generated. If you think your system is misconfigured somehow, it may be easier to just start over fresh.

SAM26K, 2019-01-06:
They are all conditional:conditional's. Can't deactivate or edit.
[image]
[image]

kizniche, 2019-01-06:
Not sure what's wrong with your system. None of mine on the master branch are preventing me from deactivating conditionals.

SAM26K, 2019-01-06:
Also all communication connections work ok.  The function that was using deg C still triggers when condition is met. with conversion to deg F.

SAM26K, 2019-01-06:
Oh well guess I'll do a reinstall unless you have any other ideas.

SAM26K, 2019-01-06:
Was able to restore previous working v7.03 release and changed the
comparison parameters from C to F values to trigger events. Then I
reupdated to the master branch.  The conditional functions switched ok
using deg F but same problem with the function > conditional:conditional
controls became unresponsive to deactivation and editing.
I think for now I'm just going to use the release version 7.3 and degrees C
which is working fine.  Its not worth all the work to do a complete
reinstall of the software and hours of recreating controls just to have deg
F.  I have 3 TH16s and 2 are inside the tent so I can use the other one to
have a F reading and not use its parameters for conditional functions.
I'm probably the first one to use TH16s with Mycodo and Tasmota so will
give it some time to see if anyone else has the problem.
Really appreciate all your work to get this hardware module working and if
you come up with any other ideas besides a complete reinstall please let me
know.
Thanks and    Best Regards

SAM26K, 2019-01-07:
Hi Kyle,
Found a fluke with Mycodo on my system.  At one point I created a 433Mhz output control just to see what it looked like.  I later deleted it.  For some reason it still shows up as an output selection on the dashboard objects.  Is there something I can delete or edit to get rid of it?  It maybe causing my function control problem.
Thanks.

SAM26K, 2019-01-07:
See screen shot. The phantom output is the line below the first TH16.
[image]

[image]

SAM26K, 2019-01-07:
Follow up.  I created a dashboard output  control for the phantom 433Mhz device and got an error in the daemon log as follows:

2019-01-06 23:36:22,296 - mycodo.daemon - INFO - Mycodo daemon v7.0.3 starting
2019-01-06 23:36:22,977 - mycodo.daemon - INFO - Anonymous statistics enabled
2019-01-06 23:36:23,025 - mycodo.daemon - INFO - Starting rpyc server
2019-01-06 23:36:43,020 - mycodo.output - INFO - Output controller activated in 19563.6 ms
2019-01-06 23:36:43,778 - mycodo.conditional_36aedeb1 - INFO - Activated in 153.8 ms
2019-01-06 23:36:44,425 - mycodo.conditional_f200d074 - INFO - Activated in 474.7 ms
2019-01-06 23:36:44,603 - mycodo.conditional_9acfa451 - INFO - Activated in 82.8 ms
2019-01-06 23:36:44,736 - mycodo.conditional_5f998c4a - INFO - Activated in 82.1 ms
2019-01-06 23:36:44,870 - mycodo.conditional_a6f92a78 - INFO - Activated in 82.3 ms
2019-01-06 23:36:45,011 - mycodo.conditional_69a7d632 - INFO - Activated in 88.7 ms
2019-01-06 23:36:45,011 - mycodo.daemon - INFO - All activated Conditional controllers started
2019-01-06 23:36:45,012 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-01-06 23:36:45,678 - mycodo.input_64a5f8ec - INFO - Activated in 580.9 ms
2019-01-06 23:36:46,003 - mycodo.input_65491931 - INFO - Activated in 230.4 ms
2019-01-06 23:36:46,802 - mycodo.input_1ab8f7c4 - INFO - Activated in 701.0 ms
2019-01-06 23:36:48,305 - mycodo.input_dc446b55 - INFO - Activated in 1396.7 ms
2019-01-06 23:36:48,314 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-06 23:36:48,318 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-06 23:36:48,319 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-06 23:36:48,320 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-06 23:36:48,823 - mycodo.daemon - INFO - Mycodo daemon started in 26.526 seconds
2019-01-06 23:36:48,849 - mycodo.daemon - INFO - 48.83 MB RAM in use
2019-01-07 00:11:18,497 - mycodo.output - WARNING - Invalid pin for output 10 (Wireless 315/433MHz LPD/SRD (rpi-rf)): None.

SAM26K, 2019-01-07:
Since I know there is a fluke with my install, I'm re-installing from scratch again.  It might be a fluke with the 433Mhz module in that it is not getting completely deleted from the system.  Plan is to reinstall 7.0.3 and then force upgrade to the master again.  Will let you know how it goes.
Thanks.

SAM26K, 2019-01-08:
Unfortunately the same problem as above after upgrading from 7.0.3 to 7.0.4,  existing function conditional:conditional's become unresponsive to deactivation and editing. Did a complete reinstall of 7.03 which works (except for testing deg C).  What code modules are involved so I can take a look?
TIA

SAM26K, 2019-01-08:
Last 30 lines of daemon logs:
2019-01-08 00:21:59,338 - mycodo.daemon - INFO - All Math controllers stopped
2019-01-08 00:21:59,351 - mycodo.conditional_850e517f - INFO - Deactivated in 16.1 ms
2019-01-08 00:21:59,416 - mycodo.conditional_18e09f78 - INFO - Deactivated in 81.2 ms
2019-01-08 00:21:59,417 - mycodo.input_0c7f8863 - INFO - Deactivated in 82.4 ms
2019-01-08 00:21:59,417 - mycodo.input_ee7d6b18 - INFO - Deactivated in 81.9 ms
2019-01-08 00:21:59,430 - mycodo.input_2d39f112 - INFO - Deactivated in 95.3 ms
2019-01-08 00:21:59,431 - mycodo.daemon - INFO - All Input controllers stopped
2019-01-08 00:21:59,432 - mycodo.daemon - INFO - All Trigger controllers stopped
2019-01-08 00:21:59,433 - mycodo.daemon - INFO - All Conditional controllers stopped
2019-01-08 00:22:14,434 - mycodo.daemon - INFO - Mycodo terminated in 15.632 seconds

2019-01-08 00:34:50,303 - mycodo.daemon - INFO - Mycodo daemon v7.0.4 starting
2019-01-08 00:34:53,655 - mycodo.daemon - INFO - Anonymous statistics enabled
2019-01-08 00:34:53,680 - mycodo.daemon - INFO - Starting rpyc server
2019-01-08 00:35:14,962 - mycodo.output - INFO - Output controller activated in 20801.7 ms
2019-01-08 00:35:15,717 - mycodo.conditional_18e09f78 - INFO - Activated in 152.2 ms
2019-01-08 00:35:16,198 - mycodo.conditional_850e517f - INFO - Activated in 372.9 ms
2019-01-08 00:35:16,200 - mycodo.daemon - INFO - All activated Conditional controllers started
2019-01-08 00:35:16,200 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-01-08 00:35:17,246 - mycodo.input_0c7f8863 - INFO - Activated in 888.8 ms
2019-01-08 00:35:17,842 - mycodo.input_1c95ca88 - INFO - Activated in 417.7 ms
2019-01-08 00:35:19,064 - mycodo.input_2d39f112 - INFO - Activated in 1045.1 ms
2019-01-08 00:35:20,617 - mycodo.input_ee7d6b18 - INFO - Activated in 1235.6 ms
2019-01-08 00:35:20,625 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-08 00:35:20,629 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-08 00:35:20,630 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-08 00:35:20,631 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-08 00:35:21,138 - mycodo.daemon - INFO - Mycodo daemon started in 30.831 seconds
2019-01-08 00:35:21,292 - mycodo.daemon - INFO - 48.42 MB RAM in use

SAM26K, 2019-01-08:
Issue 595 sounds like the same problem. Maybe there is something common between the two systems that's causing it?

kizniche, 2019-01-08:
I fixed a minor issue with Conditionals and Triggers having a form requirement before being able to be submitted. If you upgrade to master you can use the fix, if it's preventing you form interacting with Conditionals (or other Functions).

SAM26K, 2019-01-09:
Upgrading from 7.03 to the master is failing.  Execute suggested commands in feedback text and it ends up with "wrong database" in system settings. Will try to upgrade to 7.0.4 release first and then to the master.
Feedback from failed install. Tried both suggestions and didnt work:
[2019-01-09 02:02:49] Failed: Error while updating database with alembic.
[2019-01-09 02:02:49] 
[2019-01-09 02:02:49] 
[2019-01-09 02:02:49] There was an error during the upgrade.
[2019-01-09 02:02:49] Initial steps to try to fix:
[2019-01-09 02:02:49] 1. Reboot
[2019-01-09 02:02:49] 2. If that doesn't fix the issue, run the following command:
[2019-01-09 02:02:49] sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh upgrade
[2019-01-09 02:02:49] 3. If that command returns that you are running the latest version, run the following command:
[2019-01-09 02:02:49] sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh

SAM26K, 2019-01-09:
Great news! Upgrading from 7.0.3 (working) to release 7.0.4 (has form req and doesn't  work) to master 7.04 works! No install errors. One remaining problem is that the slave Rpi with the GPIO relays is no longer responding to remote commands but that should be easy to fix.  Very happy for now.