kizniche/Mycodo #593: Conditional: Measurement?

pmunz75, 2019-01-06:
Hello Kyle,

where are the settings in version 7.0.3 to find?
Setup ---> Functions ---> Conditional: Measurement?
thanks for your help

Screenshoot from 6.4.7

[image]

kizniche, 2019-01-08:
I don't quite understand the question? What is the issue?

pmunz75, 2019-01-08:
Sorry my english is bad,
I can not find this settings anymore. ( Conditional: Measurement )
where can it be found now?
if humidity is too low ---> I would like to switch my humidifier (output).

Thanks and sorry my english

kizniche, 2019-01-08:
I understand now. You will need to go to Setup -> Functions and add a "Conditional: Conditional".

Then, expand the Conditional, and next to "Add Condition", select "Mesurement", then click "Add Condition". Next to "Add Action", select "Output: Duration", then click "Add Action". Then configure both the measurement and action. You will then need to copy the ID from the measurement (It will look like ```{asdf1234} Mesaurement``` next to the particular measurement. Then make your conditional statement as follows:

```python
if {asdf1234} i snot None:
    if {asdf1234} < 40:
        print(1)
```

This will cause the output (humidifier) to be activated when the measurement (humidity) is lower than 40 (%).

pmunz75, 2019-01-08:
thank you for your help.
it works great!
Many Thanks

kizniche, 2019-01-08:
You can edit the following line:

https://github.com/kizniche/Mycodo/blob/76f1af2d7947caba51399172da9aa7e5eb5d0afa/mycodo/utils/function_actions.py#L509

to this:

```python
    logger_actions.info(message)
```

Save, restart the backend (```sudo service mycodo restart```), and then every time actions are triggered, it will print then entire Conditional message to the daemon log (Config -> Mycodo Logs -> Daemon). Does this help?

kizniche, 2019-01-08:
You can also have an action be to create a note, which will save the entire message as a note. This is useful because you can then display the notes on a graph and see when the conditional was triggered in relation to your data.

closed 2019-01-09 by kizniche