kizniche/Mycodo #595: can not be deactivated / activated / Delete / Save ...

pmunz75, 2019-01-06:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.0.3
 
#### Problem Description
Function and other can not be deactivated / activated / Delete / Save ...
if no action is selected.
 
[image]

kizniche, 2019-01-06:
I haven't been able to replicate this issue. What browsers have you tested this with? This behavior isn't found with the latest versions of Firefox or Chrome.

pmunz75, 2019-01-06:
i use latest versions of Firefox.

kizniche, 2019-01-06:
Test another browser to determine if it is an issue with your browser.

pmunz75, 2019-01-06:
i tested Safari , the same problem .

kizniche, 2019-01-06:
Sounds like an issue with your install, since you are the only user experiencing this issue. Reinstall Mycodo and see if that fixes the issue.

pmunz75, 2019-01-06:
ok thanks.

kizniche, 2019-01-08:
I was finally able to replicate this issue.

closed 2019-01-08 by kizniche

not5, 2019-01-10:
I applied the new files and it's still not quite right. Still having error on trying to add an Action.

kizniche, 2019-01-10:
An action for what?

kizniche, 2019-01-10:
Does "applied the new files" mean you upgraded to master?

not5, 2019-01-10:
Sorry, should have been more specific. I overwrote the files associated with your change (the three flask forms).

Not working for a conditional. Must select an input in order to do anything, like add an action or even activate/deactivate.

kizniche, 2019-01-10:
There were more edits than those. Best to upgrade to master if you want to test, otherwise, wait until the next version release.