kizniche/Mycodo #597: Err 500: Baudrate is NoneType

DanMaslowski, 2019-01-07:
## Mycodo Issue Report:
Mycodo Version: 7.0.3 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
Database Version: 8e5a8351ad7a
Daemon Status: Running 
Daemon Process ID: 934 
Daemon RAM Usage: 45.396 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 50.292 MB 
Frontend Virtualenv: Yes

#### Problem Description
Please list:

  - attempting to calibrate my PH probe. Atlas Scientific
  - PH probe runs fine outside of mycodo
  
 
### Errors
  Error (Full Traceback):

Traceback (most recent call last):
  File "/root/Mycodo/env/lib/python3.5/site-packages/serial/serialutil.py", line 287, in baudrate
    b = int(baudrate)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 129, in setup_atlas_ph
    selected_input, 'temperature', temp, 'continuous', 1)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 290, in dual_commands_to_sensor
    atlas_command = AtlasScientificCommand(input_sel)
  File "/root/Mycodo/mycodo/utils/calibration.py", line 24, in __init__
    input_dev.uart_location, baudrate=input_dev.baud_rate)
  File "/root/Mycodo/mycodo/devices/atlas_scientific_uart.py", line 24, in __init__
    timeout=5)
  File "/root/Mycodo/env/lib/python3.5/site-packages/serial/serialutil.py", line 219, in __init__
    self.baudrate = baudrate
  File "/root/Mycodo/env/lib/python3.5/site-packages/serial/serialutil.py", line 289, in baudrate
    raise ValueError("Not a valid baudrate: {!r}".format(baudrate))
ValueError: Not a valid baudrate: None

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Install
 2. Add a phprobe
 3. calibrate it via the UI
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2019-01-07:
Thanks for the bug report. I'll look into this. Will you be willing to test potential fixes?

DanMaslowski, 2019-01-07:
Absolutly

kizniche, 2019-01-08:
I just pushed a potential fix. Can you [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and see if it fixes your issue?

DanMaslowski, 2019-01-08:
Mycodo Version: 7.0.4 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
Database Version: b4d958997cf0
Daemon Status: Running 
Daemon Process ID: 1983 
Daemon RAM Usage: 39.232 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 51.168 MB 
Frontend Virtualenv: Yes

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/root/Mycodo/env/lib/python3.5/site-packages/serial/serialutil.py", line 287, in baudrate
    b = int(baudrate)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 129, in setup_atlas_ph
    selected_input, 'temperature', temp, 'continuous', 1)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 290, in dual_commands_to_sensor
    atlas_command = AtlasScientificCommand(input_sel)
  File "/root/Mycodo/mycodo/utils/calibration.py", line 25, in __init__
    baudrate=input_dev.baud_rate)
  File "/root/Mycodo/mycodo/devices/atlas_scientific_uart.py", line 24, in __init__
    timeout=5)
  File "/root/Mycodo/env/lib/python3.5/site-packages/serial/serialutil.py", line 219, in __init__
    self.baudrate = baudrate
  File "/root/Mycodo/env/lib/python3.5/site-packages/serial/serialutil.py", line 289, in baudrate
    raise ValueError("Not a valid baudrate: {!r}".format(baudrate))
ValueError: Not a valid baudrate: None

DanMaslowski, 2019-01-08:
As an aside, I am using the ftdi abstraction of the atlas ph monitor. So:

root@raspberrypi:/home/pi/PI-PH-Alert# ls
ftdi.py  i2cAlertPi.py  i2c.py  LICENSE  lob.txt  README.md  test.py  uart.py
root@raspberrypi:/home/pi/PI-PH-Alert# python ftdi.py

Welcome to the Atlas Scientific Raspberry Pi FTDI Serial example.

    Any commands entered are passed to the board via UART except:
    Poll,xx.x command continuously polls the board every xx.x seconds
    Pressing ctrl-c will stop the polling

    Press enter to receive all data in buffer (for continuous mode)

Discovered FTDI serial numbers:

Index:  0  Serial:  DO00GCZM

Index:  1  Serial:  DO00GDBS
===================================
Please select a device index: 1

>> Opened device  DO00GDBS
>> Any commands entered are passed to the board via FTDI:
Enter command: R
3.946
*OK

DanMaslowski, 2019-01-08:
OK, so, I didn't remove the device the first time I tried it, which meant probably stale entries. (Baud was not set). So I went in and removed all devices and toasted dash readings as well. Added the sensor. This time the baud was set (9600).  No reading on the dash page. Now I get a 504 Gateway Time-out

"504 Gateway Time-out
nginx/1.10.3"

Going to try removing everything and restarting.

kizniche, 2019-01-08:
Looking at https://github.com/AtlasScientific/Raspberry-Pi-sample-code/ I can see that the FTDI will probably not work with the current Input module, as I only have code for I2C and UART. I can add the FTDI code.

DanMaslowski, 2019-01-08:
Yeah, i am figuring that out as well. I figured I would try a custom import module.  Looking at how you do configuration. Cool stuff by the way. Thanks! Will let you know if I get anywhere (as an aside, I also have the temp probe on an isolated USB board so... ftdi as well).

kizniche, 2019-01-08:
I'm about half way (maybe more) to adding FTDI support to the Atlas pH, PT1000 temp, and EC modules.

DanMaslowski, 2019-01-08:
You are a machine! I am just now figuring out how you did this. I was manually hacking your atlas_ph.py to get it running. I will keep poking at it, but, let me know when you have your kit ready and I will test it for you.

kizniche, 2019-01-08:
I think I got full support for FTDI... testing now. I'll push for you to test by upgrading to master in a few moments.

kizniche, 2019-01-08:
I just pushed the FTDI update. At the moment, when you add the pH (FTDI) Input, you will have to manually enter the FTDI Device ID. Looking above, it appears yours is "DO00GDBS". Once we get this working, I'll add the ability for Mycodo to detect attached devices and list them in a dropdown. But for now, it's just a simple text input box.

kizniche, 2019-01-08:
Wait, there needs to be another update. Don't upgrade to master yet.

kizniche, 2019-01-08:
Okay, just pushed some things I missed. You can go ahead and upgrade to master to test.

kizniche, 2019-01-08:
actually, my web server isn't starting. Hold off until I can get it working.

kizniche, 2019-01-08:
Okay, it seems to be working with that last commit. It should be safe to test now.

DanMaslowski, 2019-01-08:
Pulling....

DanMaslowski, 2019-01-08:
No upgrade is available. You are running the latest release, version 7.0.4

kizniche, 2019-01-08:
Did you [follow these steps](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch)?

DanMaslowski, 2019-01-08:
Yes. I probably stepped on something while you were pushing. Regardless, I toasted the config and am reinstalling from master. Let you know soon.

kizniche, 2019-01-08:
BTW, neither of the last two commits were impactful to FTDI, so you can still test with what version you are installing. I'll mention to upgrade if it's necessary.

kizniche, 2019-01-08:
Just pushed a fix for FTDI when calibrating. If testing calibration (if the input module works), you will need to upgrade to master before testing.

DanMaslowski, 2019-01-08:
OK. Reinstalled, forced upgrade and pulled latest. Probes show up as FTDI devices. Device id in setup page is blank. Input in to the textbox labeled FTDI device (tool tips suggests a /dev/ttyUSBx device by the way). I put the device ID in for the PH probe (DO00GDBS), saved the file and got this message:

    Success: Input controller activated in SQL database
    Success: Input controller with ID 631e3470-1b87-49ac-8f49-2a7273d66217 activated.

Went to calibration page and was able to select "2 (Atlas pH)". Tried to input a temp Error 500. Tried clear calibration. Error 500. (stack trace at the bottom).

Tried the temp controller. Put in:DO00GCZM (temp controller ID). Ui says it is activated. Error messages below. Live page shows the following:
Atlas PT-1000 (Input 7bb3c63b)
Input (ATLAS_PT1000), 15.0 second interval
Measurement | Timestamp
0 °C (Temperature) CH1 | No Data Last 16 sec
Atlas pH (Input 631e3470)
Input (ATLAS_PH), 15.0 second interval
Measurement | Timestamp
0 pH (Ion Concentration) CH1 | No Data Last 16 sec

Temp error messages
×

    Success: Input controller activated in SQL database

×

    Error: Could not activate Input controller with ID 7bb3c63b-3101-4859-918e-96572e281e65: name 'dev' is not defined

stack trace:

Mycodo Version: 7.0.4 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
Database Version: 5a70a31c71e8
Daemon Status: Running 
Daemon Process ID: 5013 
Daemon RAM Usage: 48.588 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 52.772 MB 
Frontend Virtualenv: Yes
Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 84, in setup_atlas_ph
    atlas_command = AtlasScientificCommand(selected_input)
  File "/root/Mycodo/mycodo/utils/calibration.py", line 24, in __init__
    input_dev.ftdi_location)
  File "/root/Mycodo/mycodo/devices/atlas_scientific_ftdi.py", line 20, in __init__
    dev.flush()
NameError: name 'dev' is not defined
On calibration I get this error:
Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 127, in setup_atlas_ph
    selected_input, 'temperature', temp, 'continuous', 1)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 309, in dual_commands_to_sensor
    atlas_command = AtlasScientificCommand(input_sel)
  File "/root/Mycodo/mycodo/utils/calibration.py", line 22, in __init__
    input_dev.ftdi_location)
  File "/root/Mycodo/mycodo/devices/atlas_scientific_ftdi.py", line 20, in __init__
    dev.flush()
NameError: name 'dev' is not defined

Will look at things more tonight...

kizniche, 2019-01-08:
Don't try to calibrate before you verify the new Input module is acquiring measurements. After activating the Input, are measurements appearing on the Live page? If not, please paste any errors from the daemon log, immediately following Input activation.

kizniche, 2019-01-08:
I see this minor issue is preventing Input activation. I'll push a fix shortly.

DanMaslowski, 2019-01-08:
Daemon logs:
2019-01-08 15:09:23,786 - mycodo.input_1ad6618a - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-08 15:09:38,814 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 15:09:53,743 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 15:10:08,775 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 15:10:08,776 - mycodo.input_1ad6618a - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-08 15:10:23,806 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'

kizniche, 2019-01-08:
I just pushed a fix for the ``` 'dev' is not defined``` issue. Upgrade to master and see if measurements can be acquired.

DanMaslowski, 2019-01-08:
Toasted system. Reinstalled. Forced upgrade. Added PH Probe. Forced measurment read. No joy. Log from Daemon:

2019-01-08 16:21:10,895 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-01-08 16:21:10,896 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-08 16:21:10,896 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-08 16:21:10,896 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-08 16:21:10,897 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-08 16:21:11,398 - mycodo.daemon - INFO - Mycodo daemon started in 1.762 seconds
2019-01-08 16:21:11,401 - mycodo.daemon - INFO - 38.41 MB RAM in use
2019-01-08 16:22:18,893 - mycodo.input_bd975b9f - INFO - Activated in 1563.2 ms
2019-01-08 16:22:18,895 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:22:23,410 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:22:32,630 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:22:32,631 - mycodo.input_bd975b9f - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.

DanMaslowski, 2019-01-08:
Atlas pH (Input bd975b9f)
Input (ATLAS_PH), 15.0 second interval
Measurement | Timestamp
3.978 pH (Ion Concentration) CH1 | 2019/1/8 9:37:14

DanMaslowski, 2019-01-08:
Temp Sensor: No joy

Atlas PT-1000 (Input 93046081)
Input (ATLAS_PT1000), 15.0 second interval
Measurement | Timestamp
0 °C (Temperature) CH1 | No Data Last 16 sec

  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:33:02,674 - mycodo.input_bd975b9f - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-08 16:33:11,704 - mycodo.daemon - INFO - Received command to terminate daemon
2019-01-08 16:33:12,586 - mycodo.daemon - INFO - All LCD controllers stopped
2019-01-08 16:33:12,587 - mycodo.daemon - INFO - All PID controllers stopped
2019-01-08 16:33:12,588 - mycodo.daemon - INFO - All Math controllers stopped
2019-01-08 16:33:12,593 - mycodo.input_bd975b9f - INFO - Deactivated in 7.1 ms
2019-01-08 16:33:12,594 - mycodo.daemon - INFO - All Input controllers stopped
2019-01-08 16:33:12,595 - mycodo.daemon - INFO - All Trigger controllers stopped
2019-01-08 16:33:12,595 - mycodo.daemon - INFO - All Conditional controllers stopped
2019-01-08 16:33:12,608 - mycodo.output - INFO - Output controller deactivated in 11.9 ms
2019-01-08 16:33:12,609 - mycodo.daemon - INFO - Mycodo terminated in 0.905 seconds

DanMaslowski, 2019-01-08:
Calibration of PH:  Success inputting temp. Looks like a some stripping of chars is in order:
Success: Calibration: temperature (23.96 C): ['*OK\r']
Success: Command: continuous: []
Atlas pH Sensor
2. Calibration 1 (Atlas pH)

Insert the probe into the pH 7.0 solution, then click Ready

Latest pH measurement: Error acquiring measurement

Running testing first measurement times outs.  Then at the end of the calibration of 7.0 solution:

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 130, in setup_atlas_ph
    selected_input, 'mid', '7.0', 'continuous', 2)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 309, in dual_commands_to_sensor
    atlas_command = AtlasScientificCommand(input_sel)
  File "/root/Mycodo/mycodo/utils/calibration.py", line 34, in __init__
    self.board_version, self.board_info = self.get_board_version()
  File "/root/Mycodo/mycodo/utils/calibration.py", line 50, in get_board_version
    info = self.ph_sensor_ftdi.read_lines()[0]
IndexError: list index out of range.

DanMaslowski, 2019-01-08:
Last 100 or so lines of the Daemon logs:  AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:30:47,685 - mycodo.input_bd975b9f - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-08 16:31:02,715 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:31:17,648 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:31:32,681 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:31:32,683 - mycodo.input_bd975b9f - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-08 16:31:47,713 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:32:02,645 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:32:17,679 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:32:17,681 - mycodo.input_bd975b9f - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-08 16:32:32,710 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:32:47,641 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:33:02,672 - mycodo.inputs.atlas_ph_ftdi_None - ERROR - InputModule raised an exception when taking a reading: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_ph.py", line 198, in get_measurement
    if self.atlas_sensor_uart.setup:
AttributeError: 'NoneType' object has no attribute 'setup'
2019-01-08 16:33:02,674 - mycodo.input_bd975b9f - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-01-08 16:33:11,704 - mycodo.daemon - INFO - Received command to terminate daemon
2019-01-08 16:33:12,586 - mycodo.daemon - INFO - All LCD controllers stopped
2019-01-08 16:33:12,587 - mycodo.daemon - INFO - All PID controllers stopped
[... cut ...]
2019-01-08 16:36:15,576 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-08 16:36:15,577 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-08 16:36:15,578 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-08 16:36:15,579 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-08 16:36:16,085 - mycodo.daemon - INFO - Mycodo daemon started in 3.463 seconds
2019-01-08 16:36:16,089 - mycodo.daemon - INFO - 41.28 MB RAM in use
2019-01-08 16:38:59,552 - mycodo.input_93046081 - INFO - Activated in 1410.1 ms

kizniche, 2019-01-08:
I just pushed a fix for the PT1000 Input module.

As for this error:

```
File "/root/Mycodo/mycodo/utils/calibration.py", line 34, in init
self.board_version, self.board_info = self.get_board_version()
File "/root/Mycodo/mycodo/utils/calibration.py", line 50, in get_board_version
info = self.ph_sensor_ftdi.read_lines()[0]
IndexError: list index out of range.
```

Can you run the FTDI script and use the command "i" (just the lowercase letter i), and see what the response is?

DanMaslowski, 2019-01-08:
Welcome to the Atlas Scientific Raspberry Pi FTDI Serial example.

    Any commands entered are passed to the board via UART except:
    Poll,xx.x command continuously polls the board every xx.x seconds
    Pressing ctrl-c will stop the polling

    Press enter to receive all data in buffer (for continuous mode)

Discovered FTDI serial numbers:

Index:  0  Serial:  DO00GCZM

Index:  1  Serial:  DO00GDBS
===================================
Please select a device index: 1

>> Opened device  DO00GDBS
>> Any commands entered are passed to the board via FTDI:
iEnter command:
7.012
7.014
?I,pH,2.11
*OK
7.012
Enter command:

kizniche, 2019-01-08:
I'm not quite sure what I'm reading here.

DanMaslowski, 2019-01-08:
Sorry, this is the output of the script in the raspberry pi sample code from atlas. 
python ftdi.py

Did you mean a different FTDI script?

Ran it again:
Welcome to the Atlas Scientific Raspberry Pi FTDI Serial example.

    Any commands entered are passed to the board via UART except:
    Poll,xx.x command continuously polls the board every xx.x seconds
    Pressing ctrl-c will stop the polling

    Press enter to receive all data in buffer (for continuous mode)

Discovered FTDI serial numbers:

Index:  0  Serial:  DO00GCZM

Index:  1  Serial:  DO00GDBS
===================================
Please select a device index: 1

>> Opened device  DO00GDBS
>> Any commands entered are passed to the board via FTDI:
Enter command: i
7.009
7.007
7.009
?I,pH,2.11
*OK
7.007
Enter command: i
7.008
7.007
7.008
?I,pH,2.11
*OK
7.008
Enter command:

kizniche, 2019-01-08:
I'm referring to the formatting. That's the correct script, but it looks odd. It's also not returning the board version as it should.

DanMaslowski, 2019-01-08:
It looks to me like it is out putting a bunch of readings followed by ?I,RTD,2.10 (for the temp) and ?I,PH,2.11 for the ph probe.

So, parse the returns until you get the desired ?I,xxx,xxx?

DanMaslowski, 2019-01-08:
In calibration it blew up:
Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/root/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 127, in setup_atlas_ph
    selected_input, 'temperature', temp, 'continuous', 1)
  File "/root/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 309, in dual_commands_to_sensor
    atlas_command = AtlasScientificCommand(input_sel)
  File "/root/Mycodo/mycodo/utils/calibration.py", line 34, in __init__
    self.board_version, self.board_info = self.get_board_version()
TypeError: 'NoneType' object is not iterable

DanMaslowski, 2019-01-08:
For temp in line 97 of atlas_pt1000.py you never enter FTDI setup and it looks like this will break UART. Suggest changing to "== FTDI":

if self.interface == 'UART':
            if self.atlas_sensor_ftdi.setup:
                self.atlas_sensor_ftdi.send_cmd('R')

DanMaslowski, 2019-01-08:
Tested it locally and it works...

kizniche, 2019-01-08:
Just pushed a fix for the PT1000 and potential fix for pH calibration.

kizniche, 2019-01-08:
Just pushed a potential fix.

kizniche, 2019-01-08:
> Temp fix worked. Calibration failed gracefully. Log:

Can you describe what you are doing when you get an error. Specifically, can you explain how to replicate the above error?

DanMaslowski, 2019-01-08:
Enter Setup--> Calibration&Setup--> Select One
Select atlas ph sensor
Begin Calibration
Enter 23.97  ---> Continue to next stage
Error: Calibration: temperature (23.97 C): Default message
Atlas pH Sensor
Error: Calibration Failed

Refer to the logs for details

Logs:
2019-01-08 22:25:01,032 - mycodo.daemon - INFO - Mycodo daemon v7.0.4 starting
2019-01-08 22:25:01,382 - mycodo.daemon - INFO - Anonymous statistics enabled
2019-01-08 22:25:01,408 - mycodo.daemon - INFO - Starting rpyc server
2019-01-08 22:25:01,796 - mycodo.output - INFO - Output controller activated in 124.5 ms
2019-01-08 22:25:02,298 - mycodo.daemon - INFO - All activated Conditional controllers started
2019-01-08 22:25:02,298 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-01-08 22:25:04,018 - mycodo.input_3653c8dd - INFO - Activated in 1628.3 ms
2019-01-08 22:25:05,550 - mycodo.input_ad02a970 - INFO - Activated in 1396.5 ms
2019-01-08 22:25:05,556 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-08 22:25:05,558 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-08 22:25:05,559 - mycodo.inputs.atlas_electrical_conductivity_ftdi_DO00GCZM - ERROR - InputModule raised an exception when taking a reading: list index out of range
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/root/Mycodo/mycodo/inputs/atlas_pt1000.py", line 105, in get_measurement
    elif str_is_float(lines[0]):
IndexError: list index out of range
2019-01-08 22:25:05,559 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-08 22:25:05,564 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-08 22:25:06,065 - mycodo.daemon - INFO - Mycodo daemon started in 5.032 seconds
2019-01-08 22:25:06,072 - mycodo.daemon - INFO - 42.11 MB RAM in use

kizniche, 2019-01-08:
Can you explain what this means?

```
Error: Calibration: temperature (23.97 C): Default message
Atlas pH Sensor
Error: Calibration Failed
```

Can you take a screenshot instead?

kizniche, 2019-01-08:
I just pushed a commit that adds debugging lines to see if the proper board version is detected. Can you upgrade to master then try calibration again?

DanMaslowski, 2019-01-08:
I'm away from the computer. Logs could have been stale. Will try again and
capture screenshots

kizniche, 2019-01-09:
I don't see the "Device Info" string in the log, so it doesn't look like you attempted to calibrate during this log period.

DanMaslowski, 2019-01-09:
I did, but let me restart everything from scratch. The screenshot is the
step where it gracefully bails. You will see it all the way on the left
monitor....

kizniche, 2019-01-09:
If there are errors with the PT1000, I would recommend deactivating it while you test the pH calibration, so you don't fill your log with errors, making it harder to parse the log.

DanMaslowski, 2019-01-09:
I am starting over from scratch. Will ignore temp and just do PH. Give me ten min to pull from master.

DanMaslowski, 2019-01-09:
Pulled from master. Set up PH. Grabbed a reading from the data page. Perfect. Went to Dash. Set up a ph widget. Perfect. Went to setup. selected PH mon.  Entered temp 23.96. Selected next. Error (screen shot attached). Suspect it is bailing before logging as logs are empty and attached:

2019-01-09 03:48:03,716 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-09 03:48:03,717 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-09 03:48:03,717 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-09 03:48:03,717 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-09 03:48:04,218 - mycodo.daemon - INFO - Mycodo daemon started in 1.713 seconds
2019-01-09 03:48:04,226 - mycodo.daemon - INFO - 38.15 MB RAM in use
2019-01-09 03:50:08,624 - mycodo.daemon - INFO - Received command to terminate daemon
2019-01-09 03:50:08,725 - mycodo.daemon - INFO - All LCD controllers stopped
2019-01-09 03:50:08,726 - mycodo.daemon - INFO - All PID controllers stopped
2019-01-09 03:50:08,726 - mycodo.daemon - INFO - All Math controllers stopped
2019-01-09 03:50:08,726 - mycodo.daemon - INFO - All Input controllers stopped
2019-01-09 03:50:08,727 - mycodo.daemon - INFO - All Trigger controllers stopped
2019-01-09 03:50:08,727 - mycodo.daemon - INFO - All Conditional controllers stopped
2019-01-09 03:50:08,741 - mycodo.output - INFO - Output controller deactivated in 14.2 ms
2019-01-09 03:50:08,742 - mycodo.daemon - INFO - Mycodo terminated in 0.118 seconds

2019-01-09 03:53:12,412 - mycodo.daemon - INFO - Mycodo daemon v7.0.4 starting
2019-01-09 03:53:12,775 - mycodo.daemon - INFO - Anonymous statistics enabled
2019-01-09 03:53:12,799 - mycodo.daemon - INFO - Starting rpyc server
2019-01-09 03:53:13,200 - mycodo.output - INFO - Output controller activated in 126.2 ms
2019-01-09 03:53:13,701 - mycodo.daemon - INFO - All activated Conditional controllers started
2019-01-09 03:53:13,702 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-01-09 03:53:13,702 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-09 03:53:13,702 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-09 03:53:13,703 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-09 03:53:13,703 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-09 03:53:14,204 - mycodo.daemon - INFO - Mycodo daemon started in 1.791 seconds
2019-01-09 03:53:14,211 - mycodo.daemon - INFO - 38.20 MB RAM in use
2019-01-09 03:55:10,424 - mycodo.input_f2c6f067 - INFO - Activated in 1564.1 ms
[image]

DanMaslowski, 2019-01-09:
Wrong button. Happy to arrange a live debug session if you like.

kizniche, 2019-01-09:
That's not making any sense. You should be seeing two log lines in the daemon log, specified here:

https://github.com/kizniche/Mycodo/blob/db3afd82bdbea80f283429b5b13f509037813ac5/mycodo/utils/calibration.py#L41-L42

If you could provide me SSH access and a Mycodo login, I could rapidly diagnose the issue. I might not have enough time tonight to work on it, but if you're familiar with router port forwarding to allow SSH/HTTPS, I might be able to figure it out tonight.

DanMaslowski, 2019-01-09:
I will be happy to. Have to punch a hole through two gateways.  Let's punt until tomorrow. I have been working (*my real job) since 5 AM and need a break. Appreciate all the work you are putting into this. And honestly enjoying banging code again. Been a while. Anyway, will ping you when I get it set up...

DanMaslowski, 2019-01-30:
> What's the status of this? Is everything working?

Hi Kyle,

Very sorry, I got distracted. Yes base functionality works great.
Calibration is still down. I will find some time this weekend to punch a
hole in my firewall. In touch soon...

closed 2019-10-03 by kizniche