kizniche/Mycodo #599: Error upgrading from 7.0.4 release to forced upgrade to 7.0.4 master.

SAM26K, 2019-01-10:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.0.4
 
#### Problem Description
Note: This install is from my second RPi3B. Had to go from version 5.n straight to 7.0.4.  The first RPi 3B's install of 7.0.4 master worked.  The main difference is the first RPi was upgraded from 7.0.3 to 7.0.4 release and finally to 7.0.4 master.  The second RPi3B's install skipped 7.0.3.

Error upgrading from 7.0.4 release to forced upgrade to 7.0.4 master.
See screenshots of upgrade page and system info page:
[image]

Error 500 after attempting upgrade to master:

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 509, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: output.trigger_functions_at_start

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1473, in page_output
    output = Output.query.all()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2843, in all
    return list(self)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2995, in __iter__
    return self._execute_and_instances(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 3018, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 948, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/sql/elements.py", line 269, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1060, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1200, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1413, in _handle_dbapi_exception
    exc_info
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 265, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 248, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 509, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: output.trigger_functions_at_start [SQL: 'SELECT output.id AS output_id, output.unique_id AS output_unique_id, output.output_type AS output_output_type, output.interface AS output_interface, output.location AS output_location, output.i2c_bus AS output_i2c_bus, output.baud_rate AS output_baud_rate, output.name AS output_name, output.measurement AS output_measurement, output.unit AS output_unit, output.conversion_id AS output_conversion_id, output.channel AS output_channel, output.pin AS output_pin, output."trigger" AS output_trigger, output.amps AS output_amps, output.on_at_start AS output_on_at_start, output.on_until AS output_on_until, output.last_duration AS output_last_duration, output.on_duration AS output_on_duration, output.protocol AS output_protocol, output.pulse_length AS output_pulse_length, output.on_command AS output_on_command, output.off_command AS output_off_command, output.pwm_command AS output_pwm_command, output.trigger_functions_at_start AS output_trigger_functions_at_start, output.pwm_hertz AS output_pwm_hertz, output.pwm_library AS output_pwm_library, output.pwm_invert_signal AS output_pwm_invert_signal, output.flow_rate AS output_flow_rate \nFROM output'] (Background on this error at: http://sqlalche.me/e/e3q8)

Tried all three steps in feedback from install, no help:

[2019-01-09 23:49:47] There was an error during the upgrade.
[2019-01-09 23:49:47] Initial steps to try to fix:
[2019-01-09 23:49:47] 1. Reboot
[2019-01-09 23:49:47] 2. If that doesn't fix the issue, run the following command:
[2019-01-09 23:49:47] sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh upgrade
[2019-01-09 23:49:47] 3. If that command returns that you are running the latest version, run the following command:
[2019-01-09 23:49:47] sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh

SAM26K, 2019-01-10:
This is feedback list while upgrading to 7.0.4 master:
[2019-01-09 23:48:05] 
[2019-01-09 23:48:05] 
[2019-01-09 23:48:05] Upgrade script executed with the 'force-upgrade-master' argument. Upgrading from github repo master.
[2019-01-09 23:48:05] 
[2019-01-09 23:48:05] #### Upgrade to v7.0.4 initiated 2019-01-09_23-47-57 ####
[2019-01-09 23:48:05] 
[2019-01-09 23:48:05] #### Beginning Upgrade: Stage 1 of 2 ####
[2019-01-09 23:48:09] Stopping the Mycodo daemon...Done.
[2019-01-09 23:49:06] Downloading latest Mycodo version to /home/pi/Mycodo-master.tar.gz...Done.
[2019-01-09 23:49:06] Creating /tmp/Mycodo-7.0.4...Done.
[2019-01-09 23:49:07] Extracting /home/pi/Mycodo-master.tar.gz to /tmp/Mycodo-7.0.4...Done.
[2019-01-09 23:49:07] Removing /home/pi/Mycodo-master.tar.gz...Done.
[2019-01-09 23:49:07] Copying /tmp/Mycodo-7.0.4/.upgrade status file to /tmp/Mycodo-7.0.4...Done.
[2019-01-09 23:49:07] Moving env directory...Done.
[2019-01-09 23:49:07] Copying databases from /home/pi/Mycodo/databases/ to /tmp/Mycodo-7.0.4/databases...Done.
[2019-01-09 23:49:07] Copying flask_secret_key from /home/pi/Mycodo/databases/ to /tmp/Mycodo-7.0.4/databases...Done.
[2019-01-09 23:49:07] Copying statistics ID...Done.
[2019-01-09 23:49:07] Copying SSL certificates...Done.
[2019-01-09 23:49:07] Copying mycodo/inputs/custom_inputs...Done.
[2019-01-09 23:49:07] #### Stage 1 of 2 Complete ####
[2019-01-09 23:49:07] 
[2019-01-09 23:49:07] #### Continuing Upgrade: Stage 2 of 2 ####
[2019-01-09 23:49:07] 
[2019-01-09 23:49:07] Moving old Mycodo from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2019-01-09_23-47-57-7.0.4...Done.
[2019-01-09 23:49:07] Moving new Mycodo from /tmp/Mycodo-7.0.4 to /home/pi/Mycodo...Done.
[2019-01-09 23:49:38] 
[2019-01-09 23:49:38] #### Upgrading Mycodo database with alembic (if needed)
[2019-01-09 23:49:46] FAILED: Multiple head revisions are present for given argument 'head'; please specify a specific target revision, '@head' to narrow to a specific head, or 'heads' for all heads
[2019-01-09 23:49:47] Failed: Error while updating database with alembic.
[2019-01-09 23:49:47] 
[2019-01-09 23:49:47] 
[2019-01-09 23:49:47] There was an error during the upgrade.
[2019-01-09 23:49:47] Initial steps to try to fix:
[2019-01-09 23:49:47] 1. Reboot
[2019-01-09 23:49:47] 2. If that doesn't fix the issue, run the following command:
[2019-01-09 23:49:47] sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh upgrade
[2019-01-09 23:49:47] 3. If that command returns that you are running the latest version, run the following command:
[2019-01-09 23:49:47] sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh

kizniche, 2019-01-10:
My only recommendation would be to restore a previous version and try upgrading to a release. There are issues that can arise from upgrading to master. This is one reason why a backup that can be restored is created every time any upgrade is initiated.

closed 2019-01-10 by kizniche

kizniche, 2019-01-10:
Backups can be found on the backup/restore page.

SAM26K, 2019-01-10:
I dont have a backup of 7.3, can I do that from github?

kizniche, 2019-01-10:
7.3 can be found under releases

https://github.com/kizniche/Mycodo/releases/tag/v7.0.3

kizniche, 2019-01-10:
Just pushed a fix for the alembic versions. Master should be good to go if that's what you're trying to use.

SAM26K, 2019-01-11:
Excellent! Installs fine now.
Thanks again.

SAM26K, 2019-01-11:
Still having problems with slave RPi upgrading to current version.
Starting from scratch with a fresh Raspbian image.  Biggest pain is
recopying all those container ids for the master rpi.

kizniche, 2019-01-11:
You can try restoring a release 7.0.4 or a prior 7.x release version and then upgrade to 7.0.5

kizniche, 2019-01-11:
Basically and non-master 7.0.0 <= version <= 7.0.4 release version