kizniche/Mycodo #600: Set up edge function for pH sensor not working (same for T1000 sensor)

alonrab, 2019-01-10:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.04
 
#### Problem Description
Please list: 
I'm trying to set up a on off output trigger of GPIO pin for pH reading. 
I have an atlas pH sensor and on/off power relay
[image]
[image]

 
### Errors

When I'm trying to set up an edge, no data input option is available

[image]

Also, have to fill all the inputs  to delete the function

  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:

How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

alonrab, 2019-01-10:
Thank you for looking into it!
Did steps 1 -6 : 
this is the data input-
[image]

Then step 1+2
[image]
 
In the option box, selected: "measurement" 
in lower option box, selected : "output duration"

otherwise an error note show to select for lower box pops. 

Then selected pH probe and saved
[image]

added action, selected GPIO and saved

[image]

When trying to save- need to change box #1 #2 at bottom left
[image]

when press save, error say 
[image]

Any suggestions? 
Many many thanks

kizniche, 2019-01-10:
This is a bug that's been fixed in the yet-to-be-released v7.0.5. For now, just select anything in those boxes (it won't matter what you select).

As for your error, it appears you did not replace the ID with your particular measurement ID. Looking at your screenshot, your measurement ID is "b927459a".

alonrab, 2019-01-10:
I did,
This is the text input I placed:
if {fcc15238} is not None:
    if {fcc15238} is < 9:
        print(1)
else:
    print(0)  # False (Do not trigger actions)

Then:
If I press activate, it show no error but wont save the change
If I press save, then it show error.

[image]

[image]

Any suggestions?
Many thanks,

kizniche, 2019-01-10:
I just updated the manual with a [Conditional Setup Guide](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#conditional-setup-guide). Let me know if it helps and if there is anything that can be done to improve it.

alonrab, 2019-01-11:
Thanks a lot! its working great.
this is what confused me:
when looking for the ID - used the sensor ID instead of ID assigned at function conditional window
Then,  input from examples:
if None not in [{bdc127e}]: 
    if {0bdc127e} > 6 :
        print(1)
worked.

closed 2019-01-20 by kizniche