kizniche/Mycodo #602: Error 500 after apdating to 7.0.5

HugoPenner, 2019-01-11:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.0.5
 
#### Problem Description
Please list:

  - after updating to 7.0.5 form 7.0.3 i got this error 

 
### Errors

 Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 509, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: input.ftdi_location

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 458, in page_dashboard
    input_dev = Input.query.all()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2843, in all
    return list(self)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2995, in __iter__
    return self._execute_and_instances(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 3018, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 948, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/sql/elements.py", line 269, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1060, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1200, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1413, in _handle_dbapi_exception
    exc_info
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 265, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 248, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 509, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: input.ftdi_location [SQL: 'SELECT input.id AS input_id, input.unique_id AS input_unique_id, input.name AS input_name, input.is_activated AS input_is_activated, input.is_preset AS input_is_preset, input.preset_name AS input_preset_name, input.device AS input_device, input.interface AS input_interface, input.period AS input_period, input.power_output_id AS input_power_output_id, input.resolution AS input_resolution, input.resolution_2 AS input_resolution_2, input.sensitivity AS input_sensitivity, input.thermocouple_type AS input_thermocouple_type, input.ref_ohm AS input_ref_ohm, input.calibrate_sensor_measure AS input_calibrate_sensor_measure, input.location AS input_location, input.gpio_location AS input_gpio_location, input.i2c_location AS input_i2c_location, input.i2c_bus AS input_i2c_bus, input.ftdi_location AS input_ftdi_location, input.uart_location AS input_uart_location, input.baud_rate AS input_baud_rate, input.pin_clock AS input_pin_clock, input.pin_cs AS input_pin_cs, input.pin_mosi AS input_pin_mosi, input.pin_miso AS input_pin_miso, input.bt_adapter AS input_bt_adapter, input.switch_edge AS input_switch_edge, input.switch_bouncetime AS input_switch_bouncetime, input.switch_reset_period AS input_switch_reset_period, input.pre_output_id AS input_pre_output_id, input.pre_output_duration AS input_pre_output_duration, input.pre_output_during_measure AS input_pre_output_during_measure, input.sht_voltage AS input_sht_voltage, input.adc_gain AS input_adc_gain, input.adc_resolution AS input_adc_resolution, input.adc_sample_speed AS input_adc_sample_speed, input.cmd_command AS input_cmd_command, input.weighting AS input_weighting, input.rpm_pulses_per_rev AS input_rpm_pulses_per_rev, input.sample_time AS input_sample_time, input.port AS input_port, input.times_check AS input_times_check, input.deadline AS input_deadline, input.custom_options AS input_custom_options \nFROM input'] (Background on this error at: http://sqlalche.me/e/e3q8)

kizniche, 2019-01-11:
It looks like you have a database version issue. Check the System Information page and see if your "Database Version" is colored red, meaning there's a database version mismatch and you're using an old (non-upgraded) database with the current version. This can occur from upgrading to master at an inopportune time or an error in the database upgrade process during a normal release upgrade. I'd recommend removing your database file (```mv ~/Mycodo/databases/mycodo.db ~/Mycodo/databases/mycodo.db.old```) and reloading the frontend to generate a new database.

kizniche, 2019-01-11:
You can also try restoring an older v7.x version of Mycodo and then upgrade to 7.0.5 from that version.

HugoPenner, 2019-01-11:
With a new data base it works thx but is there a way to fix the old one cause now i have to do all my settings again

kizniche, 2019-01-11:
See my second message.

kizniche, 2019-01-11:
If you're familiar with SQLite, you can modify the database yourself and add the ftdi_device column to the inputs table and make sure the alembic version is correct.

closed 2019-01-20 by kizniche