kizniche/Mycodo #603: PID controller acting up, turns on for a few seconds then turns off

tomkm7p, 2019-01-12:
## Mycodo Issue Report:
  - Specific Mycodo Version:6.4.5
 
#### Problem Description
Please list:

  - what were you trying to do - set up a PID for a heat lamp, 
  - specific setup details that are involved

 Tried to set up a PID to maintain a constant temp of 27*C using a DHT22 sensor . All relays and sensors work.  Attached a a small light instead of heat lamp to test the system.  The light/ relay turns on for exactly 10 seconds in the on position and then turns off for exactly 20 seconds repeatedly. 

Details:

Direction - raise
Period- 30
Max age - 120
Setpoint - 27
Band - 1
Store lower as negative - checked
Kp Gain - 1.0
Ki/KD Gain - 0
Integrator Max -  100
Output - GPIO 6
Min On Raise - 0
Max on Duration  Raise - 0
Min Off Duration Raise - 0
Output Lower - Disabled
Setpoint Tracking - None

### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue? 
I am new to this world of Pi and very grateful and thankful to the open source community.  However, for a Noob such as myself, it is quite difficult to understand certain concepts. One such a concept is setting up a PID controller.  I have a heat lamp and I have a DHT 22.  All I want is to maintain a constant temp of 27*C.  I do not fully grasp all the PID controller settings.  In theory, all i need is 2 options on the PID contoller --- required temp *C ---- and -/+ *C.  Why is my system behaving so erratically? My required temp is set to 27*C and my -/+ *C is 1.  I can see that this program is an advanced piece of software with many options and i hope to learn it well and use it for my project, but I dont know why my system behaves so. 
Thanks

kizniche, 2019-01-12:
With a p gain of 1.0, if your duty cycle is 33% (10 sec on, 20 sec off), that would mean your temperature is 17C? If so, the PID appears to be operating correctly.

Have you read the manual? Also, there is a specific [FAQ](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#frequently-asked-questions) titled "How do I set up simple regulation?" and the section, [Quick Setup Examples](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#quick-setup-examples)?

kizniche, 2019-01-18:
Additionally, versions prior to 7.0 (<7.0) are not supported any longer. If you find an issue with any 7.x version, feel free to start a new issue, detailing the issue and including information about the version you're using.

closed 2019-01-18 by kizniche