kizniche/Mycodo #604: Getting × Error: Error: SQL: 'activated' When data control is activated.

SAM26K, 2019-01-12:
## Mycodo Issue Report:
  - Specific Mycodo Version:7.05
Installed from scratch.  Not an upgrade.  Happened after two complete re-installs.
 
#### Problem Description
Getting × Error: Error: SQL: 'activated' When data control is activated. Happened with GPIO status input and CPU/GPU temp input. 
[image]

Daemon Log:
Last 60 lines of /var/log/mycodo/mycodo.log:
2019-01-11 18:52:40,331 - mycodo.daemon - INFO - Mycodo daemon started in 2.428 seconds
2019-01-11 18:52:40,339 - mycodo.daemon - INFO - 38.53 MB RAM in use
2019-01-11 19:04:46,284 - mycodo.output - INFO - Output 4128aad4 setup on pin 5 and turned OFF (OFF=HIGH)
2019-01-11 19:05:23,414 - mycodo.output - INFO - Output db4a80dd setup on pin 6 and turned OFF (OFF=HIGH)
2019-01-11 19:06:24,296 - mycodo.output - INFO - Output 795d6db5 setup on pin 13 and turned OFF (OFF=HIGH)
2019-01-11 19:07:33,392 - mycodo.output - INFO - Output 8771fca4 setup on pin 13 and turned OFF (OFF=HIGH)
2019-01-11 19:08:26,703 - mycodo.output - INFO - Output 8771fca4 setup on pin 19 and turned OFF (OFF=HIGH)
2019-01-11 19:11:59,874 - mycodo.output - INFO - Output cf299f72 setup on pin 26 and turned OFF (OFF=HIGH)
2019-01-11 19:12:58,604 - mycodo.output - INFO - Output f6c32698 setup on pin 16 and turned OFF (OFF=HIGH)
2019-01-11 19:13:46,319 - mycodo.output - INFO - Output 57da3a52 setup on pin 20 and turned OFF (OFF=HIGH)
2019-01-11 19:13:58,083 - mycodo.output - ERROR - Cannot turn off Output f7809276-f6c3-4173-8ca4-14916b77ac87: Output not set up properly.
2019-01-11 19:14:37,377 - mycodo.output - INFO - Output f7809276 setup on pin 21 and turned OFF (OFF=HIGH)
2019-01-11 19:30:13,773 - mycodo.output - WARNING - Cannot turn on Output with ID 7955bbba-6ef1-44cb-b202-1726bfdff859. It doesn't exist
2019-01-11 19:30:24,002 - mycodo.output - WARNING - Cannot turn off Output with ID 7955bbba-6ef1-44cb-b202-1726bfdff859. It doesn't exist
2019-01-11 19:30:26,529 - mycodo.output - WARNING - Cannot turn off Output with ID 7955bbba-6ef1-44cb-b202-1726bfdff859. It doesn't exist
2019-01-11 19:33:50,565 - mycodo.output - WARNING - Cannot turn on Output with ID 7955bbba-6ef1-44cb-b202-1726bfdff859. It doesn't exist
2019-01-11 19:34:01,897 - mycodo.output - WARNING - Cannot turn off Output with ID 7955bbba-6ef1-44cb-b202-1726bfdff859. It doesn't exist
2019-01-11 19:34:42,132 - mycodo.output - WARNING - Cannot turn on Output with ID 7955bbba-6ef1-44cb-b202-1726bfdff859. It doesn't exist
2019-01-11 19:35:39,744 - mycodo.output - WARNING - Cannot turn off Output with ID 7955bbba-6ef1-44cb-b202-1726bfdff859. It doesn't exist
2019-01-11 19:35:47,999 - mycodo.output - WARNING - Cannot turn on Output with ID 7955bbba-6ef1-44cb-b202-1726bfdff859. It doesn't exist
2019-01-11 19:38:07,974 - mycodo.output - WARNING - Cannot turn off Output with ID 7955bbba-6ef1-44cb-b202-1726bfdff859. It doesn't exist
2019-01-11 19:45:48,045 - mycodo.output - WARNING - Cannot turn on Output with ID 14a83075-1f37-44a9-97e0-e7a55c3cf74f. It doesn't exist
2019-01-11 19:45:55,547 - mycodo.output - WARNING - Cannot turn off Output with ID 14a83075-1f37-44a9-97e0-e7a55c3cf74f. It doesn't exist
2019-01-11 19:31:14,784 - mycodo.daemon - INFO - Mycodo daemon v7.0.5 starting
2019-01-11 19:31:16,541 - mycodo.daemon - INFO - Anonymous statistics enabled
2019-01-11 19:31:16,630 - mycodo.daemon - INFO - Starting rpyc server
2019-01-11 19:31:17,509 - mycodo.output - INFO - Output 4128aad4 setup on pin 5 and turned OFF (OFF=HIGH)
2019-01-11 19:31:17,511 - mycodo.output - INFO - Output db4a80dd setup on pin 6 and turned OFF (OFF=HIGH)
2019-01-11 19:31:17,513 - mycodo.output - INFO - Output 795d6db5 setup on pin 13 and turned OFF (OFF=HIGH)
2019-01-11 19:31:17,515 - mycodo.output - INFO - Output 8771fca4 setup on pin 19 and turned OFF (OFF=HIGH)
2019-01-11 19:31:17,517 - mycodo.output - INFO - Output cf299f72 setup on pin 26 and turned OFF (OFF=HIGH)
2019-01-11 19:31:17,519 - mycodo.output - INFO - Output f6c32698 setup on pin 16 and turned OFF (OFF=HIGH)
2019-01-11 19:31:17,521 - mycodo.output - INFO - Output 57da3a52 setup on pin 20 and turned OFF (OFF=HIGH)
2019-01-11 19:31:17,523 - mycodo.output - INFO - Output f7809276 setup on pin 21 and turned OFF (OFF=HIGH)
2019-01-11 19:31:17,527 - mycodo.output - INFO - Output controller activated in 283.4 ms
2019-01-11 19:31:18,030 - mycodo.daemon - INFO - All activated Conditional controllers started
2019-01-11 19:31:18,031 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-01-11 19:31:18,032 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-11 19:31:18,032 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-11 19:31:18,033 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-11 19:31:18,034 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-11 19:31:18,535 - mycodo.daemon - INFO - Mycodo daemon started in 3.745 seconds
2019-01-11 19:31:18,546 - mycodo.daemon - INFO - 38.72 MB RAM in use
2019-01-11 20:04:10,790 - mycodo.output - WARNING - Cannot turn on Output with ID 1d5cd727-80c6-403c-91ed-21ab462592e8. It doesn't exist
2019-01-11 20:04:23,195 - mycodo.output - WARNING - Cannot turn off Output with ID 1d5cd727-80c6-403c-91ed-21ab462592e8. It doesn't exist
2019-01-11 20:11:44,002 - mycodo.output - WARNING - Cannot turn on Output with ID 3281f587-e582-4e19-9056-2bd2fc003afa. It doesn't exist
2019-01-11 20:11:58,283 - mycodo.output - WARNING - Cannot turn off Output with ID 3281f587-e582-4e19-9056-2bd2fc003afa. It doesn't exist
2019-01-11 20:12:44,657 - mycodo.output - WARNING - Cannot turn on Output with ID aff9e761-68b5-4c08-b609-415ec2fcf849. It doesn't exist
2019-01-11 20:12:55,828 - mycodo.output - WARNING - Cannot turn off Output with ID 3281f587-e582-4e19-9056-2bd2fc003afa. It doesn't exist
2019-01-11 20:12:58,425 - mycodo.output - WARNING - Cannot turn off Output with ID 3281f587-e582-4e19-9056-2bd2fc003afa. It doesn't exist
2019-01-11 20:13:51,837 - mycodo.output - WARNING - Cannot turn off Output with ID aff9e761-68b5-4c08-b609-415ec2fcf849. It doesn't exist
2019-01-11 20:16:23,441 - mycodo.output - WARNING - Cannot turn on Output with ID aff9e761-68b5-4c08-b609-415ec2fcf849. It doesn't exist
2019-01-11 20:16:31,057 - mycodo.output - WARNING - Cannot turn off Output with ID aff9e761-68b5-4c08-b609-415ec2fcf849. It doesn't exist
2019-01-11 20:17:55,054 - mycodo.output - WARNING - Cannot turn on Output with ID daeca18b-2c31-4d15-bbbc-9438c46654aa. It doesn't exist
2019-01-11 20:18:07,851 - mycodo.output - WARNING - Cannot turn off Output with ID daeca18b-2c31-4d15-bbbc-9438c46654aa. It doesn't exist
2019-01-11 20:20:44,021 - mycodo.output - WARNING - Cannot turn off Output with ID daeca18b-2c31-4d15-bbbc-9438c46654aa. It doesn't exist
2019-01-11 22:00:01,226 - mycodo.input_c2dd0a27 - INFO - Activated in 870.5 ms
2019-01-11 22:00:01,508 - mycodo.input_c2dd0a27 - INFO - Activated in 867.1 ms
2019-01-11 22:00:12,413 - mycodo.input_118e265d - INFO - Activated in 786.5 ms
2019-01-11 22:00:13,274 - mycodo.input_118e265d - INFO - Activated in 1423.7 ms

SAM26K, 2019-01-12:
Dashboard gauge for CPU temp reads zero even though live page has a real reading.

kizniche, 2019-01-12:
> Getting × Error: Error: SQL: 'activated' When data control is activated. 

Is there actually an issue or is there only a message?

SAM26K, 2019-01-12:
So far just the message.

SAM26K, 2019-01-12:
Looks database related and worried it will mess things up later,

SAM26K, 2019-01-13:
Update: Seeing this on master pi now too.  That pi was updated through sequential upgrades from 7.0.3 to 7.05.   Not sure if related but also notice that controls that have been deleted previously still appear in the selection box's for dashboard controls.
[image]

kizniche, 2019-01-13:
Master is 7.0.5 right now. Can you provide steps to reproduce your issue? I haven't been able to replicate it.

SAM26K, 2019-01-13:
So if I force an update to 7.0.5 master I should not be seeing the problem? Or do I have to another complete install of 7.0.5 and then upgrade to master?  BTW is there any databases that are not in the Mycodo directory I should be deleting?

kizniche, 2019-01-13:
7.0.5 is the same as master right now. Upgrading to 7.0.5 will bring the same code as upgrading to master. Is there actually a problem other than it displaying an error?

kizniche, 2019-01-13:
I'm not sure what you're referring to because you're describing multiple issues in this issue. It would be better to only discuss one problem per issue, unless it's related.

SAM26K, 2019-01-13:
No other error that I have encountered so far.  Only other fluke is that 433Mhz input that I deleted before.  What about paths outside of Mycodo for databases?

SAM26K, 2019-01-13:
I'm just afraid that fixing the sql error message will have to be resolved by deleting the install again and think its possible that a database not in the Mycodo folder may be causing the issue.

SAM26K, 2019-01-13:
As far as duplicating the steps: Master pi was upgraded fron 7.0.3 to 7.05.
Slave pi, had an old 5.n rev that I had to upgrade straight to 7.
I update both since then with scratch installs that were released version at the time (V7+).
I deleted the Mycodo folder  before scratch installs but suspect a db file may not getting deleted .

kizniche, 2019-01-13:
Steps as in:

1. click this.
2. click that.
3. Error appears "blah" or "This" occurs when it should be "That"

closed 2019-01-13 by kizniche

SAM26K, 2019-01-14:
Excellent!  No more error notes after editing python file.
Thanks very much again.

commit 08d5916c (2019-01-13): Fix message display (closes #604, fix doesn't affect functionality)