kizniche/Mycodo #606: Feature Request Add dashboard led indicator

SAM26K, 2019-01-15:
## Mycodo Feature Request
  - Specific Mycodo Version: 7.0.5
 
### Request Description
Add small desktop LED indicator with Linux command options.  Needs a data input and dashboard display device. Should be small so a lot can populate the dashboard without occupying a large amount of the screen.  Overlay option to put on top of other dashboard items would be nice if possible.  Also nice would be a disable check box on other dash controls status shading.
### Reason
Need to get boolean status from various remote wifi devices and other rpi's on a local LAN.
Currently the shading of dashboard controls seem to reflect the status of the Mycodo daemon's device status  on the local rpi and does not always reflect the state of the actual remote device.  Using Linux script, the remote device can be interrogated directly. 
Thanks.

SAM26K, 2019-01-15:
Example bash file of getting relay status from Sonoff TH16 with Tasmota firmware:
```
#!/bin/bash
#Relay Status for Tasmota Sonoff TH16
out=$(curl -s http://192.168.1.36/cm?cmnd=Power1)
#echo $out
if [[ $out == *"ON"* ]]; then
  echo "1"
else
 if [[ $out == *"OFF"* ]]; then
  echo "0"
 fi
fi

```
Returns 0 for relay off 1 for relay on..

Example code for getting GPIO status from a remote rpi running Mycodo:
```
#!/bin/bash
out=$(sshpass -p password ssh mycodo@192.168.1.51 mycodo-client --output_state 4128aad4-56ae-4dc0$
#echo $out
if [[ $out == *"Server returned: on"* ]]; then
  echo "1"
else
 if [[ $out == *"Server returned: off"* ]]; then
  echo "0"
 fi
fi

Example code for getting GPIO status from a remote rpi running Mycodo:
```
#!/bin/bash
out=$(sshpass -p password ssh mycodo@192.168.1.51 mycodo-client --output_state 4128aad4-56ae-4dc0$
#echo $out
if [[ $out == *"Server returned: on"* ]]; then
  echo "1"
else
 if [[ $out == *"Server returned: off"* ]]; then
  echo "0"
 fi
fi
```

SAM26K, 2019-01-15:
Example of why Linux command input is needed to check status of Sonoff TH16 or other remote device.  Need just an LED for dash panel so it doesn't take up so much real estate.  Need the code ability as different remote devices will have different requirements to return status.
[image]

SAM26K, 2019-01-28:
> Will be included in the v7.1.5 release.

Excellent!  Thank you very mu

closed 2019-01-29 by kizniche

SAM26K, 2019-01-30:
New dashboard widget working well, Thanks very much!
Related question: Is there any way to speed up the refresh time of linux/python user controls? Sometimes it takes > 30 seconds for the indicator to respond to a change in state of the remote pi
If I use a terminal window to send the command to the remote pi, I get an accurate real time response. But if the command is embedded in a data input or functional control it can take up to 60 seconds to respond.
Not sure this can be improved so wanted to get your opinion before posting another issue/request.
Thanks Again!

kizniche, 2019-01-30:
There shouldn't be any delay in the Mycodo backend/frontend. This is most likely due to your configuration. Can you attach screenshots of all the relevant options for how you have your setup (indicator options, input options, output options, etc.)?

SAM26K, 2019-01-31:
Will get back to you on this.  For now I have a work around that's working really well.  ie all code in a two  linux  bash files. 
On control with the following sequence:
1. Send command to turn on a drip line solenoid.
2. Confirm the solenoid is turned on/open drip line.
3. Turn on water pump.
Off control:
1. Send command to turn off water pump.
2. Confirm pump is off.
3. Turn off solenoid.
Needs error handling code but working well with not much delay between events.
Thanks again!