kizniche/Mycodo #607: Add HomeLap-pH Hardware

pmunz75, 2019-01-15:
Hello Kily ,
** HomeLap-pH ** ---> 
Hardware Link : http://homelab.link/homelab-ph-rev-3.shtml
Software Link  : http://homelab.link/software.shtml

I have already done hardware and at home, for Testing

is it possible to integrate this pH module?
Price performance is very good! ( 30 -35 $ ) Isolate Module

It works perfectly on the Raspberry and Arduino with the supplied software.
Sensor values are very accurate.

[image]

[image]

I think with this module you would help some.
pH measurements are very expensive and demanding.
Thanks for your feedback.

kizniche, 2019-01-16:
Their software seems rather integrated into their UI/Apache interface. I was just looking at their bash code and to acquire a measurement from get_pH.sh, many other files are referenced. It would be nice if the measurement acquisition code could be contained within one file.

kizniche, 2019-01-16:
Obtaining a measurement in Python would be ideal. I see in get_adc01.sh the addresses for I2C communication are defined.

pmunz75, 2019-01-16:
Reading the module from Python

do you mean this ?

It is possible to read HomeLab-pH data also using Python. The script has to execute the above shell 

#!/usr/bin/python

import os, json

# execute the shell command and assign the result
data = os.popen('/var/www/homelab/cgi-bin/get_pH.sh -json').read()

# just to see the result
print (data)

# convert the json string
d = json.loads(data)

# see the pH value
print (d['pH'])

# see the temperature
print (d['t'])

kizniche, 2019-01-16:
No, that merely executes the shell script... The one I mentioned that relies on many other shell scripts.

kizniche, 2019-01-16:
If you extracted the shell scripts yourself, you can always add the Linux Command Input and have it acquire a measurement by executing the file containing the python code you just pasted:

```python
#!/usr/bin/python
import os, json

data = os.popen('/path/to/scripts/get_pH.sh -json').read()
print (data)
d = json.loads(data)
print (d['pH'])
```

kizniche, 2019-01-16:
[image]

kizniche, 2019-01-16:
I just think this would be difficult to integrate into Mycodo because there's no Python module and the shell scripts are distributed among many files. The Linux Command Input is a good solution and was partly designed for issues like this.

kizniche, 2019-01-16:
The temperature sensor is a 1-wire sensor. I suspect it may be a DS18B20, but have yet to find an exact naming of the sensor model. If it is a DS sensor, you may be able to add one of those specific Input modules for it to acquire measurements from it. I'll report back if I find exactly what sensor it is.

SAM26K, 2019-01-17:
Also, how's the linearity between ph 4-7?  The oem code seems to be single point but could have more cal pts for better accuracy with more code if needed.

SAM26K, 2019-01-17:
Does that kit come with any probes or is it a separate purchase?

kizniche, 2019-01-17:
I just got confirmation from the developer that the temperature sensor is a DS18B20, so you should be able to use the DS18B20 Input module instead of querying the shell script.

kizniche, 2019-01-27:
create a script with the contents from [this comment](https://github.com/kizniche/Mycodo/issues/607#issuecomment-454967944), then in the Linux Command input, the command should be ```~/path/to/the/script.sh```

pmunz75, 2019-01-27:
ohh thanks, the script user and group is mycodo ?

kizniche, 2019-01-27:
it doesn't matter the user/group of the script.

pmunz75, 2019-01-27:
Hello,
the shell value:
pi@raspberrypi:~/Mycodo/mycodo/scripts$ ./ph_messung.sh 
{"taskStatus":"ok","time":1548632469.608,"v":22.9,"t":24.4,"pH":6.75,"calibID":"3b2f061a5b5f0b70c6a4fcd1acd22cc9","calibDate":"2019-01-28"}

24.4
pi@raspberrypi:~/Mycodo/mycodo/scripts$ 

mycode no value , the daemon.log

2019-01-27 23:42:37,502 - mycodo.linux_command_ecc8764e - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.

kizniche, 2019-01-27:
It looks like your command returns this

```
{"taskStatus":"ok","time":1548632469.608,"v":22.9,"t":24.4,"pH":6.75,"calibID":"3b2f061a5b5f0b70c6a4fcd1acd22cc9","calibDate":"2019-01-28"}

24.4
```

when it should return this

```
24
```

kizniche, 2019-01-27:
Ah, remove the first ```print()``` from the script:

```python
#!/usr/bin/python
import os, json

data = os.popen('/path/to/scripts/get_pH.sh -json').read()
d = json.loads(data)
print (d['pH'])
```

kizniche, 2019-01-27:
also, it looks like you're returning temperature instead of pH. Change your code to match what I wrote.

closed 2019-06-07 by kizniche