kizniche/Mycodo #608: BMP280 sensor reporting higher than expected temperature

wflost, 2019-01-15:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 Mycodo Version: 6.4.7
Python Version: 3.5.3 (default, Jan 19 2017, 14:11:04) [GCC 6.3.0 20170124]
Database Version: 15bd5d08a4f0
Daemon RAM Usage: 46.104 MB
Daemon Virtualenv: Yes
Frontend RAM Usage: 48.66 MB
Frontend Virtualenv: Yes

#### Problem Description
Please list:
Temperature for the BMP sensor is always 2 degrees warmer than the AM2315 sensor and another digital thermometer that I have in the same location. 

I have found the following note https://github.com/letscontrolit/ESPEasy/issues/164 where it describes this issue of the BMP280 sensor reporting higher temperatures
 
Screen dump
####
[image]

Red is SHT sensor please ignore
Blue is AM2315 Sensor and I get a similar temperature from a standalone digital thermometer
Green is the BMP280 which is higher than expected, I would expect to see it close to the blue line

####

### Errors
  - None

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Installed BMP280 and monitor the sensor
 
 
### Additional Notes
Thanks Kyle

wflost, 2019-01-15:
The update interval for all my sensors are 120 sec

kizniche, 2019-01-16:
How can this be applied to the python module?

wflost, 2019-01-16:
Hi Kyle,

I am not sure, I am not familiar enough with Mycodo or the sensor to answer 

The only solutions I have been able to think of is to assume there is a 2 degree difference and build that into all my settings

Happy to close if this is not possible in Python

Thanks

kizniche, 2019-01-16:
I'm sure it's possibile, but I'm not sure how to actualize it.

Theoi-Meteoroi, 2019-03-07:
This shouldn't be a surprise. There is a very distinct difference between accuracy and precision.   The readings are precise.  What is accurate?  Well, you need a trusted standard measurement device to have accurate.  Measurement is always a comparison between the subject signal and a standard, if you are concerned about error.  And not all signals are linear. The devices above appear to track each other.  There are self-heating issues and other error sources that factor into a system.  I have purchased parts from asia that show offsets like this.  If I were to bet, the AM2315 would be my choice for most likely accurate.

You shouldn't expect Mycodo to tell you anything but the actual reading, IMHO.

wflost, 2019-03-11:
Hi Theoi-Meteoroi,

Agree, Mycodo should provide the data from the sensor, in the link above there is a reference to changing the mode of the sensor to reduce the over heating effect and provide a data more similar to the AM2315. I am not familiar enough with Python or Mycodo code to provide more assistance with the coding and implementation of the mode selection

kizniche, 2019-08-17:
I made several attempts to change this setting with the code I use in Mycodo's module without luck.

However, I did just find from the issue in the original post, [this comment](https://github.com/pimoroni/bmp280-python/issues/2#issuecomment-487531382) that references a potential solution with the [bmp280-python module](https://github.com/pimoroni/bmp280-python). Could you give this a shot and see if it fixes your issue?

kizniche, 2019-08-17:
It looks like their [PR #3](https://github.com/pimoroni/bmp280-python/pull/3) added this feature, called "forced" mode, which is detailed in [this example file](https://github.com/pimoroni/bmp280-python/pull/3/commits/3b4620e3846b8c8f86a22fb8937b7c802885a98a#diff-da45f9ef701c6dcce02c6044027ca717). This PR hasn't been merged, so it's not available. Hmm.

closed 2020-02-26 by kizniche

kizniche, 2020-07-25:
@wflost I stumbled upon this issue from another one and went back to the PR I linked previously and saw the forced mode edit has been applied to the library. So, I added a new BMP280 Input that includes this library and the ability to select whether or not to set forced mode. I'm not sure if you still use this sensor, but figured I should add it anyway. Cheers.

wflost, 2020-07-26:
Thanks Kyle, I haven't got it connected anymore, let me see if I still have
it
Thanks