kizniche/Mycodo #609: error generated with conditional testing boolean linux input

SAM26K, 2019-01-18:
## Mycodo Issue Report:
  - Mycodo Version: 7.0.5 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
Database Version: 0797d251d77d
Daemon Status: Running 
Daemon Process ID: 848 
Daemon RAM Usage: 109.584 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 58.52 MB 
Frontend Virtualenv: Yes
 
#### Problem Description
Throws error when testing function conditional:comditional.
### Errors
[image]
[image]

```
Bash code used for TH-1 status test (works)
#!/bin/bash
#Relay Status for Tasmota Sonoff TH16
out=$(curl -s http://192.168.1.36/cm?cmnd=Power1)
#echo $out
if [[ $out == *"ON"* ]]; then
   echo "1"
else
 if [[ $out == *"OFF"* ]]; then
   echo "0"
 fi
fi
```
The linux code data input switches a front panel ok but doesn't seem to work in a function conditional:conditional test.  Deleting the boolean test is the only thing that works.  Is there something wrong with the code in the conditional control?
Thanks.
[image]

kizniche, 2019-01-18:
I found this issue last night when testing #605. This issue arises when more than one Conditions are used in the Conditional Statement. The last commit to master (303caf5) fixed this issue, if you want to make a manual fix.

SAM26K, 2019-01-18:
It looks like it's working, great work! . Will need a few hours to let the conditionals trigger a few times.

On the new conditional:conditional improvements are you looking at selectable multiple actions?
i.e.  I'm putting together a drip irrigation system where I want to:
1. Turn on one or more drip line solenoids (max 8 per raspberry pi).
2,  Pause for x seconds.
3.  Check status of solenoid (open or closed).
4.  If solenoid is open then turn on water pump for x seconds and exit loop.
5  If solenoid is closed go back to step 1 .  Repeat  step 1-4 for x number of times.
6. If solenoid never opened, log event as an error.

Want to make sure at least  1 solenoid is open before turning on pump to avoid deadheading  pump. 
Thanks very much for functionality so far.

kizniche, 2019-01-18:
This is out of the scope of a conditional controller. It's best you create your own script to perform these checks. Mycodo cound be used to acquire measurements, periodically call the script, and operate GPIO pin states.

kizniche, 2019-01-18:
Actually, what you describe can possibly be done with a conditional... I'll have to add a new function "get_measurement()" that can be called from within the Conditional Statement to return the latest measurement. That would allow time.sleep(x) calls to be performed to permit new measurements to be acquired. Currently all {Measurement ID}s are replaced with the actual measurement prior to code execution, but incorporating functions allows real-time measurement acquisition as the code runs. Thanks for the idea.

kizniche, 2019-01-18:
I can formulate some test code to perform your steps, above, with the new function once I implement it. I'm about to start some work polishing the new conditional functionality I started yesterday and finish some testing.

SAM26K, 2019-01-18:
Thanks very much for your time working on that.  Look forward to testing.

kizniche, 2019-01-19:
Check out [this comment](https://github.com/kizniche/Mycodo/issues/605#issuecomment-455733562) and the revised [Conditionals section of the manual](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#conditional) (and examples there) for info on the new functionality.

SAM26K, 2019-01-19:
> For your particular interest:
>
>    1. Turn on one or more drip line solenoids (max 8 per raspberry pi).
>    2. Pause for x seconds.
>    3. Check status of solenoid (open or closed).
>    4. If solenoid is open then turn on water pump for x seconds and exit
>    loop.
>    5 If solenoid is closed go back to step 1 . Repeat step 1-4 for x
>    number of times.
>    5. If solenoid never opened, log event as an error.
>
> This is how I would achieve it:
>
> Condition:
>
>    1. Check the state of the solenoid somehow (ID=asdf1234)
>
> Actions:
>
>    1. Output: Opens the solenoid for the drip line for 140 seconds
>    (ID=zxcv4567)
>    2. Output: Turns on a pump for 120 seconds (ID=qwer9876)
>    3. Email Notification (ID=poiu1029)
>
> Conditional Statement:
>
> run_action("{zxcv4567}")  # Open drip line for 140 seconds
> solenoid_opened = False  # Used to determine if solenoid ever opened
> for range(3):  # Loop 3 times
>     check_drip = measure("{asdf1234}")
>     if check_drip is not None and bool(check_drip):  # Drip line open
>         solenoid_opened = True  # Store that solenoid did open
>         run_action("{qwer9876}")  # Turn on water pump
>         break  # The solenoid opened and the pump turned on, so leave the for loop
>     time.sleep(2)  # check_drip was None or not open, wait 2 seconds
> if not solenoid_opened:
>     run_action("{poiu1029}")  # Solenoid never opened, send email

Looks great so far.  Unfortunately I didn't realise that upgrading to
master would wipe out my custom script folder.  Restored previous version
of Mycodo and relocating the folder outside the Mycodo path.  Will take
some work to correct all the paths in the various Mycodo controls to get
things working again.  From what I've read so far the new conditional
functionality should do exactly what I'm looking for.
Thanks again!

SAM26K, 2019-01-19:
For now simply copying the custom script folder  to the new version Mycodo folder using VNC seems to work. :) .  Going with that for now.

kizniche, 2019-01-19:
When you say "custom script folder" are you referring to custom input modules you've imported. Is this a bug, or is this a folder you created within the Mycodo directory? I would not recommend storing anything within the Mycodo directory if you don't want it to end up in /var/Mycodo-backup after an upgrade. Best to keep them somewhere in your home directory (~/).

SAM26K, 2019-01-19:
The various python and linux files I wrote for various controls.  Not input modules.  Instead of writing the code into a control directly they are either python  or linux bash files and the controls call those files.  It was a mistake to locate the  files inside the home/Mycodo folder..

SAM26K, 2019-01-19:
Should have known better, my mistake.

SAM26K, 2019-01-19:
For now I was able to just copy the restored user folder back into the new Mycodo folder with VNC and they work.  Shortcut to get   going again.

closed 2019-01-20 by kizniche

SAM26K, 2019-01-20:
Hi Kyle,
Having problems with new conditional functionality.  Control saves ok without errors but has daemon errors as follows:  It never triggers even though the input is >3V.  No email and no update to the print text file.  Seems to fail

 at first line.  Is there something wrong with the python syntax?

Daemon errors:
```
2019-01-20 11:37:05,289 - mycodo.conditional_564933ff - ERROR - Error evaluating conditional statement. Replaced Conditional Statement:

import os
import sys
sys.path.append(os.path.abspath('/var/mycodo-root'))
from mycodo.mycodo_client import DaemonControl
control = DaemonControl()

def measure(condition_id):
    return control.get_condition_measurement(condition_id)

def run_all_actions():
    print(1)

def run_action(action_id):
    control.trigger_action(action_id)

if 4083e620-2edc-4fac-be77-b10b05996716 is not None:
  F = "/home/pi/test22.txt"
  open(F,"a")
  G = print((4083e620-2edc-4fac-be77-b10b05996716))
  print(4083e620-2edc-4fac-be77-b10b05996716, F)
#F = open('/home/pi/test22.txt','w')
#F.write('hello world')
#F.close()
#  F.write("G")
  if 4083e620-2edc-4fac-be77-b10b05996716 > 1.1:
    for _ in range(3):  # Loop 3 times
      run_action("3e89520f-1b6d-4ac1-a3e4-faf23b36474d")  # Turn on solenoid
      check_drip = measure("f07c7234-1076-4ef8-ab1e-38d87b72f170")  # Measure pin state and store
      if check_drip is not  None and bool(check_drip):  # Drip line open, Good!
         solenoid_opened = True  # Store that solenoid did open
         run_action("6af714d6-7374-4863-a58d-fb12107db06f")  # Turn on water pump
         run_action("6d713dd9-fe60-424d-9a06-d00fd46a065c")  #send email.
         break  # The solenoid opened and the pump turned on, so leave the loop
         time.sleep(2)  # No measurement or check_drip was False/0, wait 2 seconds
  if not solenoid_opened:
      run_action("6d713dd9-fe60-424d-9a06-d00fd46a065c")  # Solenoid never opened, send email

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_conditional.py", line 260, in check_conditionals
    exec(cond_statement_replaced, globals())
  File "<string>", line 17
    if 4083e620-2edc-4fac-be77-b10b05996716 is not None:
                   ^
SyntaxError: invalid syntax

```
This is the conditional code block:
```
`if {4083e620} is not None:
  F = "/home/pi/test22.txt"
  open(F,"a")
  G = print(({4083e620}))
  print({4083e620}, F)
#F = open('/home/pi/test22.txt','w')
#F.write('hello world')
#F.close()
#  F.write("G")
  if {4083e620} > 1.1:
    for _ in range(3):  # Loop 3 times
      run_action("{3e89520f}")  # Turn on solenoid
      check_drip = measure("{f07c7234}")  # Measure pin state and store
      if check_drip is not  None and bool(check_drip):  # Drip line open, Good!
         solenoid_opened = True  # Store that solenoid did open
         run_action("{6af714d6}")  # Turn on water pump
         run_action("{6d713dd9}")  #send email.
         break  # The solenoid opened and the pump turned on, so leave the loop
         time.sleep(2)  # No measurement or check_drip was False/0, wait 2 seconds
  if not solenoid_opened:
      run_action("{6d713dd9}")  # Solenoid never opened, send email`
```
Screen shot of control:
[image]

SAM26K, 2019-01-20:
Action part of control above:
[image]

kizniche, 2019-01-20:
```
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_conditional.py", line 260, in check_conditionals
    exec(cond_statement_replaced, globals())
  File "<string>", line 17
    if 4083e620-2edc-4fac-be77-b10b05996716 is not None:
                   ^
SyntaxError: invalid syntax
```

That is indeed an invalid syntax. Review the [Conditional](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#conditional) section of the manual to understand the changes to measurement acquisition.

kizniche, 2019-01-20:
I'd recommend upgrading to 7.1.0 and attempt to save your Conditional. You are currently on a pre-release commit from master, and I only support issue from the current release.

SAM26K, 2019-01-21:
Read up on the new manual entries and started with your code that had no
errors. Added code with requirement for another parameter (moisture level)
to trigger the sequence of events to start watering and finally got it to
work without errors.  I think I was missing colas, bracket errors and
indentation errors. Not a fan of the python indentation requirement.
New functionality with the conditionals is great though, really needed that
because I'm using multiple wireless devices.  New functionality eliminates
the danger of loosing wifi connection by allowing the ability to
interrogate the remote devices that they have indeed executed the
command(s).

SAM26K, 2019-01-21:
With the interest for anyone reading this thread in the future, this conditional code block worked for me with Mycodo V7.1..0.  It's annoying to read a thread and no working solution at the end:

```
M = measure("{4083e620}")
if M is not None:
 if M > 1.0:
  run_action("{3e89520f}")  # Open drip line for n seconds
  solenoid_opened = False  # Used to determine if solenoid ever opened
  for _ in range(3):  # Loop 3 times
   check_drip = measure("{f07c7234}")  # Measure pin state and store
   if check_drip is not None and bool(check_drip):  # Drip line open, Good!
    time.sleep(5)
    solenoid_opened = True  # Store that solenoid did open
    run_action("{6af714d6}")  # Turn on water pump
    break  # The solenoid opened and the pump turned on, so leave the loop
    time.sleep(2)  # No measurement or check_drip was False/0, wait 2 seconds
```

SAM26K, 2019-01-21:
Thanks very much again, Kyle.

kizniche, 2019-01-21:
Keep in mind when using the measure() function that it pulls the more recent measurement from the measurement database. Therefore, if your input has a Period of, say, 30 seconds, calling measure() will not induce another measurement but merely reference the last measurement that occurred every 30 seconds. So, if you're using time.sleep(), make sure you are setting it to a sufficient time in order to have a new measurement acquired and you're not just pulling the same stale measurement. Now, for the GPIO State Condition, this is measured every time measure() is called, so you can query this function as often as you want and it will pull the real-time state of the GPIO pin. I should probably document this behavior better in the manual.

kizniche, 2019-01-21:
Also, there's an error in your code. You have time.sleep() indented once too many. Recheck my example code. Your time.sleep() is not reachable because it occurs after the break.

kizniche, 2019-01-21:
This is why 1 space indentation is not advised, as it's difficult to tell the indentation level. I recommend 2 or 4 spaces (I use 4).

kizniche, 2019-01-21:
I'm also unsure why you have a 5 second pause.

SAM26K, 2019-01-21:
Added the 5 second delay to visually confirm the state changes. There is a couple of seconds lag across the network I was allowing for.  It seems to be working ok but keeping your suggestions in mind.

kizniche, 2019-01-21:
Back to a computer, I can comment on the fixed code. This also contains executing the action to email that there has been an issue.

```python
M = measure("{4083e620}")
if M is not None and M > 1.0:
  run_action("{3e89520f}")  # Open drip line for n seconds
  solenoid_opened = False  # Used to determine if solenoid ever opened
  for _ in range(3):  # Loop 3 times
    check_drip = measure("{f07c7234}")  # Measure pin state and store
    if check_drip is not None and bool(check_drip):  # Drip line open, Good!
      time.sleep(5)
      solenoid_opened = True  # Store that solenoid did open
      run_action("{6af714d6}")  # Turn on water pump
      break  # The solenoid opened and the pump turned on, so leave the loop
    time.sleep(2)  # No measurement or check_drip was False/0, wait 2 seconds

  if not solenoid_opened:
    run_action("{asdf1234}")  # Notify by email there is an issue opening solenoid
```

SAM26K, 2019-01-22:
Fixed the indent problem, thanks.  The added email function is for testing
only.

There seems to be a problem with deactivation.  The conditional continues
to run even though it's deactivated.  Only restarting the backend after
deactivation seems to stop it.