kizniche/Mycodo #610: 504 Gateway Time-out when saving changes to conditional controls

SAM26K, 2019-01-22:
## Mycodo Issue Report:
Mycodo Version: 7.1.0 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
Database Version: 70c828e05255
Daemon Status: Running 
Daemon Process ID: 852 
Daemon RAM Usage: 96.096 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 69.648 MB 
Frontend Virtualenv: Yes

#### Problem Description
504 Gateway Time-out when saving changes to conditional controls.  The only way to recover is to change the url from https://192.168.1.17/function to https://192.168.1.17 or something other than function.  Happened while editing the action id on line 11 below.  Tried reloading several times and page would not load.  Loading the ip address without /function opens with "live" page and then the functions page can be opened from there.
```
M = measure("{4083e620}")
if M is not None:
 if M > 1.0:
  run_action("{3e89520f}")  # Open drip line for 140 seconds
  solenoid_opened = False  # Used to determine if solenoid ever opened
  for _ in range(3):  # Loop 3 times
   check_drip = measure("{f07c7234}")  # Measure pin state and store
   if check_drip is not None and bool(check_drip):  # Drip line open, Good!
    time.sleep(5)
    solenoid_opened = True  # Store that solenoid did open
    run_action("{d908f290}")  # Turn on water pump
    break  # The solenoid opened and the pump turned on, so leave the loop
    time.sleep(2)  # No measurement or check_drip was False/0, wait 2 seconds

 if not solenoid_opened:
    run_action("{asdf1234}")  # Notify by email there is an issue opening solenoid
```
 
[image]

Daemon log:
2019-01-21 23:15:30,732 - mycodo.daemon - INFO - Mycodo daemon v7.1.0 starting
2019-01-21 23:15:31,724 - mycodo.daemon - INFO - Anonymous statistics enabled
2019-01-21 23:15:31,774 - mycodo.daemon - INFO - Starting rpyc server
2019-01-21 23:15:53,579 - mycodo.output - INFO - Output controller activated in 21004.1 ms
2019-01-21 23:15:54,314 - mycodo.conditional_36aedeb1 - INFO - Activated in 129.3 ms
2019-01-21 23:15:55,006 - mycodo.conditional_f200d074 - INFO - Activated in 290.3 ms
2019-01-21 23:15:55,354 - mycodo.conditional_cfc6377c - INFO - Activated in 204.0 ms
2019-01-21 23:15:55,684 - mycodo.conditional_9acfa451 - INFO - Activated in 155.7 ms
2019-01-21 23:15:55,938 - mycodo.conditional_5f998c4a - INFO - Activated in 156.4 ms
2019-01-21 23:15:56,199 - mycodo.conditional_564933ff - INFO - Activated in 158.4 ms
2019-01-21 23:15:56,609 - mycodo.conditional_a6f92a78 - INFO - Activated in 248.2 ms
2019-01-21 23:15:57,101 - mycodo.conditional_69a7d632 - INFO - Activated in 257.5 ms
2019-01-21 23:15:57,104 - mycodo.daemon - INFO - All activated Conditional controllers started
2019-01-21 23:15:57,108 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-01-21 23:15:58,611 - mycodo.input_64a5f8ec - INFO - Activated in 1198.8 ms
2019-01-21 23:15:59,253 - mycodo.input_65491931 - INFO - Activated in 471.7 ms
2019-01-21 23:16:00,874 - mycodo.input_1ab8f7c4 - INFO - Activated in 1422.8 ms
2019-01-21 23:16:02,322 - mycodo.input_dc446b55 - INFO - Activated in 1056.7 ms
2019-01-21 23:16:03,594 - mycodo.input_5ed0b362 - INFO - Activated in 1085.7 ms
2019-01-21 23:16:04,474 - mycodo.input_defab647 - INFO - Activated in 525.3 ms
2019-01-21 23:16:05,106 - mycodo.input_f8c56ec5 - INFO - Activated in 439.4 ms
2019-01-21 23:16:05,865 - mycodo.input_827843a8 - INFO - Activated in 555.3 ms
2019-01-21 23:16:06,985 - mycodo.input_7e3dc86d - INFO - Activated in 895.9 ms
2019-01-21 23:16:07,771 - mycodo.input_b1ca893c - INFO - Activated in 526.5 ms
2019-01-21 23:16:08,413 - mycodo.input_adff79c3 - INFO - Activated in 444.2 ms
2019-01-21 23:16:09,066 - mycodo.input_07360015 - INFO - Activated in 456.1 ms
2019-01-21 23:16:09,708 - mycodo.input_977ca0cc - INFO - Activated in 442.2 ms
2019-01-21 23:16:10,479 - mycodo.input_f7d399a3 - INFO - Activated in 552.7 ms
2019-01-21 23:16:11,216 - mycodo.input_066f2eb0 - INFO - Activated in 541.4 ms
2019-01-21 23:16:11,937 - mycodo.input_6d345eb7 - INFO - Activated in 526.3 ms
2019-01-21 23:16:11,939 - mycodo.daemon - INFO - All activated Input controllers started
2019-01-21 23:16:11,940 - mycodo.daemon - INFO - All activated Math controllers started
2019-01-21 23:16:11,954 - mycodo.daemon - INFO - All activated PID controllers started
2019-01-21 23:16:11,956 - mycodo.daemon - INFO - All activated LCD controllers started
2019-01-21 23:16:12,459 - mycodo.daemon - INFO - Mycodo daemon started in 41.723 seconds
2019-01-21 23:16:12,467 - mycodo.daemon - INFO - 68.12 MB RAM in use
2019-01-21 23:20:48,407 - mycodo.conditional_36aedeb1 - INFO - Deactivated in 66.1 ms
2019-01-21 23:35:52,146 - mycodo.conditional_36aedeb1 - INFO - Activated in 247.4 ms
2019-01-21 23:36:18,820 - mycodo.conditional_564933ff - INFO - Deactivated in 16.9 ms
2019-01-21 23:36:26,223 - mycodo.conditional_564933ff - INFO - Activated in 220.2 ms
2019-01-21 23:36:38,296 - mycodo.conditional_564933ff - INFO - Deactivated in 57.0 ms
2019-01-21 23:37:30,740 - mycodo.conditional_564933ff - INFO - Activated in 1185.9 ms
2019-01-21 23:37:30,895 - mycodo.conditional_564933ff - INFO - Activated in 1111.5 ms
2019-01-21 23:38:35,855 - mycodo.conditional_564933ff - INFO - Deactivated in 105.0 ms
2019-01-21 23:41:57,441 - mycodo.conditional_a6f92a78 - INFO - Deactivated in 32.8 ms
2019-01-22 00:09:49,894 - mycodo.conditional_a6f92a78 - INFO - Activated in 380.4 ms
2019-01-22 00:09:50,009 - mycodo.conditional_a6f92a78 - INFO - Activated in 297.9 ms

SAM26K, 2019-01-22:
Just noticed that edit to line 11 above was not successful.  Restored previous value which was invalid since I had deleted and recreated the action control.

kizniche, 2019-01-22:
I see the issue. It iterates 100 times with random values. Now that sleep() is at the correct indentation level, it's performing the sleep. Since there are 100 iterations, of a range(3) for loop, this would take 600 seconds to complete, well beyond the server timeout. I'll figure out a fix.

SAM26K, 2019-01-22:
Thanks for looking into that.  Having other problems that I think may be related to shortcutting the time out by opening the "live" page.  eg pump TH16-3 solenoid is not toggling.  Using 5 second delays for troubleshooting.  Maybe code block changes are not fully updating the system somewhere?

kizniche, 2019-01-22:
Open a new issue if you continue to have problems, as this one has been resolved.