kizniche/Mycodo #611: Feature Request Add Import/export for controls (Data. Outputs, Functions, etc)

SAM26K, 2019-01-24:
## Mycodo Feature Request
  - Specific Mycodo Version:
Mycodo Version: 7.1.2 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
Database Version: 70c828e05255
Daemon Status: Running 
Daemon Process ID: 830 
Daemon RAM Usage: 92.796 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 51.272 MB 
Frontend Virtualenv: Yes

Provide means to import and export user created items  especially input, output and function controls.

Reason:
A significant amount of time can be spent creating user controls if a complete re-install is required.  Even if the controls may have incompatibilities, correcting the incompatibilities one by one is better than having to start from scratch if there is a large number of controls involved.

kizniche, 2019-01-24:
The ability already exists to export/import the settings database. The only limitation is the Mycodo version must match the database version, however any version of Mycodo can be downloaded and installed, database imported, then upgraded to the latest version.

SAM26K, 2019-01-24:
Does that include the new conditional features?  Pardon my ignorance but how does one import and export just the controls?

SAM26K, 2019-01-24:
eg can I export controls from one RPi to another one?

kizniche, 2019-01-24:
It exports the settings database, so everything is preserved, even login users. Go to Import/Export and save the settings database. Go to your newly installed Mycodo and import settings database. Viola.

SAM26K, 2019-01-24:
Awesome. verifying now, Thanks very much.

closed 2019-01-24 by kizniche

SAM26K, 2019-01-30:
Thanks again for the new LED front panel indicator.  Exactly what I requested and working as expected.
Related question: Is there any way to speed up the refresh time of linux/python user controls?  Sometimes it takes > 30 seconds for the indicator to respond to a change in state on the remote pi
If I use a terminal window to send the command to the remote   pi, I get an accurate real time response.  But if the command is embedded in a data input or functional control it can take up to 60 seconds to respond. 
Not sure this can be improved so wanted to get your opinion before posting another issue/request.
Thanks Again!