kizniche/Mycodo #613: Atlas i2c Err Nr 121

RynFlutsch, 2019-01-25:
Mycodo 7.0.5
Yesterday i tried atlas PT1000 and EC via i2c:
Sensor reading unseccessful:
Both return ErrNr 121 Remote i/o error.
Ec probe wired with sda, scl 1
and PT1000 with sda, scl 0.
No overlay or multiplexer.

kizniche, 2019-01-25:
Only current versions are supported, so please upgrade to the latest version and if an issue still exists, leave a detailed description of the steps to replicate and any and all errors encountered.

closed 2019-01-29 by kizniche

RynFlutsch, 2019-02-23:
Hey Kyle, i am on 7.3.0 now.
The Error is still present. 
I am using:
EZO Circuits
Isolated EZO Carrier Boards
And the industrial version of pH & EC probes.
There is no problem with the Mycodo Input Modul for industrial probes? If not i will add a wiring diagram and my configuration.

kizniche, 2019-02-23:
Can you provide SSH access and a Mycodo login for me to remotely connect and diagnose the issue?

kizniche, 2019-02-23:
Are you able to acquire measurements with any code?

RynFlutsch, 2019-02-25:
I am not sure, i will check. Maybe you can connect via vpn to my network? 
No, cause iam a wannabe =)
But i can test all things you tell me.

RynFlutsch, 2019-02-25:
[image]

kizniche, 2019-02-26:
What is the output of ```sudo i2c-detect -y 1```?

kizniche, 2019-02-26:
I've had an Atlas scientific PT-1000 connected via I2C for over a year without issues, so I'm sure it works. We just have to figure out what's wrong with your setup.

RynFlutsch, 2019-02-27:
Okay:
[image]

kizniche, 2019-02-27:
Your sensor isn't being detected by Linux. Can you provide photos of how you have this wired with the pins clearly shown in the photos?

kizniche, 2019-02-28:
Are you sure the EZO circuit is in I2C mode? Setting the circuit to use I2C is in the manual on the Atlas Scientific website.

RynFlutsch, 2019-03-01:
Thank you for the input, i'll investigate and provide feedback soon.

OnlinePirateDogFromSpaceAndBeyond2020, 2022-10-28:
Damn you really tried to help this noob and he bounced

OnlinePirateDogFromSpaceAndBeyond2020, 2022-10-28:
I'm getting the exact same issue except mycodo is detecting i2c devices on 3 addresses and im getting eern 121
336 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_100 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:01:54,336 - ERROR - mycodo.controllers.controller_input_8ecdce41 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-10-28 13:01:54,636 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_102 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:01:54,637 - ERROR - mycodo.controllers.controller_input_0431299f - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-10-28 13:02:09,086 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_99 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:09,386 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_100 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:09,586 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_102 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:24,125 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_99 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:24,320 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_100 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:24,620 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_102 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:39,064 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_99 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:39,065 - ERROR - mycodo.controllers.controller_input_5e4f331d - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-10-28 13:02:39,365 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_100 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:39,366 - ERROR - mycodo.controllers.controller_input_8ecdce41 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-10-28 13:02:39,666 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_102 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:39,667 - ERROR - mycodo.controllers.controller_input_0431299f - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-10-28 13:02:54,077 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_99 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:54,377 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_100 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:02:54,577 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_102 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:03:09,121 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_99 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:03:09,416 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_100 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:03:09,617 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_102 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:03:24,060 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_99 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:03:24,061 - ERROR - mycodo.controllers.controller_input_5e4f331d - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-10-28 13:03:24,351 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_100 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:03:24,351 - ERROR - mycodo.controllers.controller_input_8ecdce41 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-10-28 13:03:24,649 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_102 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:03:24,649 - ERROR - mycodo.controllers.controller_input_0431299f - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-10-28 13:03:39,106 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_99 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:03:39,387 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_100 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:03:39,588 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_102 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
[... cut ...]
2022-10-28 13:06:24,655 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_102 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-10-28 13:06:24,656 - ERROR - mycodo.controllers.controller_input_0431299f - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-10-28 13:06:39,095 - DEBUG - mycodo.devices.atlas_scientific_i2c_1_99 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error

OnlinePirateDogFromSpaceAndBeyond2020, 2022-10-28:
https://stackoverflow.com/questions/52735862/getting-ioerror-errno-121-remote-i-o-error-with-smbus-on-python-raspberry-w