kizniche/Mycodo #614: Installation problems V 7.1.3 ( Error 502: Bad Gateway Error )

pmunz75, 2019-01-26:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.1.3
  - Raspberry New P3
 
#### Problem Description
Please list: Installation problems V 7.1.3 ( Error 502: Bad Gateway Error )
 
### Errors
Error 502: Bad Gateway Error

#### Restarting nginx
#### Restarting mycodoflask

#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...

#### Setting permissions

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
Job for mycodo.service failed because the control process exited with error code.
See "systemctl status mycodo.service" and "journalctl -xe" for details.
#### Mycodo Installer finished Sam Jän 26 12:53:12 CET 2019
#### Total install time: 16 minutes and 11 seconds

************************************
** Mycodo successfully installed! **
************************************

The full install log is located at:
/home/pi/Mycodo/install/setup.log

Go to https://10.11.12.175/, or whatever your Raspberry Pi's
IP address is, to create an admin user and log in.
pi@raspberrypi:~ $ cd /
pi@raspberrypi:/ $ sudo nano /home/pi/Mycodo/install/setup.log
pi@raspberrypi:/ $ sudo systemctl status mycodo.service
● mycodo.service - Mycodo server
   Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-01-26 13:14:47 CET; 24min ago
  Process: 2829 ExecStart=/var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py (code=exited, status=1/FAILURE)

Jän 26 13:14:46 raspberrypi systemd[1]: Starting Mycodo server...
Jän 26 13:14:47 raspberrypi python[2829]: Traceback (most recent call last):
Jän 26 13:14:47 raspberrypi python[2829]:   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 40, in <module>
Jän 26 13:14:47 raspberrypi python[2829]:     import rpyc
Jän 26 13:14:47 raspberrypi python[2829]: ImportError: No module named 'rpyc'
Jän 26 13:14:47 raspberrypi systemd[1]: mycodo.service: Control process exited, code=exited status=1
Jän 26 13:14:47 raspberrypi systemd[1]: Failed to start Mycodo server.
Jän 26 13:14:47 raspberrypi systemd[1]: mycodo.service: Unit entered failed state.
Jän 26 13:14:47 raspberrypi systemd[1]: mycodo.service: Failed with result 'exit-code'.
pi@raspberrypi:/ $ 

###########

pi@raspberrypi:/ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:/ $ 

############

Thanks.

cbaymac, 2019-01-26:
Same issue

kizniche, 2019-01-26:
If the web server can't start, the first page load cannot occur to create the Mycodo database. If the database doesn't exist, the Mycodo daemon will not start. Those errors about the daemon not starting are expected. Can you investigate why your web server isn't starting?

kizniche, 2019-01-26:
You can try to start the web server from the command line with the following commands:

```bash
cd ~/Mycodo/mycodo
sudo /var/mycodo-root/env/bin/gunicorn \

--workers 1 \

--worker-class gthread \

--threads 2 \

--timeout 300 \

--pid /var/lock/mycodoflask.pid \

--bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```

pmunz75, 2019-01-26:
sudo: /var/mycodo-root/env/bin/gunicorn: command not found

pmunz75, 2019-01-26:
Jän 26 13:14:47 raspberrypi python[2829]: Traceback (most recent call last):
Jän 26 13:14:47 raspberrypi python[2829]: File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 40, in
Jän 26 13:14:47 raspberrypi python[2829]: import rpyc
Jän 26 13:14:47 raspberrypi python[2829]: ImportError: No module named 'rpyc'
Jän 26 13:14:47 raspberrypi systemd[1]: mycodo.service: Control process exited, code=exited status=1
Jän 26 13:14:47 raspberrypi systemd[1]: Failed to start Mycodo server.

kizniche, 2019-01-26:
If you don't have gunicorn or rpyc you had something seriously go wrong with the install. I suggest you delete/rename your Mycodo directory and install again.

pmunz75, 2019-01-26:
I've already tried that 5 times.
the same error

kizniche, 2019-01-26:
There should be an indication of the issue in the install log.

pmunz75, 2019-01-26:
I just install everything completely new for testing.
please wait 15 minutes for info

pmunz75, 2019-01-26:
ok i have a new RP3 , new SD-Card and complete new installation of Mycodo.
The same issue.: Error 502
the log file : 
[setup.log](https://github.com/kizniche/Mycodo/files/2799151/setup.log)

kizniche, 2019-01-26:
Here's at least one issue:

```
Building wheel for bcrypt (PEP 517): finished with status 'error'
```

kizniche, 2019-01-26:
Perhaps you need:

```
sudo apt install python3-setuptools
```

kizniche, 2019-01-26:
This may the better method to get it:

```
~/Mycodo/env/bin/pip install --upgrade setuptools
```

pmunz75, 2019-01-26:
the same :  error 502

pi@raspberrypi:~ $ sudo apt install python3-setuptools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Suggested packages:
  python-setuptools-doc
The following NEW packages will be installed:
  python3-setuptools
0 upgraded, 1 newly installed, 0 to remove and 28 not upgraded.
Need to get 215 kB of archives.
After this operation, 944 kB of additional disk space will be used.
Get:1 http://debian.anexia.at/raspbian/raspbian stretch/main armhf python3-setuptools all 33.1.1-1 [215 kB]
Fetched 215 kB in 0s (336 kB/s)            
Selecting previously unselected package python3-setuptools.
(Reading database ... 37042 files and directories currently installed.)
Preparing to unpack .../python3-setuptools_33.1.1-1_all.deb ...
Unpacking python3-setuptools (33.1.1-1) ...
Setting up python3-setuptools (33.1.1-1) ...
pi@raspberrypi:~ $ sudo reboot

pmunz75, 2019-01-26:
Requirement already up-to-date: setuptools in ./Mycodo/env/lib/python3.5/site-packages (40.6.3)

kizniche, 2019-01-26:
You need to run setup.sh again after installing setuptools

pmunz75, 2019-01-26:
ok

kizniche, 2019-01-26:
Try installing 3.1.4:

```
~/Mycodo/env/bin/pip install bcrypt==3.1.4
```

pmunz75, 2019-01-26:
ok i will test now

pmunz75, 2019-01-26:
ok i have install : 
~/Mycodo/env/bin/pip install --upgrade setuptools
~/Mycodo/env/bin/pip install bcrypt==3.1.4
sudo reboot
/home/pi/Mycodo/install /.setup.sh

it does not work here : 

#### Restarting nginx
#### Restarting mycodoflask

#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...

kizniche, 2019-01-26:
We have to know why the web server doesn't start.

```
cd ~/Mycodo/mycodo
sudo /var/mycodo-root/env/bin/gunicorn \
--workers 1 \
--worker-class gthread \
--threads 2 \
--timeout 300 \
--pid /var/lock/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```

pmunz75, 2019-01-26:
ok please wait

kizniche, 2019-01-26:
Also, I assume if you don't copy any output for commands I give that there you've scrutinized them and determined there weren't any errors.

pmunz75, 2019-01-26:
pi@raspberrypi:/ $ cd ~/Mycodo/mycodo
pi@raspberrypi:~/Mycodo/mycodo $ sudo /var/mycodo-root/env/bin/gunicorn \
> --workers 1 \
> --worker-class gthread \
> --threads 2 \
> --timeout 300 \
> --pid /var/lock/mycodoflask.pid \
> --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
sudo: /var/mycodo-root/env/bin/gunicorn: command not found
pi@raspberrypi:~/Mycodo/mycodo $

pmunz75, 2019-01-26:
but not only I have the problem.

las error : sudo: /var/mycodo-root/env/bin/gunicorn: command not found

kizniche, 2019-01-26:
Just to be sure, you followed the install instructions and performed this before attempting to install Mycodo?

```
sudo apt update && sudo apt upgrade
```

pmunz75, 2019-01-26:
yes, I did everything

kizniche, 2019-01-26:
Try:

```
~/Mycodo/env/bin/pip install gunicorn==19.9.0
```

pmunz75, 2019-01-26:
ok
install : ~/Mycodo/env/bin/pip install gunicorn==19.9.0
reboot and setup.sh

error : 
#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
5 times and ready setup.

same issue : error 502

ngnix error log : 

2019/01/26 16:10:22 [crit] 3323#3323: *22 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 10.11.12.10, server: , request: "GET /favicon.ico HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/favicon.ico", host: "10.11.12.175"
2019/01/26 16:12:23 [crit] 12194#12194: *1 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 127.0.0.1, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/", host: "localhost"

/var/log/syslog : 

Jan 26 16:11:51 raspberrypi influxd[355]: [httpd] ::1 - - [26/Jan/2019:16:11:51 +0000] "HEAD /ping HTTP/1.1" 204 0 "-" "curl/7.52.1" 17383865-2185-11e9-800b-b827eb50867f 79
Jan 26 16:11:53 raspberrypi influxd[355]: [httpd] ::1 - - [26/Jan/2019:16:11:53 +0000] "GET /ping HTTP/1.1" 204 0 "-" "InfluxDBShell/1.7.2" 1839dbfa-2185-11e9-800c-b827eb50867f 81
Jan 26 16:11:53 raspberrypi influxd[355]: ts=2019-01-26T16:11:53.234960Z lvl=info msg="Executing query" log_id=0DEiDZbG000 service=query query="CREATE DATABASE mycodo_db"
Jan 26 16:11:53 raspberrypi influxd[355]: [httpd] ::1 - - [26/Jan/2019:16:11:53 +0000] "POST /query?chunked=true&db=&epoch=ns&q=CREATE+DATABASE+mycodo_db HTTP/1.1" 200 57 "-" "InfluxDBShell/1.7.2" 183a3022-2185-11e9-800d-b827eb50867f 4107
Jan 26 16:11:54 raspberrypi influxd[355]: [httpd] ::1 - - [26/Jan/2019:16:11:54 +0000] "GET /ping HTTP/1.1" 204 0 "-" "InfluxDBShell/1.7.2" 1939f7e3-2185-11e9-800e-b827eb50867f 79
Jan 26 16:11:54 raspberrypi influxd[355]: ts=2019-01-26T16:11:54.913521Z lvl=info msg="Executing query" log_id=0DEiDZbG000 service=query query="CREATE USER mycodo WITH PASSWORD [REDACTED]"
Jan 26 16:11:55 raspberrypi influxd[355]: [httpd] ::1 - - [26/Jan/2019:16:11:54 +0000] "POST /query?chunked=true&db=mycodo_db&epoch=ns&q=CREATE+USER+mycodo+WITH+PASSWORD+%5BREDACTED%5D HTTP/1.1" 200 57 "-" "InfluxDBShell/1.7.2" 193a3b67-2185-11e9-800f-b827eb50867f 341262
Jan 26 16:12:11 raspberrypi systemd[1]: Reloading.
Jan 26 16:12:12 raspberrypi systemd[1]: Reloading.
Jan 26 16:12:12 raspberrypi crontab[12000]: (root) LIST (root)
Jan 26 16:12:12 raspberrypi crontab[12002]: (root) LIST (root)
Jan 26 16:12:13 raspberrypi crontab[12004]: (root) REPLACE (root)
Jan 26 16:12:13 raspberrypi crontab[12007]: (root) LIST (root)
Jan 26 16:12:13 raspberrypi crontab[12009]: (root) LIST (root)
Jan 26 16:12:13 raspberrypi crontab[12011]: (root) LIST (root)
Jan 26 16:12:13 raspberrypi crontab[12013]: (root) REPLACE (root)
Jan 26 16:12:16 raspberrypi systemd[1]: Reloading.
Jan 26 16:12:16 raspberrypi systemd[1]: Reloading.
Jan 26 16:12:16 raspberrypi systemd[1]: Reloading.
Jan 26 16:12:17 raspberrypi systemd[1]: Reloading.
Jan 26 16:12:17 raspberrypi systemd[1]: Reloading.
Jan 26 16:12:17 raspberrypi systemd[1]: Reloading.
Jan 26 16:12:18 raspberrypi systemd[1]: Stopping A high performance web server and a reverse proxy server...
Jan 26 16:12:18 raspberrypi systemd[1]: Stopped A high performance web server and a reverse proxy server.
Jan 26 16:12:18 raspberrypi systemd[1]: Starting A high performance web server and a reverse proxy server...
Jan 26 16:12:18 raspberrypi systemd[1]: Started A high performance web server and a reverse proxy server.
Jan 26 16:12:23 raspberrypi systemd[1]: Started gunicorn daemon for Mycodo flask web user interface.
Jan 26 16:12:24 raspberrypi gunicorn[12204]: [2019-01-26 16:12:24 +0000] [12204] [INFO] Starting gunicorn 19.9.0
Jan 26 16:12:24 raspberrypi gunicorn[12204]: [2019-01-26 16:12:24 +0000] [12204] [INFO] Listening at: unix:/var/run/mycodoflask.sock (12204)
Jan 26 16:12:24 raspberrypi gunicorn[12204]: [2019-01-26 16:12:24 +0000] [12204] [INFO] Using worker: gthread
Jan 26 16:12:24 raspberrypi gunicorn[12204]: [2019-01-26 16:12:24 +0000] [12215] [INFO] Booting worker with pid: 12215
Jan 26 16:12:24 raspberrypi gunicorn[12204]: [2019-01-26 16:12:24 +0000] [12215] [ERROR] Exception in worker process
Jan 26 16:12:24 raspberrypi gunicorn[12204]: Traceback (most recent call last):
Jan 26 16:12:24 raspberrypi gunicorn[12204]:   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
Jan 26 16:12:24 raspberrypi gunicorn[12204]:     worker.init_process()
Jan 26 16:12:24 raspberrypi gunicorn[12204]:   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/gthread.py", line 104, in init_process
Jan 26 16:12:24 raspberrypi gunicorn[12204]:     super(ThreadWorker, self).init_process()
Jan 26 16:12:24 raspberrypi gunicorn[12204]:   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/base.py", line 129, in init_process
Jan 26 16:12:24 raspberrypi gunicorn[12204]:     self.load_wsgi()
Jan 26 16:12:24 raspberrypi gunicorn[12204]:   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/base.py", line 138, in load_wsgi
Jan 26 16:12:24 raspberrypi gunicorn[12204]:     self.wsgi = self.app.wsgi()
Jan 26 16:12:24 raspberrypi gunicorn[12204]:   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/app/base.py", line 67, in wsgi
[... cut ...]
Jan 26 16:15:15 raspberrypi influxd[355]: ts=2019-01-26T16:15:15.423454Z lvl=info msg="Retention policy deletion check (start)" log_id=0DEiDZbG000 service=retention trace_id=0DEjwwsl000 op_name=retention_delete_check op_event=start
Jan 26 16:15:15 raspberrypi influxd[355]: ts=2019-01-26T16:15:15.423828Z lvl=info msg="Retention policy deletion check (end)" log_id=0DEiDZbG000 service=retention trace_id=0DEjwwsl000 op_name=retention_delete_check op_event=end op_elapsed=0.458ms
Jan 26 16:16:15 raspberrypi systemd[1]: Started Session c4 of user pi.

kizniche, 2019-01-26:
You need to try the gunicorn command I gave after installing gunicorn.

cbaymac, 2019-01-26:
I've tried both Stretch Lite and Stretch Full. No joy.

kizniche, 2019-01-27:
This should be resolved with the just-released v7.1.4. It was an issue with bcrypt 3.1.5 and 3.1.6. I reverted back to using 3.1.4, which shouldn't be an issue with security since the newer versions seem to only add support for compiling on different systems.

kizniche, 2019-01-27:
Sorry for the issue. I should have tested and found this before releasing. This didn't affect upgraded systems, only new installs. I'll be sure to test this the next time I update python module versions.

closed 2019-01-27 by kizniche

kizniche, 2019-01-28:
Strangely, several of my Pis ***can*** install bcrypt 3.1.5 and 3.1.6. This is odd. I have no idea what's different about these systems from stock Raspbian.

JosephArturo, 2019-04-19:
@cbaymac 
@pmunz75 

I had the same problem.. I am using an OrangePI, had to do way more to get Mycodo working but the last error was the same you guys had. 

The real issue was:
Building wheel for bcrypt (PEP 517): finished with status 'error'

This also happend when doing:
~/Mycodo/env/bin/pip install bcrypt==3.1.4

I also did:
~/Mycodo/env/bin/pip install gunicorn==19.9.0

Finally resolved it with:
sudo apt-get install libffi-dev  build-essential
sudo ~/Mycodo/env/bin/pip install bcrypt==3.1.4

then I reinstalled the complete package.

kizniche, 2019-04-19:
> I am using an OrangePI, had to do way more to get Mycodo working

Can you elaborate on what you did. I will see if I can incorporate any changes that allows Mycodo to install on non-Raspberry Pi systems.

JosephArturo, 2019-04-23:
Hi Kyle,

Well just getting Mycodo to the point that it would run was trial and error.. now it runs but does not function, understandably so.

Install:
The OrangePI team ported Raspberian to the OrangePI but only the server edition, there are some requirements missing.. 

I had the follow errors installing;
- I had to install GIT otherwise I was not able to curl the install script
- python 2.7 (had to manually upgrade it to 3.4 and pip also)
- no raspi-config by default (had to manually install it and it's limited).
- no NginX ( also manually installed it).
- I also had the bcrypt error, manual build command corrected this.

Getting it to work on non RaspberryPi's:

I would port the app for off brand PI devboards to something like armbian (https://www.armbian.com)
Most off brand dev boards have limited (opensource) implementations of RPi.GPIO and this was the main issue with running your software on a board like the OrangePI. I was able to rename and import OPi.GPIO for example, but their implementation was so far off the original that it just did not work and I have no time to dive deeper into making it work at the moment. During the process of installing a sensor I for example got an execution stack error related to  'RPI_INFO', this is not implemented in the OPi.GPIO library. There are several implementations of RPi.GPIO for other boards.
https://opi-gpio.readthedocs.io/en/latest/index.html
http://wiki.friendlyarm.com/wiki/index.php/RPi.GPIO_:_NanoPi_NEO/NEO2/Air_GPIO_Programming_with_Python

Same with the PICAMERA module, most offbrand boards cannot use this module so making it not a requirement for an off brand installation is a must, I disabled the import in the python code and it allowed me to startup mycodo.. as I personally do not use a camera this was not an issue.

For now I switched back to an Raspberry Pi 2 cause I needed your application for a small server room, and was spending to much time to just getting it to run on another dev board but it would be nice to see such a possibility in the near future.

Hope it was any help to you, have a great week and thanks for the awesome application.