kizniche/Mycodo #615: Feature Request: Add ability to create multiple duplicate controls

SAM26K, 2019-01-27:
**## Mycodo Feature Request:**

  **- Specific Mycodo Version:
Mycodo Version: 7.1.4** 
Python Version: 3.5.3 (default, Jan 19 2017, 14:11:04) [GCC 6.3.0 20170124] 
Database Version: 70c828e05255
Daemon Status: Running 
Daemon Process ID: 14570 
Daemon RAM Usage: 77.912 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 52.44 MB 
Frontend Virtualenv: Yes

**Feature Request:** Add ability to create multiple duplicate controls.
When working with multi channel devices it would greatly reduce the work load to be able to create duplicate controls from existing controls.  If only the difference between controls have to be edited, it would greatly reduce the time required by the user to configure a Mycodo RPi instance.
Thanks for considering request.

kizniche, 2019-01-27:
Can you provide an example of what you're describing?

SAM26K, 2019-01-27:
It looks like output controls have the option to create multiple output control instances with one click but the feature is missing from data and function control creation. That's a start but would like to see the ability to extend to duplicating an existing control with user settings included. eg:
[image]

SAM26K, 2019-01-27:
The only difference between the multiple controls is the name and the exact path to the linux bash file.
Currently each control has to be created one at a time and each configured with the same info except for minor differences.

closed 2020-07-23 by kizniche