kizniche/Mycodo #617: PWM will not start

alonrab, 2019-01-30:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 ver. 7.15 on Pi3B v 1.2 
#### Problem Description
Hi All, I'm not sure what is causing this and would like to see if you have any advise? 
Trying to set up PWM to new pump controller (4-20 ma) with 0-3.3v signal (through noyito voltage to current convertor) 
When I add PWM output to pin 18 and turn on there is an error - cant find pigpio.
Did fresh install and still happens
When I put:
sudo pigpiod
in command port it works 
Any suggestions to solve this?

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)
step 1: add output (GPIO pin 19)
 
[image]

step2: set % and turn on

[image]

test error log: deamon,
[image]

step 3: in command port sudo pigpiod
[image]

step 4: turn output on again with 50%
[image]

random output that turn off after a few seconds.
 
For pin 18 the output turned green (on) but same problems accrue
 
Fool error tests:
tested that PWM work:
python code:

import RPi.GPIO as IO          #calling header file which helps us use GPIO’s of PI

import time                            #calling time to provide delays in program

IO.setwarnings(False)           #do not show any warnings

IO.setmode (IO.BCM)         #we are programming the GPIO by BCM pin numbers. (PIN35 as ‘GPIO19’)

IO.setup(19,IO.OUT)           # initialize GPIO19 as an output.

p = IO.PWM(19,22000)          #GPIO19 as PWM output, with 100Hz frequency
p.start(50)                             #generate PWM signal with 0% duty cycle

yielded the expected 1.62vdc at GPIO pin 19

Any suggestions?
many thanks!
### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2019-01-30:
Can you attach a screenshot of your Config -> Configure -> Raspberry Pi page?

kizniche, 2019-01-30:
Can you also attach your setup.log?

alonrab, 2019-01-30:
config config raspberry pi
[image]

set-up log
### Mycodo installation initiated Wed 30 Jan 10:12:50 EST 2019
#### Mycodo installation began Wed 30 Jan 10:15:12 EST 2019

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap currently set to 100 MB. Changing to 512 MB and restarting
Stopping dphys-swapfile (via systemctl): dphys-swapfile.service.
Starting dphys-swapfile (via systemctl): dphys-swapfile.service.

#### Updating apt repositories
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian stretch/main armhf Packages [205 kB]
Get:4 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [42.0 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:6 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.5 kB]
Fetched 12.0 MB in 9min 23s (21.4 kB/s)
Reading package lists...

#### Uninstalling apt version of pip (if installed)
Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring
  python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  python-pip*
0 upgraded, 0 newly installed, 1 to remove and 117 not upgraded.
After this operation, 671 kB disk space will be freed.
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
(Reading database ... 60%
(Reading database ... 65%
(Reading database ... 70%
(Reading database ... 75%
(Reading database ... 80%
(Reading database ... 85%
(Reading database ... 90%
(Reading database ... 95%
(Reading database ... 100%
(Reading database ... 105943 files and directories currently installed.)

Removing python-pip (9.0.1-2+rpt2) ...

Processing triggers for man-db (2.7.6.1-2) ...

#### Installing prerequisite apt packages and update pip
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Get:3 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [42.0 kB]
Fetched 67.4 kB in 1s (39.7 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
[... 709 lines cut ...]

#### Restarting nginx
#### Restarting mycodoflask

#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Successfully connected to http://localhost

#### Setting permissions

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
#### Mycodo Installer finished Wed 30 Jan 10:36:55 EST 2019
#### Total install time: 21 minutes and 43 seconds

************************************
** Mycodo successfully installed! **
************************************

The full install log is located at:
/home/pi/Mycodo/install/setup.log

Go to https://192.168.0.105/, or whatever your Raspberry Pi's
IP address is, to create an admin user and log in.

kizniche, 2019-01-30:
From your setup.log it doesn't seem as though this is a fresh install of Raspbian you installed Mycodo on. Let me know if this is not true.

alonrab, 2019-01-30:
maybe I am doing the fresh install wrong?
Did today:
Format, install pi, set up raspi config (vnc, i2c, wifi, expand memory), install mycodo, reboot

kizniche, 2019-01-30:
That is correct. I just needed verification. Try this command:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh install-pigpiod
```

kizniche, 2019-01-30:
Then reboot. That should get it working. This I believe is the issue, that this command is missing from setup.sh

kizniche, 2019-01-30:
I found the actual issue. pigpiod is not supposed to install at startup, but rather be a dependency for the PWM Output. When the PWM Output (or any of the Inputs that use pigpiod) is attempted to be added, the dependency system should install pigpiod for the first time. I just pushed a fix that will be incorporated into the 7.1.6 release.

kizniche, 2019-01-30:
An easy workaround for now is to just install a DHT22 Input before trying to install a PWM Output. The dependency system will install pigpiod for the DHT22.

alonrab, 2019-01-30:
did install + reboot.
There are no error in log but no PWM. 
Just saw your note, thank you very very much! this is great! will look into this as I'm not familiar with DHT22 but need to resume this later on tonight, 
best,

kizniche, 2019-01-30:
Your messing with pigpiod outside Mycodo may have affected it's ability to install pigpiod internally through Mycodo.

I also noticed you are using a hardware PWM pin but don't have the Hardware PWM Library selected in the Output options. Read the PWM section of the manual for explanations of this.

kizniche, 2019-01-30:
FYI, these are the Inputs that have pigpiod as a dependency and it should have been installed when any of the following Inputs were installed: DHT11, DHT22, HTU21D, Signal: PWM, Signal: Revolutions

alonrab, 2019-01-31:
Thanks for the updated version, worked after running the command sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh install-pigpiod
Did fresh install and set up again. 
this is what i found:
step 1.  Adding output PWM at GPIO pin 19 at 50%  - same problem in deamon log , cant find pigpio
step2:  Delete output, add sensor DHT22 at GPIO 4; check deamon log, can't find pigpio
step3:  run command: sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh install-pigpiod
            as the end of installation notes prompted a warning
            run command: udo systemctl daemon-reload
           Add PWM output, GPIO 19, 50% and turn on
           power output steady at 1.52V
seems to be resolved
Thanks

kizniche, 2019-02-01:
Okay, I think I finally fixed the issue. It ***should*** detect the missing pigpio dependency from a fresh install when a PWM output or an Input that requires pigpiod is attempted to be installed.

alonrab, 2019-02-04:
Hi, Tried today again with fresh install with version 7.1.7
Had the same problem with adding the PWM GPIO output 
for adding GPIO PWM output ,cant find pigpiod
checked that it was installed and it was (zero updates installed)
Deamon Log:

  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 919, in output_on
    trigger_conditionals=trigger_conditionals)
  File "/var/mycodo-root/mycodo/controller_output.py", line 450, in output_on_off
    self.output_switch(output_id, 'on', duty_cycle=duty_cycle)
  File "/var/mycodo-root/mycodo/controller_output.py", line 616, in output_switch
    self.pwm_hertz[output_id])
  File "/usr/lib/python3/dist-packages/pigpio.py", line 1542, in set_PWM_frequency
    _pigpio_command(self.sl, _PI_CMD_PFS, user_gpio, frequency))
  File "/usr/lib/python3/dist-packages/pigpio.py", line 983, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'
2019-02-04 12:47:35,702 - mycodo.output - ERROR - Cound not connect to pigpiod
2019-02-04 12:47:35,705 - mycodo.output - INFO - PWM 1feb705b setup on pin 21
2019-02-04 12:47:35,959 - mycodo.output - ERROR - Cound not connect to pigpiod
2019-02-04 12:47:35,960 - mycodo.output - INFO - PWM 1feb705b setup on pin 21
2019-02-04 12:47:44,810 - mycodo.daemon - ERROR - Could not turn output on: 'NoneType' object has no attribute 'send'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 919, in output_on
    trigger_conditionals=trigger_conditionals)
  File "/var/mycodo-root/mycodo/controller_output.py", line 450, in output_on_off
    self.output_switch(output_id, 'on', duty_cycle=duty_cycle)
  File "/var/mycodo-root/mycodo/controller_output.py", line 616, in output_switch
    self.pwm_hertz[output_id])
  File "/usr/lib/python3/dist-packages/pigpio.py", line 1542, in set_PWM_frequency
    _pigpio_command(self.sl, _PI_CMD_PFS, user_gpio, frequency))
  File "/usr/lib/python3/dist-packages/pigpio.py", line 983, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'
2019-02-04 12:51:12,106 - mycodo.output - ERROR - Cound not connect to pigpiod
2019-02-04 12:51:12,107 - mycodo.output - INFO - PWM cdbdddc8 setup on pin 21

kizniche, 2019-02-04:
Thanks for the update. I'll do another fresh install later so we can finally lay this bug to rest.

kizniche, 2019-02-04:
Did it prompt you to install pigpio via the dependency page when you attempted to add the PWM Output?

alonrab, 2019-02-04:
Report all dependencies for PWM were installed
Device: SIGNAL_PWM (Signal (PWM))
All dependencies have been installed for this device.

kizniche, 2019-02-04:
That doesn't tell me if pigpio was a dependency that was installed.

kizniche, 2019-02-04:
It indicating the PWM is on when there was an error preventing it is a bug. Pushing a fix for that.

My previous question seems like it was answered, but it's not clear that it was. When you first attempted to add a PWM Output, there are two scenarios that could have occurred immediately following clicking "Add" on the Output page:

1. You were taken to the dependency page and instructed to install the pigpio dependency before being able to add the PWM Output.
2. You were not taken to the dependency page and the PWM Output was added.

alonrab, 2019-02-04:
ok thanks!

kizniche, 2019-02-04:
Keep in mind this fix does not resolve the issue, it merely doesn't allow the output to show that it's been turned on when that error occurs. My question is still unanswered from comment https://github.com/kizniche/Mycodo/issues/617#issuecomment-460394125

alonrab, 2019-02-04:
Ia this about the dependency prompt?
clarify: there was no prompt to install dependency or any prompt to install/update
I was not taken to install/ scenario 2

alonrab, 2019-02-05:
Hi, Did fresh install today (format;install - on other installs)  no luck with PWM GPIO
see 1) deamon report; 2) setup.log

1) deamon report:

2019-02-05 19:46:40,041 - mycodo.daemon - INFO - Mycodo daemon v7.2.0 starting
2019-02-05 19:46:40,428 - mycodo.daemon - INFO - Anonymous statistics enabled
2019-02-05 19:46:40,457 - mycodo.daemon - INFO - Starting rpyc server
2019-02-05 19:46:40,839 - mycodo.output - INFO - Output controller activated in 109.4 ms
2019-02-05 19:46:41,340 - mycodo.daemon - INFO - All activated Conditional controllers started
2019-02-05 19:46:41,341 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-02-05 19:46:41,342 - mycodo.daemon - INFO - All activated Input controllers started
2019-02-05 19:46:41,343 - mycodo.daemon - INFO - All activated Math controllers started
2019-02-05 19:46:41,343 - mycodo.daemon - INFO - All activated PID controllers started
2019-02-05 19:46:41,344 - mycodo.daemon - INFO - All activated LCD controllers started
2019-02-05 19:46:41,845 - mycodo.daemon - INFO - Mycodo daemon started in 1.803 seconds
2019-02-05 19:46:41,849 - mycodo.daemon - INFO - 38.57 MB RAM in use
2019-02-05 19:56:59,188 - mycodo.output - ERROR - Cound not connect to pigpiod
2019-02-05 19:56:59,189 - mycodo.output - INFO - PWM ab92da4b setup on pin 16
2019-02-05 19:57:12,633 - mycodo.daemon - ERROR - Could not turn output on: 'NoneType' object has no attribute 'send'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 919, in output_on
    trigger_conditionals=trigger_conditionals)
  File "/var/mycodo-root/mycodo/controller_output.py", line 438, in output_on_off
    self.output_switch(output_id, 'on', duty_cycle=duty_cycle)
  File "/var/mycodo-root/mycodo/controller_output.py", line 616, in output_switch
    self.pwm_hertz[output_id])
  File "/usr/lib/python3/dist-packages/pigpio.py", line 1542, in set_PWM_frequency
    _pigpio_command(self.sl, _PI_CMD_PFS, user_gpio, frequency))
  File "/usr/lib/python3/dist-packages/pigpio.py", line 983, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'

2) setup.log

### Mycodo installation initiated Tue  5 Feb 16:10:49 UTC 2019
#### Mycodo installation began Tue  5 Feb 16:10:51 UTC 2019

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap currently set to 100 MB. Changing to 512 MB and restarting
Stopping dphys-swapfile (via systemctl): dphys-swapfile.service.
Starting dphys-swapfile (via systemctl): dphys-swapfile.service.

#### Updating apt repositories
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:2 http://archive.raspberrypi.org/debian stretch/main armhf Packages [205 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:5 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [42.0 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.5 kB]
Fetched 12.0 MB in 2min 15s (88.9 kB/s)
Reading package lists...

#### Uninstalling apt version of pip (if installed)
Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring
  python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  python-pip*
0 upgraded, 0 newly installed, 1 to remove and 131 not upgraded.
After this operation, 671 kB disk space will be freed.
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
[... 737 lines cut ...]

#### Restarting nginx
#### Restarting mycodoflask

#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...
#### Successfully connected to http://localhost

#### Setting permissions

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
#### Mycodo Installer finished Tue  5 Feb 16:25:45 UTC 2019
#### Total install time: 14 minutes and 54 seconds

************************************
** Mycodo successfully installed! **
************************************

The full install log is located at:
/home/pi/Mycodo/install/setup.log

Go to https://192.168.0.104/, or whatever your Raspberry Pi's
IP address is, to create an admin user and log in.

alonrab, 2019-02-05:
Tried adding DHT22 input and turning on, no error, but did not improved outcome for adding PWM

kizniche, 2019-02-05:
You were not redirected to th dependency page and prompted to install pigpio when you attempted to create a PWM output?

I tested this last night and it worked for me. I will do one more fresh install and see if I can replicate the behavior.

kizniche, 2019-02-05:
The Atlas pH Input has a dependency, but not pigpio.

https://github.com/kizniche/Mycodo/blob/506dc67a71e48d03a88439c58b4d2ecc7f8fb643/mycodo/inputs/atlas_ph.py#L63

Many Inputs have dependencies. This wasn't an issue with the dependency system as a whole, only adding Outputs. However, you indicating the pigpio dependency for the DHT22 wasn't installed when you attempted to install that Input is concerning.

alonrab, 2019-02-05:
yes, no error was raised for DHT22 in log.

kizniche, 2019-02-05:
That's not what I was suggesting. Disregard the log. When you add an Input that has a dependency, and that dependency isn't already installed, it should prompt you to install that dependency via the dependency page. If, when attempting to install the DHT22, it did not prompt you to install the pigpio dependency, then it indicates the issue is beyond merely the Output Dependency issue I described.

kizniche, 2019-02-07:
Which Raspbian version are you installing? I suspect you are not using the Lite version and this is the source of your issues, as python3-pigpio is coming pre-installed and that is indicating to Mycodo that it doesn't need to install its own scripts.

alonrab, 2019-02-07:
That is correct. I am not using the Lite version. This is probably why i encounter some difficulties. Did not know, thanks for pointing this - really wasn't aware they are so different.

kizniche, 2019-02-07:
I just released v7.2.1. I think I came up with a solution that works for both the Lite and non-Lite version of Raspbian, however I've only tested on the Lite version. It should also work on the non-Lite versions. If you're inclined to test, please let me know if it works.

alonrab, 2019-02-07:
Of course, will update what i find shortly

alonrab, 2019-02-07:
I think it is working as intended - Many thanks!
Did upgrade to v 7.2.1, add GPIO PWM 
was prompted to install the dependency 
after install, had to reboot -then adding GPIO, set up PWM and turn PWM % on were ok without error report. 

I'm away from the lab but will report if the output volt test is wrong - think its probably ok.
Best,

kizniche, 2019-02-07:
Thanks for the help. I'm going to consider this closed. If you find an issue, we can reopen it.

closed 2019-02-07 by kizniche