kizniche/Mycodo #619: flaky TCA 9545a?

Drakthule, 2019-01-31:
## Mycodo Issue Report:
  - 7.1.6
 
#### Problem Description
after trying two different switch doc lab mux'rs and even reloading the system, its only detecting one bme 280 despite having 3 seperate bme 280s hooked up on different bus branches 
### Errors 
I2C Bus 1: i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- UU -- -- -- --                         
I2C Bus 3: i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- UU -- -- -- --                         
I2C Bus 4: i2cdetect -y 4
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- UU -- -- 76 --                         
I2C Bus 5: i2cdetect -y 5
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- UU -- -- -- --                         
I2C Bus 6: i2cdetect -y 6
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- UU -- -- -- --                         
### Additional Notes
originally thought maybe it wasnt getting enough power but aside from the leds on the muxer shining brighter switching from 3.3 to 5 volts did nothing and one only bus is actually lighting up. i believe ive gotten the prerequisites, added the linked files from other github to the /boot/overlay folder and made the amendment to the config.txt file, honestly not sure whats going on

kizniche, 2019-02-08:
There are 8 channels yet you only have 5 listed above.

Drakthule, 2019-02-08:
Tca9545a is only 4 channels, I figured out most of it by peaking into the DTS before compiling for both the 9545 and 9548, only real issues I've got anymore is channel 3 and 4 always end up identical, like the devices are mirrored.

Theoi-Meteoroi, 2019-03-07:
I've seen some similar weirdness with address repeats from some devices.   I had some idea of the actual bus setting and used it and did not debug this.

I'm pretty sure this has nothing to do with Mycodo. This is a device tree problem with the camrex overlay that becomes apparent with my overlay.  Yes, I noticed.  I have also seen the 9548 do something like on its own with the default overlay when connected to the hardware i2c. I get different results when I use the GPIO overlay with GPIO i2c.  One of these days I'll put the new Saleae analyzer on it and figure out the issue.  Or someone else will figure it out, probably better than I would.

closed 2019-06-07 by kizniche