kizniche/Mycodo #620: Database upgrade failed during 7.1.6 upgrade

mrplow, 2019-02-01:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.1.6
 
#### Problem Description
- upgrade failed, web interface not running

 
### Upgrade Log
```
[2019-01-31 19:16:22] Installed version: 7.1.5                          
[2019-01-31 19:16:22] Latest version: 7.1.6
[2019-01-31 19:16:22]
[2019-01-31 19:16:22] #### Upgrade to v7.1.6 initiated 2019-01-31_19-16-17 ####
[2019-01-31 19:16:22]
[2019-01-31 19:16:22] #### Beginning Upgrade: Stage 1 of 2 ####
[2019-01-31 19:16:26] Stopping the Mycodo daemon...Done.
[2019-01-31 19:16:30] Downloading latest Mycodo version to /home/pi/mycodo-7.1.6.tar.gz...Done.
[2019-01-31 19:16:30] Creating /tmp/Mycodo-7.1.6...Done.
[2019-01-31 19:16:30] Extracting /home/pi/mycodo-7.1.6.tar.gz to /tmp/Mycodo-7.1.6...Done.
[2019-01-31 19:16:30] Removing /home/pi/mycodo-7.1.6.tar.gz...Done.
[2019-01-31 19:16:30] Copying /tmp/Mycodo-7.1.6/.upgrade status file to /tmp/Mycodo-7.1.6...Done.                                               
[2019-01-31 19:16:30] Moving env directory...Done.
[2019-01-31 19:16:30] Copying databases from /home/pi/Mycodo/databases/ to /tmp/Mycodo-7.1.6/databases...Done.
[2019-01-31 19:16:30] Copying flask_secret_key from /home/pi/Mycodo/databases/ to /tmp/Mycodo-7.1.6/databases...Done.
[2019-01-31 19:16:30] Copying statistics ID...Done.
[2019-01-31 19:16:30] Copying SSL certificates...Done.
[2019-01-31 19:16:30] Copying mycodo/inputs/custom_inputs...Done.
[2019-01-31 19:16:30] #### Stage 1 of 2 Complete ####
[2019-01-31 19:16:30]
[2019-01-31 19:16:30] #### Continuing Upgrade: Stage 2 of 2 ####        
[2019-01-31 19:16:30]
[2019-01-31 19:16:30] Moving old Mycodo from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2019-01-31_19-16-17-7.1.5...Done.
[2019-01-31 19:16:30] Moving new Mycodo from /tmp/Mycodo-7.1.6 to /home/pi/Mycodo...Done.                                                       
[2019-01-31 19:17:01]
[2019-01-31 19:17:01] #### Upgrading Mycodo database with alembic (if needed)                                                                   
[2019-01-31 19:17:02] Failed: Error while updating database with alembic.
[2019-01-31 19:17:02]                                                   
[2019-01-31 19:17:02]
[2019-01-31 19:17:02] There was an error during the upgrade.
[2019-01-31 19:17:02] Initial steps to try to fix:
[2019-01-31 19:17:02] 1. Reboot
[2019-01-31 19:17:02] 2. If that doesn't fix the issue, run the following command:
[2019-01-31 19:17:02] sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh upgrade
[2019-01-31 19:17:02] 3. If that command returns that you are running the latest version, run the following command:
[2019-01-31 19:17:02] sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh
[2019-01-31 19:17:02]
```

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Upgrade from 7.1.5 to 7.1.6?
I also ran the commands suggested by the log but it didn't resolve the issue but it did appear to delete the automatically created backup

kizniche, 2019-02-01:
Does your System Information page show there is a database version mismatch (colored red instead fo green)?

Can you restore a backed up version from the web interface?

If you don't have web interface access, you can restore a backup from the command line, replacing "/var/Mycodo-backups/Mycodo-backup-2018-03-11_21-19-15-5.6.4/" with the path to the particular backup you would like restored.:

```
sudo mycodo-commands /var/Mycodo-backups/Mycodo-backup-2018-03-11_21-19-15-5.6.4/
```

kizniche, 2019-02-01:
I've had all of my development systems (4) successfully upgrade to the latest release. I also see several user systems (~10) successfully upgrade. So I think this issue is isolated to your system. Let me know what you find with the database version information.

mrplow, 2019-02-01:
After a restore and running the upgrade again there were no issues.
Thanks!

closed 2019-02-01 by mrplow