kizniche/Mycodo #621: Unexpected error when activating PID

cbaymac, 2019-02-01:
## Mycodo Issue Report:
  - Specific Mycodo Version:
7.1.5
 
#### Problem Description
Please list: PID controller seems to be working with setpoint tracking for humidity, but shows error in log and after activation on functions page. see attached screen shots.
[image]
[image]

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)
2019-02-01 09:30:49,875 - mycodo.pid_29fafdfe - INFO - Deactivated in 137793735.4 ms
2019-02-01 09:33:38,100 - mycodo.pid_29fafdfe - INFO - Activated in 309.9 ms
2019-02-01 09:33:38,101 - mycodo.pid_29fafdfe - ERROR - Run Error: lookback must be greater or equal to sampletime
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_pid.py", line 211, in run
    noiseband=self.autotune_noiseband)
  File "/var/mycodo-root/mycodo/utils/pid_autotune.py", line 59, in __init__
    raise ValueError('lookback must be greater or equal to sampletime')
ValueError: lookback must be greater or equal to sampletime
2019-02-01 09:33:38,244 - mycodo.pid_29fafdfe - INFO - Deactivated in 137962104.3 ms

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1 Deactivate pid for humidity with set point tracking

 2. step 2... Activate pid
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue? Would you please take a look at my PID setting as that may be the problem. I am trying to use set point sine wave tracking for humidity + or - 10% around a core value of 89%
[image]

kizniche, 2019-02-01:
It appears you selected Start Autotune at one point. This looks like a bug in the autotune when the sample time (Period) is greater than 120. The PID controller is crashing instead of gracefully exiting. This causes the PID Autotune variable not to be unset, which in turn causes the Autotune to activate again when the PID is merely activated (causing another crash of the controller and a repetition of the cycle). I'll see about making a fix. In the meantime, you might be able to get the controller out of this issue by setting the Period to 59, activating the PID, deactivating the PID, then setting the Period back to 120. This should allow you to activate the PID again. If that doesn't work, you can delete the PID and add a new one and that should fix the issue. 

As for your PID settings, I see some issues, but I can't go into detail at this moment. I'll respond later when I can review them further and develop a thoughtful response.

cbaymac, 2019-02-01:
The PID controller no longer worked so made a new sd card. I was trying to Auto-Tune before this error which somehow I messed up.

cbaymac, 2019-02-01:
Thank you. I would assume that when I do try autotune i should disable set point tracking and let the program go from a low humidity to a higher one?

cbaymac, 2019-02-02:
Kyle,
  I may be missing it, but it would be great to have a mycodo users forum
to discuss settings such as PID and other of the many ways mycodo can be
best set up. I often have questions that are not related to software
problems and the manual just can’t go into detail on all of the best
settings for different applications. Another example is that I have found
the BME 280 to be a much better sensor than the DHT22 for grow room
humidity settings.
  It doesn’t seem to be a good use of your time for me to be asking for
your opinion on the best PID setting for a grow room humidifier for
example. I love your software and the gourmet mushrooms are really taking
off in this cooler weather. Thanks.

kizniche, 2019-02-02:
I set up a forum in the past but then decided to just use the github issue tracker. I'm willing to give the forum another shot, so this morning I set up SSL on my site and opened a forum, https://kylegabriel.com/forum

Let me know if there are any issues registering and posting. I'll be around later to respond. If there isn't a post there, I'll return here to respond to your tuning question I said I'd get back to.

kizniche, 2019-02-02:
The fix for this issue is live and discussion moved to https://kylegabriel.com/forum/pid-tuning/help-improve-this-pid-tuning/

closed 2019-02-02 by kizniche