kizniche/Mycodo #624: Pigpiod fails

0x0f906, 2019-02-08:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.2.1
 
#### Problem Description
Please list: DHT22 fails 
Possibly related to pigpio

  - what were you trying to do
start DHT22 sensor
  - specific setup details that are involved
 DHT22

### Errors
  - Copy and pasting crash logs, or link to any specific
2019-02-07 19:28:32,658 - mycodo.input_d0d1945c - INFO - Activated in 222.3 ms
2019-02-07 19:28:32,659 - mycodo.dht22_d0d1945c - ERROR - Could not connect to pigpiod. Ensure it is running and try again.
2019-02-07 19:28:32,660 - mycodo.input_d0d1945c - ERROR - Error: 'tuple' object has no attribute 'items'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_input.py", line 303, in run
    self.unique_id, self.create_measurements_dict())
  File "/var/mycodo-root/mycodo/controller_input.py", line 427, in create_measurements_dict
    for each_channel, each_measurement in self.measurement.values.items():
AttributeError: 'tuple' object has no attribute 'items'

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
Try to start DHT22 sensor

 
### Additional Notes
Issue worked before latest upgrade.

kizniche, 2019-02-08:
Is pigpio running? Use this at the command line:

```bash
ps aux | grep pigpio
```

0x0f906, 2019-02-08:
I'm sorry I thought I added that in there. Yes pigpiod is running.

kizniche, 2019-02-08:
Have you restarted your Pi since the upgrade?

0x0f906, 2019-02-08:
Yes, I think maybe this is a pigpio issue vs a mycodo. Here's my syslog showing pigpio failing. Hmmmm

This is a fresh install of rasbian-lite updated and upgraded. 

Starting pigpiod service (low latency)...
Feb  7 19:42:52 voyager systemd[1]: Started pigpiod service (low latency).
Feb  7 19:42:52 voyager pigpiod[763]: 2019-02-07 19:42:52 initMboxBlock: init mbox zaps failed
Feb  7 19:42:52 voyager pigpiod[763]: Can't initialise pigpio library
Feb  7 19:42:52 voyager systemd[1]: pigpiod_low.service: Main process exited, code=exited, status=1/FAILURE
Feb  7 19:42:52 voyager killall[766]: pigpiod: no process found
Feb  7 19:42:52 voyager systemd[1]: pigpiod_low.service: Control process exited, code=exited status=1
Feb  7 19:42:52 voyager systemd[1]: pigpiod_low.service: Unit entered failed state.
Feb  7 19:42:52 voyager systemd[1]: pigpiod_low.service: Failed with result 'exit-code'.
Feb  7 19:42:56 voyager influxd[339]: [httpd] ::1 - mycodo [07/Feb/2019:19:42:56 -0800] "POST /write?db=mycodo_db HTTP/1.1" 204 0 "-" "python-requests/2.21.0" 9f384906-2b53-11e9-8021-b827eb410660 8669
Feb  7 19:42:57 voyager systemd[1]: pigpiod_low.service: Service hold-off time over, scheduling restart.
Feb  7 19:42:57 voyager systemd[1]: Stopped pigpiod service (low latency).
Feb  7 19:42:57 voyager systemd[1]: Starting pigpiod service (low latency)...
Feb  7 19:42:57 voyager systemd[1]: Started pigpiod service (low latency).
Feb  7 19:42:57 voyager pigpiod[780]: 2019-02-07 19:42:57 initMboxBlock: init mbox zaps failed
Feb  7 19:42:57 voyager pigpiod[780]: Can't initialise pigpio library
Feb  7 19:42:57 voyager systemd[1]: pigpiod_low.service: Main process exited, code=exited, status=1/FAILURE
Feb  7 19:42:57 voyager killall[783]: pigpiod: no process found
Feb  7 19:42:57 voyager systemd[1]: pigpiod_low.service: Control process exited, code=exited status=1
Feb  7 19:42:57 voyager systemd[1]: pigpiod_low.service: Unit entered failed state.
Feb  7 19:42:57 voyager systemd[1]: pigpiod_low.service: Failed with result 'exit-code'

kizniche, 2019-02-08:
Try going to [Config] -> Configure -> Raspberry Pi and changing the "Configure pigpiod" option from "1 ms (faster)" or "5 ms (slower)" to "Uninstall pigpiod", then click Reconfigure. Then give that a minute or two to complete. Refresh the page and confirm the default value of that selection box is "Uninstall pigpiod", and then select "1 ms (faster)" and click Reconfigure.

kizniche, 2019-02-08:
This seems to be the issue:

```
Feb 7 19:42:52 voyager killall[766]: pigpiod: no process found
```

kizniche, 2019-02-08:
I will flash a fresh Raspbian Lite, install Mycodo, and test with my DHT22 tomorrow. I'm out of time tonight. If you discover any more info that could indicate the issue, please post and I'll review it when I can.

kizniche, 2019-02-08:
If you need a functioning system until then, I'd recommend flashing Raspbian and installing v7.2.0 by downloading [that specific release](https://github.com/kizniche/Mycodo/releases/tag/v7.2.0), extract it to ~/Mycodo and execute ```sudo /bin/bash ~/Mycodo/install/setup.sh``` to install it.

0x0f906, 2019-02-08:
It must have been related to the update cycle of pigpio and the DHT22. It needed to be set 5ms and now it seems to work. Strange thing is it worked well on the default setting before the latest upgrade. Although now I'd bet a pigpio update was to blame. Thanks for the suggestion, sometimes an outside voice helps when you're going round and round.

closed 2019-02-08 by 0x0f906

kizniche, 2019-02-08:
Good to hear! If you're not using PWM at a high frequency, then you really don't need a 1ms sample rate anyway. Though that is concerning it's not working at 1 ms. I have a system right now running at 1 ms and the DHT22 is fine after upgrading to 7.2.1 from earlier versions. I'll still test a fresh install with my DHT22 sometime soon to see if I can replicate the issue. Thanks for the report and glad it was resolved so quickly.