kizniche/Mycodo #625: BMP280 - I2C adress issue

ErlendLarsen, 2019-02-08:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.2.0

 
#### Problem Description
Trying to add second BMP280 sensor, but can't select adress 0x76, only 0x77(no drop-down menu at all). 
I have soldered SD0 to VCC as said in datasheet, and it is detected by i2cdetect -y 1
There is just no list for options on BMP280, I can do this with BME280, but then I get values with 0 and values that I dont need. 
 

Thanks! 
Erlend

kizniche, 2019-02-08:
Thanks for the bug report. I'll fix this today and make a new release you can upgrade to. I'll update you when this happens.

ErlendLarsen, 2019-02-11:
Thank you so much! :)

ErlendLarsen, 2019-02-11:
Just updated, and it seems like I can't get it to work, on both adresses.
When I try to add a new input it gives me an error:
    Error: Add Input: (sqlite3.OperationalError) disk I/O error (Background on this error at: http://sqlalche.me/e/e3q8)
Also when I try to activate the input it says: 
    "Error: Could not find measurments"

I can still deactivate and reactivate the input I had added earlier, which seems to work on both addresses.

kizniche, 2019-02-12:
You tried the working circuit on both address. Did you try the non-working circuit on both addresses?

ErlendLarsen, 2019-02-12:
I have tried everything, all my sensors works with the BME and the already working BMP. But when I add a new one, that one doesnt work.

kizniche, 2019-02-12:
I'm not exactly sure what your comment means. Can you be more descriptive? In order to determine if it's the hardware of the software that's the issue, can you plug in the non-working circuit by itself and confirm if it does or doesn't work? Then plug in the working circuit and confirm if it does or doesn't work.

kizniche, 2019-02-17:
> "Error: Could not find measurments"

This error indicates there's an issue with the measurement units being used. Did you change the measurement units from their default?

> Error: Add Input: (sqlite3.OperationalError) disk I/O error (Background on this error at: http://sqlalche.me/e/e3q8)

This indicates there is an issue with your filesystem. Have you tried using a different SD card?

kizniche, 2019-02-17:
Are there any errors in the daemon log?

closed 2019-04-18 by kizniche