kizniche/Mycodo #626: Trigger failure

0x0f906, 2019-02-08:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.2.1
 
#### Problem Description
Please list: All triggers are producing the same error. Will not function. 

  - what were you trying to do: Create a trigger

 
### Errors
2019-02-08 15:53:34,652 - mycodo.trigger_6745106f - INFO - Activated in 89.4 ms
2019-02-08 15:53:35,038 - mycodo.trigger_6745106f - ERROR - Run Error: trigger_function_actions() got an unexpected keyword argument 'last_measurement'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_trigger.py", line 167, in run
    self.check_triggers()
  File "/var/mycodo-root/mycodo/controller_trigger.py", line 431, in check_triggers
    edge=gpio_state)
TypeError: trigger_function_actions() got an unexpected keyword argument 'last_measurement'

### Steps to Reproduce the issue:
How can this issue be reproduced?
Create a trigger with a start point and wait for it to cycle/trigger.
 
### Additional Notes
Sorry for another bug report! No worries or rush on my end.

kizniche, 2019-02-09:
I actually just got this error as well. I'll try to get a fix and new release out tonight, if not tomorrow.

0x0f906, 2019-02-09:
> I actually just got this error as well. I'll try to get a fix and new release out tonight, if not tomorrow.

Very cool man, thanks again as always.

closed 2019-02-09 by kizniche