kizniche/Mycodo #629: PID fails to change output

drgrumpy, 2019-02-14:
## Mycodo Issue Report:
  Version: 6.4.5
 
#### Problem Description
Please list:

PID does not switch off output
 
### Errors
No errors in the logs.

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2019-02-14:
I only support current versions of Mycodo. I'm sure you can imagine with close to 200 releases how much time I could spend investigating issues from old versions (not to mention how much wasted time if the issue has already been fixed). If you still have the issue with the latest version, I'll take a look.

closed 2019-02-14 by kizniche

drgrumpy, 2019-02-14:
OK appreciate that. But the latest version 7 is a major jump, resulting in a lot of time/work to upgrade, meaning system is out of action for a while, and may not work after upgrade.  Ideally I would shutdown take a backup or the working system and then do the upgrade. But my problem is that I can't easily retrieve the sd card to take a back up before upgrading, and if I do, I risk all sorts of issues with connections (I know this from previous experience with upgrades and taking days to get back to where I started).

Would appreciate, if you could give me a clue as to where to look in the code (in controller_pid I guess), I may be able to find it myself.

Thanks, Steve

kizniche, 2019-02-14:
I wouldn't know where to look, as that code significantly different in many areas from the current version. I also only have current versions installed on my development systems, so I wouldn't even be able to test it. I would recommend flashing a new SD card to test and saving the old one, which makes it very easy to restore the previous version by just putting the old SD card back in. I would like to help, but my time is better spent fixing current issues rather than reconfiguring my entire development system and investigating a bug in old code that I'm sure is already fixed in the current version.

kizniche, 2019-02-16:
If you want to try 7, you can just rename your ~/Mycodo directory to something else (e.g. ~/Mycodo-6.4.5), then install the latest version. If you want to go back to the previous version, delete/rename the new ~Mycodo to something else, rename ~/Mycodo-6.4.5 to ~/Mycodo and then run ```sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh``` and it will set up the old version.

drgrumpy, 2019-02-16:
OK, thanks. I will try that at some less critical time (when the weather is warmer!).

After some trials I have now got it working as I want:
First I added a conditional (daily span) to turn on the PID at 06:00 to 18:00, I expected the PID would be switched off at 18:00 but it wasn't... so I added another condition (daily point) to turn off the PID at 18:00 (that has just worked)!