kizniche/Mycodo #630: LiveCamera in Dashboard

Ericktz, 2019-02-16:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.2.2
 
#### Problem Description
Please list:

i'm setup a new camera in /dev/video0 , then in dashboard when im create a live stream get an error when refresh.
### Errors
  - List any errors you encountered.
```
Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_general.py", line 179, in video_feed
    camera_stream = import_module('mycodo.mycodo_flask.camera.camera_' + camera_options.library).Camera
  File "/home/pi/Mycodo/env/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 956, in _find_and_load_unlocked
ImportError: No module named 'mycodo.mycodo_flask.camera.camera_fswebcam'
```
the next images can you help for reference 
[image]
[image]

kizniche, 2019-02-16:
Live streaming is currently only supported with the picamera library. I haven't built that feature for the fswebcam library. I might have mentioned this in the manual, but it looks as thought I didn't add a check to prevent users from setting the camera dashboard widget to stream with the fswebcam library, or indicate why there's an error with this configuration.

kizniche, 2019-02-16:
I'll plug my USB camera into my development pi in the next week and see if I can get it working.

closed 2019-05-06 by kizniche