kizniche/Mycodo #631: error 500 when selecting function tab

SAM26K, 2019-02-20:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 Mycodo Version: 7.2.3 
Python Version: 3.5.3 (default, Jan 19 2017, 14:11:04) [GCC 6.3.0 20170124] 
Database Version: 2976b41930ad
Daemon Status: Running 
Daemon Process ID: 10805 
Daemon RAM Usage: 70.248 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 51.476 MB 
Frontend Virtualenv: Yes
#### Problem Description
Error 500 when selecting function tab.  Happens on 2 rpi 3's after upgrading.
 
### Errors
  Mycodo Version: 7.2.3 
Python Version: 3.5.3 (default, Jan 19 2017, 14:11:04) [GCC 6.3.0 20170124] 
Database Version: 2976b41930ad
Daemon Status: Running 
Daemon Process ID: 10805 
Daemon RAM Usage: 70.248 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 51.476 MB 
Frontend Virtualenv: Yes

Daemon log:
2019-02-20 13:57:44,542 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:57:49,749 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:57:49,770 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 13:57:54,545 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:57:59,542 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:04,578 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:04,580 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 13:58:09,546 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:14,543 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:19,638 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:19,640 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 13:58:24,545 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:29,542 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:34,571 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:34,572 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 13:58:39,551 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:44,540 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:49,690 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:49,692 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 13:58:54,542 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:58:59,536 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:04,541 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:04,542 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 13:59:09,538 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:14,536 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:19,580 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:19,585 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 13:59:24,544 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:29,549 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:34,711 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:34,712 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 13:59:39,545 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:44,546 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:49,637 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:49,638 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 13:59:54,550 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 13:59:59,544 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 14:00:04,563 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 14:00:04,565 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 14:00:09,541 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-20 14:00:14,539 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.

not5, 2019-02-20:
Same issue on Dashboard tab for me. Different error reason:
Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 581, in page_dashboard
    input_dev, dict_units, dict_measurements)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_general.py", line 279, in choices_inputs
    choices, each_input, dict_units, dict_measurements)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_general.py", line 398, in form_input_choices
    cnum=channel + 1, cname=channel_name)
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'

SAM26K, 2019-02-20:
Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1350, in page_function
    input_dev, dict_units, dict_measurements)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_general.py", line 279, in choices_inputs
    choices, each_input, dict_units, dict_measurements)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_general.py", line 398, in form_input_choices
    cnum=channel + 1, cname=channel_name)
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'

kizniche, 2019-02-20:
Sorry about that. I pushed a potential fix. Could one of you [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and see if the issue is resolved? Once I get confirmation, I'll make a release with the fix.

kizniche, 2019-02-20:
```
2019-02-20 13:58:34,572 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-20 13:58:39,551 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
```

These errors are not related to this issue, however you should investigate why you are receiving these errors.

not5, 2019-02-20:
Your changes fixed the error on the Dash & function pages for me.

Thanks.

closed 2019-02-21 by kizniche

SAM26K, 2019-02-21:
Fixed my problem too.  The Daemon errors where in some of the linux command
files and should be fixed now.  But now all the output commands give a
"red" feedback tip window instead of green.