kizniche/Mycodo #633: Feedback tip windows are red when toggeling output controls on/off

SAM26K, 2019-02-22:
## Mycodo Issue Report:
 Mycodo Version: 7.2.4 
Python Version: 3.5.3 (default, Jan 19 2017, 14:11:04) [GCC 6.3.0 20170124] 
Database Version: 2976b41930ad
Daemon Status: Running 
Daemon Process ID: 1237 
Daemon RAM Usage: 80.032 MB 
Daemon Virtualenv: Yes 
Frontend RAM Usage: 50.02 MB 
Frontend Virtualenv: Yes
uptime
 00:00:37 up 15:58,  1 user,  load average: 3.18, 3.11, 2.96
Kernel Information: uname -a
Linux raspberrypi3_3 4.14.70-v7+ #1144 SMP Tue Sep 18 17:34:46 BST 2018 armv7l GNU/Linux
#### Problem Description
Please list:

  -When toggling any output control (GPIO or Linux command)
[image]
 the tip window is red instead of green.
Not sure if these errors are related:

ast 30 lines of /var/log/mycodo/mycodo.log:
2019-02-22 00:01:28,659 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-22 00:01:33,661 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:01:38,651 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:01:43,664 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:01:43,666 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-22 00:01:48,675 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:01:53,659 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:01:58,663 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:01:58,668 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-22 00:02:03,669 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:08,679 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:13,655 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:13,657 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-22 00:02:18,682 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:23,656 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:28,651 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:28,653 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-22 00:02:33,652 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:38,663 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:43,653 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:43,657 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-22 00:02:48,668 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:53,683 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:58,654 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:02:58,655 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-22 00:03:03,657 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:03:08,689 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:03:13,652 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.
2019-02-22 00:03:13,654 - mycodo.input_9c25ea72 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2019-02-22 00:03:18,742 - mycodo.linux_command_9c25ea72 - ERROR - The command returned a non-numerical value. Ensure only one numerical value is returned by the command.

SAM26K, 2019-02-22:
The outputs do work as expected.

kizniche, 2019-02-22:
This issue is a result of #632, which has been fixed but not released. Your log errors are unrelated to this issue.

closed 2019-02-22 by kizniche