kizniche/Mycodo #634: Hydroponic

hkn1979, 2019-02-28:
### Additional Notes
Can you add Hydroponics system? pls.

kizniche, 2019-02-28:
Can you elaborate what you mean?

hkn1979, 2019-02-28:
I think, you know very well hydroponics system, but
I must select the product property (vegetable)
...measure the nutrients in water
... and pump the water periyodical time.
I'am newbee this area but, your project very well, maybe You want to add this propertys. thanks.

kizniche, 2019-03-01:
I don't see why Mycodo couldn't be used to operate a hydroponic system. I've never built one, so I wouldn't be able to recommend any equipment. If you have questions about Mycodo, I'll be happy to answer them.

hkn1979, 2019-03-01:
Thanks kizniche, Pls don't close comment. Maybe someone wants to write some information, we will wait.

RynFlutsch, 2019-03-05:
Hey Ho,
check atlas-scientific.com for hardware. I am at this project too but i am really busy. Next days i try to setup the industrial Probe and hope for Kyles support ;-D
You Need some stuff for a whole automated Hydroponic system. 
pH-sensor, EC-sensor, Temp-sensor, Maybe O2-sensor water-valves and pumps. And don't forget the ion exchanger or Reverse osmosis equipment. That is really complex. My advice is to start with automated irrigation

Theoi-Meteoroi, 2019-03-07:
Mycodo is pretty close to plug and play from a certain perspective.  Have you successfully built and deployed sensor applications?  Do you understand i2c and serial communications?   Are you deeply familiar with your sensing target?  Mycodo fills a need for a flexible sensor system that can perform  PID control to maintain a control point.  And way much more.  If your head just exploded, visit the fine folks at [Adafruit](adafruit.com) and come back when the term PID makes sense.  They offer the hardware to do what you want. There are other ways and parts you can use but they support what they sell if you have trouble.  Find out how to use that hardware. PH, temp or whatever it is that fits your definition of "hydro".   This is radical self-reliance, and part of how this thing works.

closed 2019-06-07 by kizniche