kizniche/Mycodo #635: Post Upgrade Error

CriticalFlaw, 2019-03-03:
Hello, I've recently done the upgrade from version 5 to 7. I can get to the login page but after logging in the following error is shown. Looks to be a database issue. Any help would be greatly appreciated. 

Error (Full Traceback):

```
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 509, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: output.on_state
```

The above exception was the direct cause of the following exception:

```
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 466, in page_dashboard
    output = Output.query.all()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2843, in all
    return list(self)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2995, in __iter__
    return self._execute_and_instances(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 3018, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 948, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/sql/elements.py", line 269, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1060, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1200, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1413, in _handle_dbapi_exception
    exc_info
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 265, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 248, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 509, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: output.on_state [SQL: 'SELECT output.id AS output_id, output.unique_id AS output_unique_id, output.output_type AS output_output_type, output.interface AS output_interface, output.location AS output_location, output.i2c_bus AS output_i2c_bus, output.baud_rate AS output_baud_rate, output.name AS output_name, output.measurement AS output_measurement, output.unit AS output_unit, output.conversion_id AS output_conversion_id, output.channel AS output_channel, output.pin AS output_pin, output.on_state AS output_on_state, output.amps AS output_amps, output.on_until AS output_on_until, output.last_duration AS output_last_duration, output.on_duration AS output_on_duration, output.protocol AS output_protocol, output.pulse_length AS output_pulse_length, output.on_command AS output_on_command, output.off_command AS output_off_command, output.pwm_command AS output_pwm_command, output.trigger_functions_at_start AS output_trigger_functions_at_start, output.state_at_startup AS output_state_at_startup, output.state_at_shutdown AS output_state_at_shutdown, output.pwm_hertz AS output_pwm_hertz, output.pwm_library AS output_pwm_library, output.pwm_invert_signal AS output_pwm_invert_signal, output.flow_rate AS output_flow_rate \nFROM output'] (Background on this error at: http://sqlalche.me/e/e3q8)
```

kizniche, 2019-03-04:
Upgrading from <7 to >=7 should have deleted your database so a database with the new schema could be created. Move your current database with ```sudo mv ~/Mycodo/databases/mycodo.db ~/Mycodo/databases/mycodo.db.bu```, restart the fronted with ```sudo service mycodoflask restart```, then reload the web page.

CriticalFlaw, 2019-03-04:
That did it. Thank you very much!

closed 2019-03-04 by CriticalFlaw