kizniche/Mycodo #636: Add Mycodo functions for use with conditional user code

kizniche, 2019-03-05:
Add functions such as historical data querying, allowing single or multiple measurements to be obtained, summed, averaged, etc. and used in conditional statement code.

My particular use at the moment is as follows (however there are many other uses):

I have a Measurement Trigger will turn on a water pump for 10 seconds every 60 seconds if an Edge Input measurement is 1 (water level low). Once the water is at the correct level this measurement is 0 and the water pump doesn't run. A faulty float sensor caused 1 to be continually measured, even when the water level was well above the float sensor, causing all the water to be transferred from my reservoir to the container that needed filling. There was no Mycodo system available to determine this was acting abnormally. Typically it takes 3 10-second operations of the pump to bring the water level from the low mark to the high mark. In this case, the pump continually ran for 10 seconds every 60 seconds all night (10 hours).

To prevent this, a conditional can be created that queries the past 10 minutes and sums the durations the pump has been on. If the sum of the durations is greater than 60 (6 times turned on in the past 10 minutes), then it will turn off the Measurement Trigger that operates the pump and send a notification email informing me the float valve isn't operating properly.

Theoi-Meteoroi, 2019-03-07:
Compelling use case.  I've puzzled about various faulty sensor scenarios.  Retrospection could add sane responses to failure cases.  But would be an advanced function, I think.

not5, 2019-04-02:
Interesting. I may be able to use this for a water sensor I put together that has some hysteresis.... it bounces around high and low when there's no water, but the average is definitely below 0.5 for any 5 second duration

zsole2, 2019-12-10:
Is there any development on this issue? Based on your recent commit, it looks like it, but could not find any reference or example on it in the guides.
I converting  my conditionals from the old system, and really like the possibilities for the new Python conditionals. Checking the cumulative ON time of an output would be indeed useful (maybe not the past 10 minutes, this would be too short for my fridges' cycle)

As a side note, is there a way to include only the self-generated message into the e-mail that is sent? Maybe making optional to add the current full details, e.g. when someone needs to debug something. This way I can only get in the notification what I need to see to quickly get the necessary info.

kizniche, 2020-01-24:
It has been released. you can find documentation on the new conditional conditions at https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#conditional-options

> is there a way to include only the self-generated message into the e-mail that is sent?

Does this mean you do not want to include the code that's appended to the message? If so, all you need to do is use:

```python
self.message = "my message"
```

instead of:

```python
self.message += "my message"
```

"+=" appends your message to the current message (which includes the conditional code), while "=" changes the message to your message. Make sure if you want to append anything after this initial message, you start using "+=" again, otherwise you will be overwriting your previous message with the new one. It may be best to just have ```self.message = ""``` at the beginning of your conditional to clear the message, then just use "+=" throughout the rest of your conditional code.

zsole2, 2020-01-25:
Thanks, these appears both answer my question. I'll test and revert soon if anything is not clear.

closed 2020-07-23 by kizniche