kizniche/Mycodo #637: Dockerized Mycodo

kizniche, 2019-03-05:
I've started a new branch to experiment with using docker and docker-compose to run Mycodo. Thus far I successfully got Mycodo to run fully in docker containers. This includes the Flask/Gunicorn frontend, Nginx, Influxdb, and the Mycodo daemon. This issue can serve as a discussion thread and a way I can keep users updated on the progress and where this is going.

Because of this, I've successfully built and ran Mycodo on my 64-bit AMD desktop computer in addition to Raspberry Pis. Theoretically Mycodo can now run on any system that can run Linux.

kizniche, 2019-03-06:
To try it out, use these commands:

```
git clone https://github.com/kizniche/Mycodo
cd Mycodo
git checkout mycodo_docker
make install-docker
make build
```

Once it's built and running, you can navigate to https://127.0.0.1/ to access the web UI.

Theoi-Meteoroi, 2019-03-07:
I did this. On the stretch distro for a long run node. This is my CO2 and particle detector station. 

pi@CO2mon:~ $ sudo mv ./Mycodo ./645_Mycodo.saved
pi@CO2mon:~ $ sudo apt-get update
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]  
Get:3 http://archive.raspberrypi.org/debian stretch/main armhf Packages [214 kB]
Get:4 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:5 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [44.4 kB]
Get:6 http://mirrordirector.raspbian.org/raspbian stretch/non-free armhf Packages [95.5 kB]
Fetched 12.1 MB in 22s (530 kB/s)                                              
Reading package lists... Done
pi@CO2mon:~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
WARNING: This update bumps to rpi-4.19.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=224931
##############################################################
Would you like to proceed? (y/N)
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    464      0 --:--:-- --:--:-- --:--:--   465
100 58.2M  100 58.2M    0     0  1995k      0  0:00:29  0:00:29 --:--:-- 1488k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.19.25+
 *** depmod 4.19.25-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 82f49d8dfd9c000a203fb1deae64a3eb87abf3c1
 *** A reboot is needed to activate the new firmware
 <reboot and apt-get update to get goodness>
<why I used apt-get update? It seemed the conservative thing to do>
<https://askubuntu.com/questions/194651/why-use-apt-get-upgrade-instead-of-apt-get-dist-upgrade>

Got this (good to know because I use that on Pi.)
apt-listchanges: News
---------------------

wpasupplicant (2:2.6-19) unstable; urgency=medium

  With this release, wpasupplicant no longer respects the system
  default minimum TLS version, defaulting to TLSv1.0, not TLSv1.2. If
  you're sure you will never connect to EAP networks requiring anything less
  than 1.2, add this to your wpasupplicant configuration:

    tls_disable_tlsv1_0=1
    tls_disable_tlsv1_1=1

  wpasupplicant also defaults to a security level 1, instead of the system
  default 2. Should you need to change that, change this setting in your
  wpasupplicant configuration:

    openssl_ciphers=DEFAULT@SECLEVEL=2

  Unlike wpasupplicant, hostapd still respects system defaults.

 -- Andrej Shadura <andrewsh@debian.org>  Sat, 15 Dec 2018 14:22:18 +0100

[... 648 lines cut ...]
    copying cffi/pkgconfig.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/commontypes.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/api.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/lock.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/setuptools_ext.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/_cffi_include.h -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/parse_c_type.h -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/_embedding.h -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/_cffi_errors.h -> build/lib.linux-armv7l-3.6/cffi
    running build_ext
    building '_cffi_backend' extension
    creating build/temp.linux-armv7l-3.6
    creating build/temp.linux-armv7l-3.6/c
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/usr/include/ffi -I/usr/include/libffi -I/usr/local/include/python3.6m -c c/_cffi_backend.c -o build/temp.linux-armv7l-3.6/c/_cffi_backend.o
    unable to execute 'gcc': No such file or directory
    error: command 'gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/local/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-_fa39_s9/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-9unz3ez9/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-_fa39_s9/cffi/
ERROR: Service 'flask' failed to build: The command '/bin/sh -c pip install --no-cache-dir -r /home/mycodo/requirements.txt' returned a non-zero code: 1
Makefile:2: recipe for target 'build' failed
make: *** [build] Error 1
pi@CO2mon:~/Mycodo $ 

---- Your move, Kyle.  I'll leave this as it is and await better ideas to continue further.  Almost is pretty good in my book.  I hear that some other embedded applications are going to containers because they are also being built on linux and obscure isn't good enough for security anymore.  People around you are already driving those applications.  A Tesla isn't just a big battery, with wheels.

kizniche, 2019-03-07:
Looks like gcc is needed. I just recently took it out of the install process, but hadn't built on the Pi to verify it was still working. I'll push a fix shortly, but this should do it:

```sudo apt-get install gcc```

kizniche, 2019-03-07:
Sorry your first build experience was a flop. I pushed a fix for the issue building on Pis. I also updated my second comment with more detailed commands to shutdown any previously-installed Mycodo to free up ports and adding the user to the ```docker``` group. These should be all the commands possibly needed to get it up and running. I'm currently building on a Pi right now, so if I encounter any issues, I'll push another fix shortly.

Theoi-Meteoroi, 2019-03-09:
No worries Kyle, I still had a good time while trying.  I'll follow your suggestions and try some more.   Its worthwhile to do, I think.  Thank you for giving containers a go with Mycodo!

Theoi-Meteoroi, 2019-03-21:
Full Moon coming.  Guess it is time.

```
pi@everlast:~ $ sudo apt-get update
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian stretch/main armhf Packages [214 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]                                     
Get:5 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [44.5 kB]  
Fetched 12.0 MB in 1min 32s (129 kB/s)                                                                                   
Reading package lists... Done
pi@everlast:~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
WARNING: This update bumps to rpi-4.19.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=224931
##############################################################
Would you like to proceed? (y/N)

 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    423      0 --:--:-- --:--:-- --:--:--   424
100 58.4M  100 58.4M    0     0  3960k      0  0:00:15  0:00:15 --:--:-- 3268k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.19.29-v7+
 *** depmod 4.19.29+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 3f0ecd1fca450c2cacd429c96b78cc4be96acd51
 *** A reboot is needed to activate the new firmware
pi@everlast:~ $ 
< rebooted >

pi@everlast:~ $ sudo service mycodo stop
Failed to stop mycodo.service: Unit mycodo.service not loaded.
pi@everlast:~ $ sudo service mycodoflask stop
Failed to stop mycodoflask.service: Unit mycodoflask.service not loaded.
pi@everlast:~ $ sudo service nginx stop
pi@everlast:~ $ sudo service influxdb stop
pi@everlast:~ $ git clone https://github.com/kizniche/Mycodo
Cloning into 'Mycodo'...
remote: Enumerating objects: 22, done.
remote: Counting objects: 100% (22/22), done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 38960 (delta 10), reused 13 (delta 6), pack-reused 38938
Receiving objects: 100% (38960/38960), 71.42 MiB | 6.54 MiB/s, done.
Resolving deltas: 100% (31298/31298), done.
pi@everlast:~ $ cd Mycodo
pi@everlast:~/Mycodo $ git checkout mycodo_docker
Branch mycodo_docker set up to track remote branch mycodo_docker from origin.
Switched to a new branch 'mycodo_docker'
pi@everlast:~/Mycodo $ sudo /bin/bash ./mycodo.sh install-dependencies
#### Installing docker
# Executing docker install script, commit: cfba462
+ sh -c apt-get update -qq >/dev/null
+ sh -c apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
W: APT had planned for dpkg to do more than it reported back (18 vs 22).
   Affected packages: nginx-full:armhf
[... 873 lines cut ...]

```
Successfully built bbcfefa76198
Successfully tagged mycodo_nginx:latest
Creating influxdb ... done
Creating flask    ... done
Creating nginx    ... done
Creating daemon   ... done
pi@everlast:~/Mycodo $ 

And guess what I got.   Magic.

Gonna go explore now, however I have a few comments just for fun

Collecting *itsdangerous*>=0.24 (from Flask==1.0.2->-r /home/mycodo/requirements.txt (line 4))

That guy stole my best bad idea for a library.

The only misadventure was not adding 'sudo' to the 'make build' instruction.   I'm starting to make a habit of writing instructions to include the 'sudo' and I bold the rest of the command-line. This makes it easy to cut and paste for lazy, root-less folks such as myself.   I don't wanna be root.  Being root warps the perceptions of usage, usually in unfortunate ways.

I have a GUI.   Nice work Kyle!  

I logged in and looked around a little so far.  Will try some of the basic stuff and see what works.

I have this big smile right now.  That saying about how 'nothing good ever happens after midnight' came from someone who was doing it wrong.  At least that is my perspective.

Theoi-Meteoroi, 2019-03-21:
Well the first thing I tried was a USB connected sensor - the MH-z19.  I cannot change the port from hardware serial due to some access issue:

Error: Modify Input: Invalid device or improper permissions to read device

I tried changing perms on the device, that isn't the issue.  Its a container thing.   And don't I like a challenge.   Containers are attractive to me because as you become more ambitious with Mycodo, the platform becomes the limiting factor, even with four cores.  For straight forward PID tasks with a few sensors, the Pi is more than cost effective.  As the sensors become more sophisticated or you are simply ambitious ( voice control,  integration with other code ) containers could allow the kind of isolation that prevents total system failure when a feature doesn't work as expected.  Like the code I write.  

I will have to come back to this later. So much to do.  But it is a beginning and that is pretty exciting.  The OS can take this to a number of higher performance platforms if ported with that in mind.  There may be a way (I think there is) to push most of the work and storage to a cloud instance and have a lower end device like the Pi be the real-world proxy.   Once you program a PID, the work is local and the rest can be more or less done elsewhere.  This provides scale, if needed and still maintain a single pane of glass for complex configurations.  Anything built in the cloud can also be done on a local server with virtualization, if you are allergic to clouds.

kizniche, 2019-03-21:
Glad to see it's running! Here are a few notes before I begin my own investigation into getting external sensors working:

> The only misadventure was not adding 'sudo' to the 'make build' instruction. I'm starting to make a habit of writing instructions to include the 'sudo' and I bold the rest of the command-line. This [...]

After you issued the command ```sudo usermod -aG docker pi``` to add your user to the docker group, you must log out and back in for it to take effect. Then you can call ```docker``` without root privileges.

> I have this big smile right now. That saying about how 'nothing good ever happens after midnight' came from someone who was doing it wrong. At least that is my perspective.

:smiley: 

> Well the first thing I tried was a USB connected sensor - the MH-z19. I cannot change the port from hardware serial due to some access issue:
> Error: Modify Input: Invalid device or improper permissions to read device

I haven't attempted to do this yet, so I'll see how I can fix this. I have ```privileged: true``` in the ```docker-compose.yml``` file, which should give access to the system devices, but I haven't done much more than this to get external sensors working on the Mycodo.

In other news, today I'm implementing my recently-built outdoor LoRaWAN gateway (photo attached) on my university farm and sending sensor data from the surrounding greenhouses to The Things Network before being acquired by Mycodo (or multiple Mycodos). I got the [BME280 working on my TTGO T-Beam dev. platform](https://github.com/kizniche/ttgo-tbeam-sensor-node-bme280) last night and just recently added support for downloading data from The Things Network into Mycodo via a [TTN Input module](https://github.com/kizniche/Mycodo/blob/aaa94219811e268fe88776b70e292668a81602ce/mycodo/inputs/ttn_data_storage.py). I'm going to be working on a way to make this Input module able to have a variable number if measurements selected (opposed to the current hard-coded ones for just the BME280), then each measurement can have its own unit/measurement configured, so there won't be any limitation to the number or type of measurements.

[image]

kizniche, 2019-03-22:
The gateway was successfully deployed on the farm and one node is currently measuring temperature, humidity, pressure, dew point, and vapor pressure deficit in one of the hydroponic greenhouses. Now to see if the system remains stable over a paltry cellular internet connection. I've already set up the Mycodo on the farm to download the data from TTN, just as I have at home, and it's working perfectly so far.

Regarding docker, I found the missing configuration settings I needed to detect devices after the container has started. I just pushed an update and successfully tested the MH-Z16/UART over USB, connected after the container started!

kizniche, 2019-05-06:
I added Telegraf and Grafana to the docker build to experiment a bit with them. They were easily integrated, and add a new set of system monitoring and dashboard capability for Mycodo data. I'm not sure if they'll stay, but for the time being, they're included in the Docker build. I also fixed several issues with the dependency install script, moved everything to using Python 3, and successfully tested on a fresh install of Raspbian for everything to work.

kizniche, 2019-11-29:
I've come back to working on this and am currently trying to modify master rather than update the mycodo_docker branch, as it's very far behind at this point.

kizniche, 2019-11-30:
After much toiling, I have the latest version of Mycodo (8.1.1) running in Docker containers, without sacrificing its ability to be normally installed on a Raspberry Pi. I'll push my edits to master soon, after I tweak some of the volumes and directory structures. Here's it running on Ubuntu 18.04 (64-bit):

[image]

kizniche, 2019-11-30:
The mycodo_docker branch is now defunct (and will be removed soon), as I've been successful at integrating the docker code to master without sacrificing any normal functionality. The Docker configuration/build files can be found under the ```docker``` folder, with [docker/README.md](https://github.com/kizniche/Mycodo/blob/master/docker/README.md) containing all the current setup/build information.

There is a lot of functionality that doesn't work, but I'll be working on this gradually. This is a good step to bringing Mycodo available to other platforms.

### Containers running on a full build

```
kiz@kimchi:~/PycharmProjects/Mycodo/docker$ docker container ls
CONTAINER ID        IMAGE                    COMMAND                  CREATED             STATUS              PORTS                                                                    NAMES
22ea6edd7511        app                      "bash -c 'PYTHONPATH…"   15 minutes ago      Up 15 minutes       0.0.0.0:9090->9090/tcp                                                   mycodo_daemon
baa0b92e76f8        docker_nginx             "nginx -g 'daemon of…"   15 minutes ago      Up 15 minutes       0.0.0.0:80->80/tcp, 0.0.0.0:443->443/tcp                                 nginx
777b051a7079        grafana/grafana:latest   "/run.sh"                15 minutes ago      Up 15 minutes       0.0.0.0:3000->3000/tcp                                                   grafana
9a153480c81c        app                      "gunicorn --workers …"   15 minutes ago      Up 15 minutes                                                                                mycodo_flask
27bc125d1cf8        telegraf:latest          "/entrypoint.sh tele…"   15 minutes ago      Up 15 minutes       8092/udp, 8125/udp, 8094/tcp                                             telegraf
075d0552cc63        docker_influxdb          "/entrypoint.sh /bin…"   16 minutes ago      Up 15 minutes       0.0.0.0:8083->8083/tcp, 0.0.0.0:8086->8086/tcp, 0.0.0.0:8090->8090/tcp   influxdb
```

### Grafana/Telegraf (only part of full stats available)

[image]

Theoi-Meteoroi, 2020-01-29:
Thanks Kyle!  I'm going to spend some time looking at how this might work on a Mac with dockerdesktop.  I'm just looking for serial port inputs right now and maybe a "sensor pod" on a FTDI-4232 with python code input to Mycodo.   I can do the initial work on the standard release on a pi and then try it on a laptop with containers.  USB-C changes what you can attach to a laptop (for instrumentation) and power has always been my biggest challenge.  Hang enough sensors on a Pi and the power gets flaky with most wall-wart supplies.   And some sensors I want to use ( AS7265x spectral sensor ) just crush the CPU on a pi with the matlab code to generate the spectral chart.   
I think I whined earlier about Grafana - saying Mycodo saved me from having to do that.  I'm using Grafana in my day job now for visualizing prometheus data.   And so now, I'm doing that.

nargetdev, 2022-01-14:
I'm just beginning my journey towards Dockerized mycodo.  I'm very keen to port to balenaOS for my RPi's but I'm shooting towards `docker-compose` on my Ubuntu machine first.

I'm specifically having trouble building `mycodo_flask`.  I think it's an environment thing, it seems to be that `pip` is missing from mycodo's version of python?

        "${MYCODO_PATH}"/env/bin/python -m pip install --upgrade pip
                                                                           ^
                                                                           |
                                                                      missing

Why does mycodo manage it's own binaries?  Vs venv or the like?

Anyways I was able to get mycodo_flask to build by commenting some of the following, but then get runtime failure.
```
RUN /home/mycodo/mycodo/scripts/upgrade_commands.sh create-files-directories
RUN /home/mycodo/mycodo/scripts/upgrade_commands.sh update-apt
RUN /home/mycodo/mycodo/scripts/upgrade_commands.sh update-packages
RUN /home/mycodo/mycodo/scripts/upgrade_commands.sh install-docker-ce-cli
#RUN /home/mycodo/mycodo/scripts/upgrade_commands.sh docker-update-pip
#RUN /home/mycodo/mycodo/scripts/upgrade_commands.sh docker-update-pip-packages
RUN /home/mycodo/mycodo/scripts/upgrade_commands.sh ssl-certs-generate
#RUN /home/mycodo/mycodo/scripts/upgrade_commands.sh docker-compile-translations
RUN /home/mycodo/mycodo/scripts/upgrade_commands.sh compile-mycodo-wrapper
```

kizniche, 2022-01-14:
@nargetdev You should use the code from the master branch, not the latest release. You're currently using old docker code. Try the mast branch and it should work.

nargetdev, 2022-01-14:
Ah.. forgot to `git pull`

kizniche, 2022-01-14:
I made a bunch of docker changes since the last release: bd38694a56fc81dcda1bacbb74b2337942847440, 11e4114cd55e1ea7e20243aa7cdd6c15847f4f9b, and d2f9dbcb28de3d2cd42a5a50938153e59f5ebd04. So thre should be significant improvements with building and more recent software versions.

nargetdev, 2022-01-15:
Yes it worked this time.  Nice work Kyle.

I'll stay posted regarding balena.  Should be relatively straight forward.
A few minor structural changes needed - `docker-compose.yml` needs to be in the root of the repo, and all bind mounts (ref to explicit host path) needs removed.

nargetdev, 2022-01-15:
For what it's worth I managed to get the following fork to successfully upload the docker images to the RPi via `balena push ...` but there's a myriad of runtime issues that crop up, probably because I ripped out all the bind mounts from the `docker-compose.yml`

https://github.com/nargetdev/balena-mycodo

More to come as I dig in further..

jamie-mint, 2022-07-19:
I'm getting errors with the host networking in the compose context.  (because compose renames network hostnames)

the below example is illustrative:
```
[mycodo_telegraf] 2022-07-19T00:19:47Z E! [outputs.influxdb] When writing to [http://localhost:8086]: failed doing req: Post "http://localhost:8086/write?db=telegraf": dial tcp 127.0.0.1:8086: connect: connection refused
```

It should be found at:
mycodo_influxdb:8086

I'm wondering as you'd said previously @kizniche you don't want to break compatibility with the "normal" build.  Without making a ton of defines/exceptions for every single mention of localhost/127.0.0.1 towards microservices names in docker-compose context what do you think is the path of least resistance here?

kizniche, 2022-07-19:
Fortunately just a few weeks ago I added the ability to configure influxdb settings in the Mycodo database. This includes the ability to set the hostname to connect to influxdb I haven't tested it, but I suspect if you change it from "localhost" to "mycodo_influxdb", it should work. You'll need to install from the master branch, because this hasn't been released yet.

jamie-mint, 2022-07-19:
Brilliant, yes I was just discovering this.

Another piece of the equation is that my embedded docker framework/platform (balenaOS) doesn't accept a parent dir context (i.e. `context: ../`).  Just converted some paths hopefully no clobber.  I'll report back when my build is running again.

P.S. Nice job moving on influxdb 2.x too.

nargetdev, 2022-07-19:
No joy changing hostname to `mycodo_influxdb`

I think the problem might be more fundamental though..

```
'last_data' raised and error: 'Misc' object has no attribute 'measurement_db_name'
```

nargetdev, 2022-07-19:
I can see this code is intertwined with the influxdb v1 and v2 logic switches.  I think i'm going to take a step back and explore influxdb outside of the mycodo context (simple mock data) get my bearings and then come back in

kizniche, 2022-07-19:
I just pushed a potential fix for the undeclared variable issue

jamie-mint, 2022-07-22:
I'm able to docker-compose successfully on `influx2` branch but I think there's a new issue on `master`, as `mycodo_flask` crashes now on initialization:

```
Traceback (most recent call last):
  File "/home/mycodo/env/lib/python3.9/site-packages/gunicorn/arbiter.py", line 589, in spawn_worker
    worker.init_process()
  File "/home/mycodo/env/lib/python3.9/site-packages/gunicorn/workers/base.py", line 134, in init_process
    self.load_wsgi()
  File "/home/mycodo/env/lib/python3.9/site-packages/gunicorn/workers/base.py", line 146, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/mycodo/env/lib/python3.9/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/mycodo/env/lib/python3.9/site-packages/gunicorn/app/wsgiapp.py", line 58, in load
    return self.load_wsgiapp()
  File "/home/mycodo/env/lib/python3.9/site-packages/gunicorn/app/wsgiapp.py", line 48, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/mycodo/env/lib/python3.9/site-packages/gunicorn/util.py", line 359, in import_app
    mod = importlib.import_module(module)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/mycodo/mycodo/start_flask_ui.py", line 13, in <module>
    app = create_app()  # required by the wsgi config and main()
  File "/home/mycodo/mycodo/mycodo_flask/app.py", line 61, in create_app
    register_extensions(app)
  File "/home/mycodo/mycodo/mycodo_flask/app.py", line 86, in register_extensions
    populate_db()
  File "/home/mycodo/mycodo/databases/models/__init__.py", line 147, in populate_db
    if not Misc.query.count():
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/orm/query.py", line 3163, in count
    return self._from_self(col).enable_eagerloads(False).scalar()
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/orm/query.py", line 2888, in scalar
    ret = self.one()
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/orm/query.py", line 2865, in one
    return self._iter().one()
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/orm/query.py", line 2903, in _iter
    result = self.session.execute(
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1692, in execute
    result = conn._execute_20(statement, params or {}, execution_options)
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1631, in _execute_20
    return meth(self, args_10style, kwargs_10style, execution_options)
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/sql/elements.py", line 325, in _execute_on_connection
    return connection._execute_clauseelement(
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1498, in _execute_clauseelement
    ret = self._execute_context(
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1862, in _execute_context
    self._handle_dbapi_exception(
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 2043, in _handle_dbapi_exception
    util.raise_(
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/util/compat.py", line 207, in raise_
    raise exception
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1819, in _execute_context
    self.dialect.do_execute(
  File "/home/mycodo/env/lib/python3.9/site-packages/sqlalchemy/engine/default.py", line 732, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: misc.measurement_db_name
[SQL: SELECT count(*) AS count_1
FROM (SELECT misc.id AS misc_id, misc.dismiss_notification AS misc_dismiss_notification, misc.force_https AS misc_force_https, misc.hide_alert_info AS misc_hide_alert_info, misc.hide_alert_success AS misc_hide_alert_success, misc.hide_alert_warning AS misc_hide_ale
rt_warning, misc.hide_tooltips AS misc_hide_tooltips, misc.grid_cell_height AS misc_grid_cell_height, misc.login_message AS misc_login_message, misc.max_amps AS misc_max_amps, misc.output_usage_cost AS misc_output_usage_cost, misc.output_usage_currency AS misc_outp
ut_usage_currency, misc.output_usage_dayofmonth AS misc_output_usage_dayofmonth, misc.output_usage_volts AS misc_output_usage_volts, misc.output_usage_report_gen AS misc_output_usage_report_gen, misc.output_usage_report_span AS misc_output_usage_report_span, misc.o
utput_usage_report_day AS misc_output_usage_report_day, misc.output_usage_report_hour AS misc_output_usage_report_hour, misc.sample_rate_controller_conditional AS misc_sample_rate_controller_conditional, misc.sample_rate_controller_function AS misc_sample_rate_cont
roller_function, misc.sample_rate_controller_input AS misc_sample_rate_controller_input, misc.sample_rate_controller_math AS misc_sample_rate_controller_math, misc.sample_rate_controller_output AS misc_sample_rate_controller_output, misc.sample_ra [... cut ...] er AS mis
c_measurement_db_user, misc.measurement_db_password AS misc_measurement_db_password, misc.measurement_db_dbname AS misc_measurement_db_dbname
FROM misc) AS anon_1]
(Background on this error at: https://sqlalche.me/e/14/e3q8)
[2022-07-22 13:29:01 -0600] [7] [INFO] Worker exiting (pid: 7)
Error: Command 'influxd version' returned non-zero exit status 127.
[2022-07-22 13:29:01 -0600] [1] [INFO] Shutting down: Master
[2022-07-22 13:29:01 -0600] [1] [INFO] Reason: Worker failed to boot.
```

jamie-mint, 2022-07-22:
I would imagine this is because your volume is already initialized with some data?

kizniche, 2022-07-22:
>I'm able to docker-compose successfully on influx2 branch

The influx2 branch is very old (currently 25 commits behind master, which includes several fixes for docker). The influx2 branch changes have also already been incorporated into the master branch and it needs to be deleted. So, all testing should be performed on the master branch.

kizniche, 2022-07-22:
>I think there's a new issue on master, as mycodo_flask crashes now on initialization

This error should not exist on master, as I fixed this undeclared variable with https://github.com/kizniche/Mycodo/commit/0fc6105b142b35e5b39759fbee067d252921f1cd

Is your repo clone up to date with this latest commit?

jamie-mint, 2022-07-22:
Yea these are from as sanitized an environment as I could conjur (at `0fc6105b142b35e5b39759fbee067d252921f1cd`)

One of the things I saw in the logs from mycodo_flask
`/bin/sh: 1: influxd: not found`

jamie-mint, 2022-07-22:
I'm building on both Pop!_OS (basically ubuntu) and also raspbian.

`uname -a` # Pop!_OS 
Linux system76 5.15.8-76051508-generic #202112141040~1639505278~21.10~0ede46a SMP Tue Dec 14 22:38:29 U x86_64 x86_64 x86_64 GNU/Linux

`uname -a` # raspbian 
Linux planck64 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux

I would like to get things running locally on my macbook as well.. but that's last in line.

kizniche, 2022-07-22:
I think I fixed the issues. I now have docker building and running Mycodo without issue, on the master branch.

```bash
git clone https://github.com/kizniche/Mycodo
cd Mycodo/docker
sudo /bin/bash ./setup.sh dependencies
sudo make build
```

- Go to https://PI_IP_Address/
- Create admin user
- Login with user
- Go to [gear icon] -> Configure -> Time Series Database Options and change Hostname from "localhost" to "mycodo_influxdb", save
- Go to Input page, add "CPU Load" Input, activate
- Go to Live page
- See current CPU load data

jamie-mint, 2022-07-22:
No joy.  Without copying the whole traceback I'm still in a similar spot on Pop!_OS and Debian (RPi)

`Error: cannot import name 'Misc' from partially initialized module 'mycodo.databases.models' (most likely due to a circular import) (/home/mycodo/mycodo/databases/models/__init__.py)`

Perhaps it's the leftover data in the volumes.  Trying to delete the volumes now and see..

jamie-mint, 2022-07-22:
Yes!  It was the volumes..

WARNING - this next command can and will destroy data.  Measure twice.. cut once.

`docker volume rm $(docker volume ls -q)`

From a clean state (no prior volumes), and after going `Configure` --> `Time Series Database Options` and set Hostname: `mycodo_influxdb`

We are correctly writing metrics to the database.  I tested only my co2 sensor (FTDI) so far but it's working correctly.

Thankyou @kizniche

nargetdev, 2022-07-26:
I managed to get i2c working.  I created a pull request so you can check it out, not really to merge but i'm pretty stoked about one click deploy to any (or many) balenaOS devices simultaneously with single pane of glass management.

nargetdev, 2022-07-26:
That PR i mentioned requires the python virtual environment (not included) which I had setup on my other RPi.  So that should be packaged in..

So far GPIO (normal read/write) works ok (using the `/sys/class/gpio/<pin_num>/value`)

PWM I was only able to get working through the RPi.GPIO python module

nargetdev, 2022-07-26:
This is exciting though.. it's enough to switch over to docker based builds for me.

nargetdev, 2022-07-29:
Is any initialization required for a new influxdb2.3 database?

i.e. can the influxdb Dockerfile simply be amended as:

`FROM influxdb:2.3`

[... rest of file ...]

nargetdev, 2022-07-29:
I'm getting unauthorized access as follows and I'm wondering if it's because connections are not coming from `localhost`, but rather `mycodo_daemon` hostname

`ts=2022-07-29T22:12:26.941630Z lvl=error msg="api error encountered" log_id=0bz~YeIW000 handler=session error="unauthorized access"`

kizniche, 2022-07-29:
The docker influxdb install script [run.sh](https://github.com/kizniche/Mycodo/blob/master/docker/influxdb/run.sh) was only designed for influxdb 1.x, and since 1.x and 2.x have incompatible commands, the commands used to create the db/user/password likely don't work with 2.x.

kizniche, 2022-07-29:
See here for the difference between the two:

https://github.com/kizniche/Mycodo/blob/eb1b7c6b405a80bd2bbca0ccbabb31f485264281/mycodo/scripts/upgrade_commands.sh#L428-L465

nargetdev, 2022-07-30:
Yes I just found the install commands that are invoked on the upstream (non-docker) `install/setup.sh` script

```
if [[ ${INFLUX} == '1)' ]]; then
    ${INSTALL_CMD} update-influxdb-1 2>&1 | tee -a "${LOG_LOCATION}"
    ${INSTALL_CMD} update-influxdb-1-db-user 2>&1 | tee -a "${LOG_LOCATION}"
elif [[ ${INFLUX} == '2)' ]]; then
    ${INSTALL_CMD} update-influxdb-2 2>&1 | tee -a "${LOG_LOCATION}"
    ${INSTALL_CMD} update-influxdb-2-db-user 2>&1 | tee -a "${LOG_LOCATION}"
```

I tested this out (outside of docker), but i still somehow end up with an influxdb1.8 install

Anyways, no worries.  In due time I may take a look at porting influxdb2.x to the docker side, but ver 1.x is fine for now.

kizniche, 2022-08-26:
>one click deploy to any (or many) balenaOS devices simultaneously with single pane of glass management.

I'm not familiar with this. Can you give give a breakdown of how this works or how to do it?

>balenaOS doesn't accept a parent dir context (i.e. context: ../)

Why is this? Is the only solution to put the docker root directory (or just docker-compose.yaml) as the root of the Mycodo repository? I guess this wouldn't be too much of an issue if all the other directories were placed inside a docker/ directory. The directory structure of [nargetdev/balena-mycodo](https://github.com/nargetdev/balena-mycodo) currently has all these docker directories in the root of Mycodo. Could these be moved to docker/ and still work? If so, I'd be more open to merging the pull request since it keeps the docker and non-docker potions of Mycodo more segregated. Docker is still an experimental feature that needs some improvement, so is still a secondary feature at this point.

nargetdev, 2022-11-03:
Hey there.  So for the time being I've actually graduated to a [lightweight Kubernetes (k3s)](https://k3s.io/) based deployment to manage my RPis and I'm finding it to be delicious, albeit with a learning curve (k8s manifests are much more verbose than docker-compose.yml)

k8s is not opinionated about a specific file structure in the way that BalenaOS mandates - so it's definitely possible now without mutilating the structure / mucking up bare metal compatibility.

My workflow for my own services has generally involved a build command like the following:

```
docker buildx build --push \
            --platform linux/arm64 \
            --tag nargetdev/cam-gbucket-dgraph:0.2.1   .
```
... in order to build cross platform image for RPi from my Macbook (very zippy build)

My motivation for switching to k3s from balena initially was their 10 device limit, but since then I've realized quite a few advantages that come packaged with k8s ecosystem in terms of observability, and extensive tooling of the mainstream de-facto standard for container management.

P.S. regarding BalenaOS / BalenaCloud - best way to grok I think is to just make an account at https://www.balena.io/ and flash some RPis with their OS and push some containers to the runtime (they have tons of tutorials).  It's a very nice docker distro with bells and whistles (like a filesystem watcher that automatically rebuilds and pushes images to your devices as you edit source), but they do have a 10 device limit for their cloud PaaS - at which point you could just run your own open source instance, but I haven't tried, as I'm all in on k3s for now.

nargetdev, 2022-11-04:
Ah.. one gotcha - it seems the `cgroup_memory=1 cgroup_enable=memory` which is required in `/boot/cmdline.txt` for k3s mucks up the nginx or flask or something with the (bare metal) frontend.  So normal Mycodo on the host or Mycodo via docker would have to be selected at boot.. but as long as one is aware that they are mutually exclusive at runtime I don't think it's a problem

nargetdev, 2022-11-09:
@kizniche do you have a map for which volume mounts are populated on initial creation by the file system of the container? I didn't realize how Docker handles this until just now (I.e. initialized volume will clobber Docker file system, vs empty volume will assume Docker file system). I haven't looked at this closely yet, but I noticed it when containers were crashing because /home/mycodo/env/python was getting clobbered. The python executable for example should be in a RO location. What is needing to be persisted across containers and for what reasons? I'd like to limit or eliminate the possibility of clobber / runtime inconsistency due to initialized / uninitialized volumes.

Cheers. To containers.

kizniche, 2022-11-10:
Are you referring to these?

https://github.com/kizniche/Mycodo/blob/9b42db8daeb7d035f6f1e06d2af4fc2a73dee10b/docker/docker-compose.yml#L38-L48

kizniche, 2022-11-16:
@nargetdev could you please provide some more information about building/running on kubernetes? I'd like to explore this option more, but my experience is virtually all with Docker, so I could use some help getting this going and want to avoid fumbling around and hitting common issues that can be avoided with some guidance. Thanks.

jonny-sexton, 2022-11-20:
Hi,

I am having a few issues running Mycodo on my Raspberry Pi Zero via docker. I have run the following commands successfully:
```
git clone https://github.com/kizniche/Mycodo
cd Mycodo/docker
sudo /bin/bash ./setup.sh dependencies
```

However, when I then ran `sudo make build` I got the following error:
```
The requested image's platform (linux/arm/v7) does not match the detected host platform (linux/arm/v6) and no specific platform was requested
```
Which I fixed by repacing `FROM influxdb:1.8.10` with `FROM mendhak/arm32v6-influxdb` in `/Mycodo/docker/influxdb/Dockerfile`

I now have another error that I am not able to resolve:
```
ERROR: Service 'mycodo_flask' failed to build: The command '/bin/sh -c /home/mycodo/mycodo/scripts/upgrade_commands.sh install-docker-ce-cli' returned a non-zero code: 1
```

I checked and the file `/home/mycodo/mycodo/scripts/upgrade_commands.sh` does not exist. It appears that the line:
```
COPY ./mycodo/scripts/upgrade_commands.sh /home/mycodo/mycodo/scripts/upgrade_commands.sh
``` 
in `Mycodo/docker/Dockerfile` is not correct, as it should be:
```
COPY ../mycodo/scripts/upgrade_commands.sh /home/mycodo/mycodo/scripts/upgrade_commands.sh
```
instead. I have of coursed tried correcting this, running `sudo make build -C docker` from the parent directory instead, which should fix the relevant path issue. I also tried copying the files manually. However, I am still not able to correct this issue (I always end up having the same error message).

I will try the standard installation method instead now, which should be sufficient for my purposes. I just thought I would let you know in case I am doing anything wrong here.

Looking forward to working with Mycodo!

Cheers,

Jonny

kizniche, 2022-11-20:
I've actually been working on Docker all day today. The branch I'm working on is https://github.com/kizniche/Mycodo/tree/docker-01 but I have several changes still uncommitted. I just got everything to build nicely on a Pi 4 and will be pushing changes soon, then merging into master shortly after. You may need to perform your change still for 32-bit ARM devices like your Pi, unless we can figure out a way to detect this automatically.

jonny-sexton, 2022-11-20:
Brilliant, thanks for the speedy response! I retried the steps outlined above on the docker-01 branch, and the orginial issue is solved. However I now have a new error message:
```
ERROR: Service 'mycodo_daemon' failed to build: The command '/bin/sh -c /home/mycodo/mycodo/scripts/upgrade_commands.sh install-docker' returned a non-zero code: 100
```
I had a quick look, but tbh I don't understand where the issue lies.

jonny-sexton, 2022-11-20:
> unless we can figure out a way to detect this automatically.
You can detect which chip architecture a linux machine has with the `arch` command. On the pi zero I get the following:
```
armv6l
```
On the pi 4 you should get 
```
aarch64
```
I believe.

kizniche, 2022-11-20:
I just committed (and merged into master) my final changes for Docker. I haven't tested on a Pi Zero.

>I now have a new error message: ERROR: Service 'mycodo_daemon' failed to build...

Is this all there was in the log? No other errors?

jonny-sexton, 2022-11-21:
OK, I will try again then with the master branch. I ran the docker-compose up command with the --verbose option and here are the last few lines of code before the error appears:
```
compose.cli.verbose_proxy.proxy_callable: docker build -> <generator object APIClient._stream_helper at 0xb4849530>
compose.cli.verbose_proxy.proxy_callable: docker close <- ()
compose.cli.verbose_proxy.proxy_callable: docker close -> None
compose.service.build: Building mycodo_daemon
compose.cli.verbose_proxy.proxy_callable: docker build <- (path='/home/Jonny/Mycodo', tag='app', rm=True, forcerm=False, pull=False, nocache=False, dockerfile='docker/Dockerfile', cache_from=None, labels=None, buildargs={}, network_mode=None, target=None, shmsize=None, extra_hosts=None, container_limits={'memory': None}, gzip=False, isolation=None, platform=None)
docker.api.build._set_auth_headers: Looking for auth config
docker.api.build._set_auth_headers: No auth config in memory - loading from filesystem
docker.utils.config.find_config_file: Trying paths: ['/home/Jonny/.docker/config.json', '/home/Jonny/.dockercfg']
docker.utils.config.find_config_file: No config file found
docker.api.build._set_auth_headers: Sending auth config ()
urllib3.connectionpool._make_request: http://localhost:None "POST /v1.38/build?t=app&q=False&nocache=False&rm=True&forcerm=False&pull=False&dockerfile=docker%2FDockerfile HTTP/1.1" 200 None
compose.cli.verbose_proxy.proxy_callable: docker build -> <generator object APIClient._stream_helper at 0xb48e4e28>
ERROR: compose.cli.main.main: Service 'mycodo_daemon' failed to build: The command '/bin/sh -c /home/mycodo/mycodo/scripts/upgrade_commands.sh install-docker' returned a non-zero code: 100
```

kizniche, 2022-11-21:
This command starts with printf, so for that to not be in the logs indicates an unusual issue.

https://github.com/kizniche/Mycodo/blob/2c200363c85c8c3988bdc8c576faf90c85419b6e/mycodo/scripts/upgrade_commands.sh#L711-L715

jonny-sexton, 2022-11-21:
I just tested with the main branch again, but still no dice:
```
Building mycodo_influxdb
Building mycodo_flask
ERROR: Service 'mycodo_flask' failed to build: The command '/bin/sh -c /home/mycodo/mycodo/scripts/upgrade_commands.sh install-docker' returned a non-zero code: 100
```

Yes that's what I thought. I'll take a closer look when I have more time, will probably be this weekend though. I'll update you if I find anything.

kizniche, 2022-11-21:
I just did a full test on a freshly installed Ubuntu 20.04 virtual machine. I changed the Linux image used and updated the build commands. I was able to build and run without issue.

jonny-sexton, 2022-12-06:
Hi @kizniche, sorry for the late reply. I still couldn't get this to work, and I have decided to write my own code for my needs.

However I think the crux of the issue was that my raspberry pi zero w (1st generation) supported 32-bit OS only, and Docker does not support 32 bit OS (I had to install Docker using unofficial 3rd party software: https://gitlab.com/docker-32bit). This caused all kinds of issues trying to fix it, so I think you should maybe add in the README that 32-bit raspberry pis are not supported for the docker installation of Mycodo.

nargetdev, 2023-01-16:
architecture heterogeneity can be addressed with `buildx` as follows..

```
#! /bin/bash
docker buildx build \
            . \
            --push \
            --platform linux/arm64, linux/arm/v7, linux/arm/v6 \
            --tag nargetdev/infinilapse:1.1.2
```

[10 more replies]