kizniche/Mycodo #638: New input: Ruuvitag BT Sensor

RynFlutsch, 2019-03-08:
Maybe a new bluetooth Sensor with
Temperature, Humidity, Pressure and Motion.
Uses Bosch BME280

https://github.com/ruuvi
https://ruuvi.com/ruuvitag-specs/

kizniche, 2019-03-27:
Do you have one of these devices?

RynFlutsch, 2019-03-27:
No not yet, but i can send you some bucks to buy one or i buy one and we work together ?

RynFlutsch, 2019-03-31:
I've ordered three.
I would like to give you one.

kizniche, 2019-04-01:
Thanks. You can contact me at kylegabriel.com/contact

kizniche, 2019-04-14:
I got the sensor you sent and I pushed a commit yesterday that added a RuuviTag Input module. I've been dealing with being ill, so I haven't been able to mess with it much since then. I've noticed there's a memory leak in the Python library ([ttu/ruuvitag-sensor](https://github.com/ttu/ruuvitag-sensor)) causing usage to skyrocket only a few hours after activating the input and causing instability issues (several inputs stop working, system became unresponsive once). I'm not investigating how to fix this memory issue.

kizniche, 2019-04-16:
I think I fixed the memory leak issue. With my very brief testing, it appears if I execute a python script that utilizes the ruuvitag-sesnor module through subprocess.Popen rather than running the ruuvitag-sensor module within the Mycodo Input module, memory doesn't exponentially increase. I've attached a graph of Mycodo's ran usage after several restarts after it became unresponsive, and the last time I restarted with the new module utilizing subprocess.

[image]

kizniche, 2019-04-16:
However, this doesn't mean there aren't now memory issues outside Mycodo. I'll monitor using ```free``` and report back. I'll commit my work once I've verified it to work and make a release so you can test.

RynFlutsch, 2019-04-17:
I can't upgrade. You currently have the latest release. 
Why not change to 7.4.3 ?

kizniche, 2019-04-17:
I haven't released 7.4.3. There are more than just this change and I'm not going to release until I've had a chance to thoroughly test the changes so I know it's not going to break anyone's system. I'm also still dealing my my illness.

kizniche, 2019-04-18:
This should be the final fix. I now have both my SHT31 Smart Gadget and the Ruuvitag both working in harmony and no memory issue.

RynFlutsch, 2019-04-21:
Ruuvitag is working ;-) Thank you, i hope you are fine again.
You know the limit for connect Ruuvitags to Mycodo?
You know some tips for the SHT 31 Gadget settings?
I would use the meassured data for PID.
Period 1800s and Logging Intervall 60s show no data on live page.

RynFlutsch, 2019-04-21:
I have some issues. 
3x Ruuvitag is working, but only without SHT31-Gadget.

RynFlutsch, 2019-04-21:
If i turn the SHT31 Gadget on, all 3 Ruuvis report on Live Page: no Date Last x seconds
`2019-04-21 10:38:57,788 - mycodo.ruuvitag_e9708b22 - ERROR - InputModule raised an exception when taking a reading: I/O operation on closed file
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 117, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/ruuvitag.py", line 182, in get_measurement
    lock.release()
  File "/var/mycodo-root/env/lib/python3.5/site-packages/locket/__init__.py", line 110, in release
    lock.release()
  File "/var/mycodo-root/env/lib/python3.5/site-packages/locket/__init__.py", line 164, in release
    _unlock_file(self._file)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/locket/__init__.py", line 46, in _unlock_file
    fcntl.flock(file_.fileno(), fcntl.LOCK_UN)
ValueError: I/O operation on closed file
2019-04-21 13:52:03,791 - mycodo.input_a93dafb2 - INFO - Activated in 386.0 ms
2019-04-21 13:52:31,445 - mycodo.sht31_smart_gadget_a93dafb2 - INFO - Sensirion AG, SmartGadget, SN: 00000000, Name: Smart Humigadget, Firmware: 1.3, Hardware: 1, Software: 0.1, Log Interval: 60.0 sec`

kizniche, 2019-04-21:
The issue is you cannot read from more than once device on the same Bluetooth device at the same time. Consequently, you have to wait for one to finish before another can start. Currently, when the SHT31SG input starts, it downloads all the data, which depending on the log interval, can be a considerable amount of time. I currently have a lock file keep assurance only one device is accessing the BT device, that operates on a timeout of 600 seconds (10 minutes). This worked with my setup, as my SHT31SG interval is 10 minutes, causing a 5-minute download at activation. I believe your SHT31SG takes longer than 10 minutes to download data, causing your RT inputs to break the lock and access the BT, which causes an error for everything. You can verify this by only activating your SHT31SG, letting it wait (20 or 30 minutes), then activating your RTs.

kizniche, 2019-04-21:
I will experiment with storing the SHT31SG timestamp to reduce download time on initial activation (only if it recently performed a data download). I can also increase the lock file timeout to something incredibly large like 30 minutes. However, with multiple RTs, you can easily exceed a 30-minute wait if the logging intervals are exceptionally short (1 minute).

kizniche, 2019-05-17:
I just released v7.5.3. Can you see if you're still experiencing any of these issues with this release?

RynFlutsch, 2019-05-19:
On 7.4.3 My Ruuvis were working.
I Tried yesterday with 7.5.3
Live Page show no data.
Slightly older Ruuvi metrics appear in the dashboard.
SHT31 i will investigate.

RynFlutsch, 2019-05-24:
Hi Kyle, sorry busy...
Error Log:
`2019-05-24 18:07:32,281 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - InputModule.get_measurement() method raised IOError: [Errno 2] No such file or directory: '/var/lock/bluetooth_dev_hci0'
2019-05-24 18:16:30,122 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not acquire lock. Breaking for future locking.
2019-05-24 18:16:30,124 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - InputModule.get_measurement() method raised IOError: [Errno 2] No such file or directory: '/var/lock/bluetooth_dev_hci0'
2019-05-24 18:18:13,198 - ERROR - mycodo.inputs.ruuvitag_6adb7a42 - Could not acquire lock. Breaking for future locking.
2019-05-24 18:18:13,199 - ERROR - mycodo.inputs.ruuvitag_6adb7a42 - InputModule.get_measurement() method raised IOError: [Errno 2] No such file or directory: '/var/lock/bluetooth_dev_hci0'`

RynFlutsch, 2019-05-28:
Daemon Error.log:
`2019-05-28 21:19:09,031 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not connect: Device disconnected
2019-05-28 21:19:22,856 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - InputModule.get_measurement() method raised IOError: [Errno 32] Broken pipe
2019-05-28 21:39:20,272 - ERROR - mycodo.inputs.ruuvitag_6adb7a42 - Could not acquire lock. Breaking for future locking.
2019-05-28 21:39:20,274 - ERROR - mycodo.inputs.ruuvitag_6adb7a42 - InputModule.get_measurement() method raised IOError: [Errno 2] No such file or directory: '/var/lock/bluetooth_dev_hci0'
2019-05-28 21:39:49,004 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not connect: Failed to connect to peripheral C9:8E:20:E7:7D:0A, addr type: random
2019-05-28 21:40:17,041 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not connect: Device disconnected
2019-05-28 21:40:43,048 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not connect: Device disconnected
2019-05-28 21:41:05,122 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not connect: Failed to connect to peripheral C9:8E:20:E7:7D:0A, addr type: random
2019-05-28 21:41:35,127 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not connect: Failed to connect to peripheral C9:8E:20:E7:7D:0A, addr type: random
2019-05-28 21:41:57,754 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not connect: Device disconnected
2019-05-28 21:42:13,429 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not connect: Failed to connect to peripheral C9:8E:20:E7:7D:0A, addr type: random
2019-05-28 21:42:53,439 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not connect: Device disconnected
2019-05-28 21:43:03,640 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - InputModule.get_measurement() method raised IOError: [Errno 32] Broken pipe
2019-05-28 22:03:20,297 - ERROR - mycodo.inputs.ruuvitag_6adb7a42 - Could not acquire lock. Breaking for future locking.
2019-05-28 22:03:20,298 - ERROR - mycodo.inputs.ruuvitag_6adb7a42 - InputModule.get_measurement() method raised IOError: [Errno 2] No such file or directory: '/var/lock/bluetooth_dev_hci0'
2019-05-28 22:03:38,517 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - Could not connect: Failed to connect to peripheral C9:8E:20:E7:7D:0A, addr type: random
2019-05-28 22:04:18,362 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - InputModule.get_measurement() method raised IOError: [Errno 32] Broken pipe
2019-05-28 22:23:50,262 - ERROR - mycodo.inputs.ruuvitag_6adb7a42 - Could not acquire lock. Breaking for future locking.
2019-05-28 22:23:50,264 - ERROR - mycodo.inputs.ruuvitag_6adb7a42 - InputModule.get_measurement() method raised IOError: [Errno 2] No such file or directory: '/var/lock/bluetooth_dev_hci0'
2019-05-28 22:40:48,499 - ERROR - mycodo.controller_output - Cannot turn off Output 3996dfdb-7bd2-4410-bd86-da74ee06ec4d: Output not set up properly.
2019-05-28 22:40:48,499 - ERROR - mycodo.controller_output - Cannot turn off Output 9bb9fe5b-4fa4-4e9e-9c16-634232fb4e46: Output not set up properly.
2019-05-28 22:40:48,500 - ERROR - mycodo.controller_output - Cannot turn off Output 8464eeca-90c3-4477-84f3-e275f27630e1: Output not set up properly.
2019-05-28 22:40:48,514 - ERROR - mycodo.controller_output - Cannot turn off Output d0c0efc9-b5bc-4366-9870-d2ca46f94c6a: Output not set up properly.
2019-05-28 22:41:08,029 - ERROR - mycodo.controller_output - Cannot turn off Output 9bb9fe5b-4fa4-4e9e-9c16-634232fb4e46: Output not set up properly.
2019-05-28 22:41:08,030 - ERROR - mycodo.controller_output - Cannot turn off Output d0c0efc9-b5bc-4366-9870-d2ca46f94c6a: Output not set up properly.
2019-05-28 22:41:08,031 - ERROR - mycodo.controller_output - Cannot turn off Output 8464eeca-90c3-4477-84f3-e275f27630e1: Output not set up properly.
2019-05-28 22:41:08,031 - ERROR - mycodo.controller_output - Cannot turn off Output 3996dfdb-7bd2-4410-bd86-da74ee06ec4d: Output not set up properly.
2019-05-28 22:46:28,747 - ERROR - mycodo.inputs.sht31_smart_gadget_b8723207 - Could not connect: Failed to connect to peripheral FD:E1:1B:A9:D4:8E, addr type: random
2019-05-28 22:46:28,748 - ERROR - mycodo.inputs.sht31_smart_gadget_b8723207 - Could not connect: Failed to connect to peripheral FD:E1:1B:A9:D4:8E, addr type: random
2019-05-28 22:46:33,169 - ERROR - mycodo.inputs.sht31_smart_gadget_27dc0994 - InputModule.get_measurement() method`

kizniche, 2019-05-31:
It's been challenging to get the Ruuvitag and SHT31 Smart Sensor to work together. I have a hunch it's my lockfile implementation, so I'm currently testing the filelock library instead of locket. So far it's working, but time will tell if data keeps coming in. I'll keep you updated.

RynFlutsch, 2019-06-01:
I tried SHT31 + Ruuvi and its not working. No Data since x seconds.
2 Ruuvis + 3 Xiaomi Miflora without the SHT31 are working ;-)

kizniche, 2019-06-02:
I think I may have found the issue with the Ruuvitag script hanging and preventing future measurements. I modified the execution code to kill the process after a timeout period of 10 seconds. I'm going to test overnight.

kizniche, 2019-06-03:
No issues so far. If that continues I'll try to get a release out today.

kizniche, 2019-06-03:
Here's a log excerpt with debugging enabled, where you can see the script not responding. Previously, this is what would stop the Ruuvitag from collecting future measurements until a daemon restart.

```
2019-06-03 09:09:16,608 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Starting measurement
2019-06-03 09:09:19,466 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Completed measurement
2019-06-03 09:09:19,592 - DEBUG - mycodo.controller_input_3569c515 - Adding measurements to InfluxDB with ID 3569c515-f62c-40c6-97d8-cf550b998522: {0: {'unit': 'C', 'measurement': 'temperature', 'timestamp_utc': None, 'value': 28.3}, 1: {'unit': 'percent', 'measurement': 'humidity', 'timestamp_utc': None, 'value': 31.5}, 2: {'unit': 'Pa', 'measurement': 'pressure', 'timestamp_utc': None, 'value': 981.82}, 3: {'unit': 'V', 'measurement': 'battery', 'timestamp_utc': None, 'value': 3.061}, 4: {'unit': 'g_force', 'measurement': 'acceleration_g_force', 'timestamp_utc': None, 'value': 1.0221942085533453}, 5: {'unit': 'g_force', 'measurement': 'acceleration_x_g_force', 'timestamp_utc': None, 'value': 0.026}, 6: {'unit': 'g_force', 'measurement': 'acceleration_y_g_force', 'timestamp_utc': None, 'value': -0.042}, 7: {'unit': 'g_force', 'measurement': 'acceleration_z_g_force', 'timestamp_utc': None, 'value': 1.021}}
2019-06-03 09:10:16,607 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Starting measurement
2019-06-03 09:10:26,655 - DEBUG - mycodo.system_pi - cmd_output() timed out after 10 seconds with command '/var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/inputs/scripts/ruuvitag_values.py --mac_address C3:31:AC:B4:17:31 --bt_adapter 0'
2019-06-03 09:11:16,648 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Starting measurement
2019-06-03 09:11:17,824 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Completed measurement
2019-06-03 09:11:17,950 - DEBUG - mycodo.controller_input_3569c515 - Adding measurements to InfluxDB with ID 3569c515-f62c-40c6-97d8-cf550b998522: {0: {'unit': 'C', 'measurement': 'temperature', 'timestamp_utc': None, 'value': 28.36}, 1: {'unit': 'percent', 'measurement': 'humidity', 'timestamp_utc': None, 'value': 31.5}, 2: {'unit': 'Pa', 'measurement': 'pressure', 'timestamp_utc': None, 'value': 981.85}, 3: {'unit': 'V', 'measurement': 'battery', 'timestamp_utc': None, 'value': 3.073}, 4: {'unit': 'g_force', 'measurement': 'acceleration_g_force', 'timestamp_utc': None, 'value': 1.027141665010236}, 5: {'unit': 'g_force', 'measurement': 'acceleration_x_g_force', 'timestamp_utc': None, 'value': 0.03}, 6: {'unit': 'g_force', 'measurement': 'acceleration_y_g_force', 'timestamp_utc': None, 'value': -0.038}, 7: {'unit': 'g_force', 'measurement': 'acceleration_z_g_force', 'timestamp_utc': None, 'value': 1.026}}
2019-06-03 09:12:16,660 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Starting measurement
2019-06-03 09:12:18,388 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Completed measurement
2019-06-03 09:12:18,514 - DEBUG - mycodo.controller_input_3569c515 - Adding measurements to InfluxDB with ID 3569c515-f62c-40c6-97d8-cf550b998522: {0: {'unit': 'C', 'measurement': 'temperature', 'timestamp_utc': None, 'value': 28.4}, 1: {'unit': 'percent', 'measurement': 'humidity', 'timestamp_utc': None, 'value': 31.5}, 2: {'unit': 'Pa', 'measurement': 'pressure', 'timestamp_utc': None, 'value': 656.16}, 3: {'unit': 'V', 'measurement': 'battery', 'timestamp_utc': None, 'value': 3.073}, 4: {'unit': 'g_force', 'measurement': 'acceleration_g_force', 'timestamp_utc': None, 'value': 1.024221167521937}, 5: {'unit': 'g_force', 'measurement': 'acceleration_x_g_force', 'timestamp_utc': None, 'value': 0.03}, 6: {'unit': 'g_force', 'measurement': 'acceleration_y_g_force', 'timestamp_utc': None, 'value': -0.04}, 7: {'unit': 'g_force', 'measurement': 'acceleration_z_g_force', 'timestamp_utc': None, 'value': 1.023}}
```

kizniche, 2019-06-03:
And here is an excerpt with locking working as expected. You can see the Smart Gadget start before the Ruuvitag, and the Ruuvitag wait until the SG finishes before trying to access the BT device.

```
2019-06-03 07:59:16,680 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Starting measurement
2019-06-03 07:59:17,940 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Completed measurement
2019-06-03 07:59:18,061 - DEBUG - mycodo.controller_input_3569c515 - Adding measurements to InfluxDB with ID 3569c515-f62c-40c6-97d8-cf550b998522: {0: {'unit': 'C', 'measurement': 'temperature', 'timestamp_utc': None, 'value': 28.61}, 1: {'unit': 'percent', 'measurement': 'humidity', 'timestamp_utc': None, 'value': 32.0}, 2: {'unit': 'Pa', 'measurement': 'pressure', 'timestamp_utc': None, 'value': 981.39}, 3: {'unit': 'V', 'measurement': 'battery', 'timestamp_utc': None, 'value': 3.085}, 4: {'unit': 'g_force', 'measurement': 'acceleration_g_force', 'timestamp_utc': None, 'value': 1.0232135651954581}, 5: {'unit': 'g_force', 'measurement': 'acceleration_x_g_force', 'timestamp_utc': None, 'value': 0.031}, 6: {'unit': 'g_force', 'measurement': 'acceleration_y_g_force', 'timestamp_utc': None, 'value': -0.039}, 7: {'unit': 'g_force', 'measurement': 'acceleration_z_g_force', 'timestamp_utc': None, 'value': 1.022}}
2019-06-03 08:00:15,033 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Starting measurement
2019-06-03 08:00:16,678 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Starting measurement
2019-06-03 08:05:16,220 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Completed measurement
2019-06-03 08:05:16,317 - DEBUG - mycodo.controller_input_ce2139b0 - Adding measurements to InfluxDB with ID ce2139b0-c728-4b5d-82cd-efe320b8ba47: {0: {'unit': 'C', 'measurement': 'temperature', 'timestamp_utc': None, 'value': 26.5}, 1: {'unit': 'percent', 'measurement': 'humidity', 'timestamp_utc': None, 'value': 41.84000015258789}, 2: {'unit': 'percent', 'measurement': 'battery', 'timestamp_utc': None, 'value': 67}, 3: {'unit': 'C', 'measurement': 'dewpoint', 'timestamp_utc': None, 'value': 12.482609727743696}, 4: {'unit': 'Pa', 'measurement': 'vapor_pressure_deficit', 'timestamp_utc': None, 'value': 2013.0937116066461}}
2019-06-03 08:05:17,998 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Completed measurement
2019-06-03 08:05:18,122 - DEBUG - mycodo.controller_input_3569c515 - Adding measurements to InfluxDB with ID 3569c515-f62c-40c6-97d8-cf550b998522: {0: {'unit': 'C', 'measurement': 'temperature', 'timestamp_utc': None, 'value': 28.7}, 1: {'unit': 'percent', 'measurement': 'humidity', 'timestamp_utc': None, 'value': 32.0}, 2: {'unit': 'Pa', 'measurement': 'pressure', 'timestamp_utc': None, 'value': 981.32}, 3: {'unit': 'V', 'measurement': 'battery', 'timestamp_utc': None, 'value': 3.073}, 4: {'unit': 'g_force', 'measurement': 'acceleration_g_force', 'timestamp_utc': None, 'value': 1.0280573913940798}, 5: {'unit': 'g_force', 'measurement': 'acceleration_x_g_force', 'timestamp_utc': None, 'value': 0.027}, 6: {'unit': 'g_force', 'measurement': 'acceleration_y_g_force', 'timestamp_utc': None, 'value': -0.038}, 7: {'unit': 'g_force', 'measurement': 'acceleration_z_g_force', 'timestamp_utc': None, 'value': 1.027}}
```

kizniche, 2019-06-03:
I haven't seen an issue in 24 hours, so I think it's stable for release. I'll prepare that now.

kizniche, 2019-06-04:
v7.5.5 released

kizniche, 2019-06-05:
I just experienced a SHT31 SG error that stopped both sensors from operating. I'm investigating it and working on a solution.

kizniche, 2019-06-06:
I think I finally fixed the SHT31 issue. I may have broken other inputs in the process, but will test this afternoon.

RynFlutsch, 2019-06-06:
Ok i test 7.5.5 today with SHT31 and Ruuvi

kizniche, 2019-06-06:
The current release is not the latest code and there will be issues. Even the master branch doesn't yet have the latest code. I need to test the changes I made with other Inputs before I commit the changes to the master branch. I'll let you know when I do this if you want to upgrade to the master branch to test.

kizniche, 2019-06-07:
I pushed a commit to master that appears to be stable. If you want to upgrade to master and test, it should be okay to do so now.

kizniche, 2019-06-07:
Getting these two sensors has been a challenge! I think I finally got them working. Though it's not just for this set, but rather any sensor that uses bluetooth, as only one can connect at a time. So, this also affects the Miflora and others that may be added in the future.

I had to remove the function timeout feature because there were issues I didn't notice when I committed my edits last night. Killing a threaded process is more difficult than I imagined, and I've about exhausted all my current skill in getting it to work. I'll have to wait and see if one of the sensors hangs in a blocking state and debug where the exact issue is and hope it can be fixed. The debug lines for the SHT31 is starting to look ridiculous, but I at least have reporting coverage that can help me determine exactly where a hang occurs. Here's an excerpt of my debug log (enable Input setting: "Log Level: Debug"):

```
2019-06-07 10:58:27,781 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Acquiring lock for /var/lock/bluetooth_dev_hci0 (3600 sec timeout)
2019-06-07 10:58:27,784 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Lock acquired for /var/lock/bluetooth_dev_hci0 in 0.003 seconds
2019-06-07 10:58:27,785 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Starting measurement
2019-06-07 10:58:27,786 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Input Initializing (Initialized: False)
2019-06-07 10:58:29,510 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Acquiring lock for /var/lock/bluetooth_dev_hci0 (3600 sec timeout)
2019-06-07 10:58:33,071 - DEBUG - mycodo.device.sht31_smart_gadget.delegate - SHT31Delegate Initialized
2019-06-07 10:58:33,072 - DEBUG - mycodo.device.sht31_smart_gadget.sht31 - __prepareGadget()
2019-06-07 10:59:00,514 - DEBUG - mycodo.device.sht31_smart_gadget.sht31 - SHT31 Initialized
2019-06-07 10:59:01,343 - INFO - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Sensirion AG, SmartGadget, SN: 00000000, Name: Smart Humigadget, Firmware: 1.3, Hardware: 1, Software: 0.1, Log Interval: 1800.0 sec
2019-06-07 10:59:01,344 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Downloading Data
2019-06-07 10:59:01,345 - DEBUG - mycodo.device.sht31_smart_gadget.sht31 - readLoggedDataInterval()
2019-06-07 10:59:01,346 - DEBUG - mycodo.device.sht31_smart_gadget.sht31 - setSyncTimeMs()
2019-06-07 10:59:01,636 - DEBUG - mycodo.device.sht31_smart_gadget.delegate - prepareLoggerReadout()
2019-06-07 10:59:01,734 - DEBUG - mycodo.device.sht31_smart_gadget.delegate - handleNotification()
...
2019-06-07 10:59:04,172 - DEBUG - mycodo.device.sht31_smart_gadget.delegate - handleNotification()
2019-06-07 10:59:04,173 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Downloaded Data
2019-06-07 10:59:04,174 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Parsing/saving data
2019-06-07 10:59:04,175 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Storing 56 temperatures
2019-06-07 10:59:05,570 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Storing 72 humidities
2019-06-07 10:59:07,086 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Calculating/storing 56 dewpoint and vpd
2019-06-07 10:59:09,458 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Parsed/saved data
2019-06-07 10:59:09,459 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Acquiring present measurements
2019-06-07 10:59:09,460 - DEBUG - mycodo.device.sht31_smart_gadget.delegate - handleNotification()
...
2019-06-07 10:59:09,464 - DEBUG - mycodo.device.sht31_smart_gadget.delegate - handleNotification()
2019-06-07 10:59:09,728 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Acquired present measurements
2019-06-07 10:59:09,729 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Disconnecting
2019-06-07 10:59:09,729 - DEBUG - mycodo.device.sht31_smart_gadget.sht31 - disconnect()
2019-06-07 10:59:09,731 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Disconnected
2019-06-07 10:59:09,732 - DEBUG - mycodo.inputs.sht31_smart_gadget_ce2139b0 - Completed measurement
2019-06-07 10:59:09,749 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Lock acquired for /var/lock/bluetooth_dev_hci0 in 40.239 seconds
2019-06-07 10:59:09,750 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Starting measurement
2019-06-07 10:59:09,877 - DEBUG - mycodo.controller_input_ce2139b0 - Adding measurements to InfluxDB with ID ce2139b0-c728-4b5d-82cd-efe320b8ba47: {0: {'value': 26.399999618530273, 'unit': 'C', 'measurement': 'temperature', 'timestamp_utc': None}, 1: {'value': 44.36000061035156, 'unit': 'percent', 'measurement': 'humidity', 'timestamp_utc': None}, 2: {'value': 87, 'unit': 'percent', 'measurement': 'battery', 'timestamp_utc': None}, 3: {'value': 13.28720012253613, 'unit': 'C', 'measurement': 'dewpoint', 'timestamp_utc': None}, 4: {'value': 1914.5568609630195, 'unit': 'Pa', 'measurement': 'vapor_pressure_deficit', 'timestamp_utc': None}}
2019-06-07 10:59:11,531 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Completed measurement
2019-06-07 10:59:11,677 - DEBUG - mycodo.controller_input_3569c515 - Adding measurements to InfluxDB with ID 3569c515-f62c-40c6-97d8-cf550b998522: {0: {'value': 28.93, 'unit': 'C', 'measurement': 'temperature', 'timestamp_utc': None}, 1: {'value': 34.0, 'unit': 'percent', 'measurement': 'humidity', 'timestamp_utc': None}, 2: {'value': 973.93, 'unit': 'Pa', 'measurement': 'pressure', 'timestamp_utc': None}, 3: {'value': 3.073, 'unit': 'V', 'measurement': 'battery', 'timestamp_utc': None}, 4: {'value': 1.028 [... cut ...]  'acceleration_z_g_force', 'timestamp_utc': None}}
2019-06-07 11:00:29,504 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Acquiring lock for /var/lock/bluetooth_dev_hci0 (3600 sec timeout)
2019-06-07 11:00:29,506 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Lock acquired for /var/lock/bluetooth_dev_hci0 in 0.002 seconds
2019-06-07 11:00:29,507 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Starting measurement
2019-06-07 11:00:32,600 - DEBUG - mycodo.inputs.ruuvitag_3569c515 - Completed measurement
```

RynFlutsch, 2019-06-09:
Yes i think the SHT31 is the Problem.
Without the SHT31 Gadget its working.

RynFlutsch, 2019-06-09:
Daemon.log Debug:
`2019-06-09 13:02:25,488 - DEBUG - mycodo.controller_input_f4e4ed01 - Adding measurements to InfluxDB with ID f4e4ed01-9b75-400c-bc45-0ce5e1870305: {0: {'timestamp_utc': None, 'measurement': 'temperature', 'unit': 'C', 'value': 24.56999969482422}, 1: {'timestamp_utc': None, 'measurement': 'humidity', 'unit': 'percent', 'value': 44.4900016784668}, 2: {'timestamp_utc': None, 'measurement': 'battery', 'unit': 'percent', 'value': 82}, 3: {'timestamp_utc': None, 'measurement': 'dewpoint', 'unit': 'C', 'value': 11.674729293897052}, 4: {'timestamp_utc': None, 'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1713.5152720628857}}
2019-06-09 13:02:54,238 - DEBUG - mycodo.inputs.sht31_smart_gadget_f4e4ed01 - Starting measurement
2019-06-09 13:03:23,411 - DEBUG - mycodo.inputs.sht31_smart_gadget_f4e4ed01 - Completed measurement
2019-06-09 13:03:23,469 - DEBUG - mycodo.controller_input_f4e4ed01 - Adding measurements to InfluxDB with ID f4e4ed01-9b75-400c-bc45-0ce5e1870305: {0: {'timestamp_utc': None, 'measurement': 'temperature', 'unit': 'C', 'value': 24.56999969482422}, 1: {'timestamp_utc': None, 'measurement': 'humidity', 'unit': 'percent', 'value': 44.65999984741211}, 2: {'timestamp_utc': None, 'measurement': 'battery', 'unit': 'percent', 'value': 82}, 3: {'timestamp_utc': None, 'measurement': 'dewpoint', 'unit': 'C', 'value': 11.73253972731222}, 4: {'timestamp_utc': None, 'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1708.2676686127243}}
2019-06-09 13:03:24,276 - DEBUG - mycodo.inputs.sht31_smart_gadget_f4e4ed01 - Starting measurement
2019-06-09 13:03:54,243 - DEBUG - mycodo.inputs.sht31_smart_gadget_f4e4ed01 - Starting measurement
2019-06-09 13:03:54,637 - DEBUG - mycodo.inputs.sht31_smart_gadget_f4e4ed01 - Not connected: Not measuring
2019-06-09 13:04:24,284 - DEBUG - mycodo.inputs.sht31_smart_gadget_f4e4ed01 - Starting measurement`

kizniche, 2019-06-09:
From the formatting of your log lines, I can tell you're not running the latest code.

RynFlutsch, 2019-06-09:
Latest version: 7.5.5 ?

kizniche, 2019-06-09:
To upgrade to the master branch, you need to follow these instructions: https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch

RynFlutsch, 2019-06-09:
Ok thanks

RynFlutsch, 2019-06-09:
I think i am up to date now:
`2019-06-09 13:03:24,276 - DEBUG - mycodo.inputs.sht31_smart_gadget_f4e4ed01 - Starting measurement
2019-06-09 13:03:25,581 - ERROR - mycodo.inputs.sht31_smart_gadget_f4e4ed01 - InputModule raised an exception when taking a reading: 'SHT31' object has no attribute '_SHT31__characteristics'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 116, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 384, in get_measurement
    self.connect()
  File "/home/pi/Mycodo/mycodo/inputs/sht31_smart_gadget.py", line 179, in connect
    addr=self.location, iface=self.bt_adapter)
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 103, in __init__
    self.__prepareGadget()
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 108, in __prepareGadget
    self.__peripheral.getCharacteristics(uuid=UUID(0x2A23))[0],
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 508, in getCharacteristics
    rsp = self._getResp('find')
  File "/var/mycodo-root/env/lib/python3.5/site-packages/bluepy/btle.py", line 416, in _getResp
    self.delegate.handleNotification(hnd, data)
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 82, in handleNotification
    self.parent.setHumidityNotification(False)
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 189, in setHumidityNotification
    self.__setHumidityNotification(tmp)
  File "/var/mycodo-root/mycodo/devices/sht31_smart_gadget.py", line 193, in __setHumidityNotification
    self.__characteristics['Humidity'].valHandle + 2,
AttributeError: 'SHT31' object has no attribute '_SHT31__characteristics'
2019-06-09 13:03:54,243 - DEBUG - mycodo.inputs.sht31_smart_gadget_f4e4ed01 - Starting measurement
2019-06-09 13:03:54,636 - ERROR - mycodo.inputs.sht31_smart_gadget_f4e4ed01 - Could not connect: Failed to connect to peripheral FD:E1:1B:A9:D4:8E, addr type: random`

kizniche, 2019-06-09:
We're there any errors in the upgrade log? I've never received that error before.

RynFlutsch, 2019-06-09:
Looks good

RynFlutsch, 2019-06-09:
I will test 2 SHT 31 Gadgets + 1 Ruuvi this night and report then.

kizniche, 2019-06-15:
I've had the latest code running for over a week without issue. I've made small tweaks to the Ruuvitag Input to throw out aberrant measurements, but besides that, no reemergence of the blocking/hanging issues seen previously.

kizniche, 2019-09-05:
I've had quite the time trying to get the Ruuvitag and SHT31-SG to play nicely together. I believe the issue may be the Ruuvitag library (and it causing BT issues that then mess up the SHT31-SG), so I'm currently running the experimental branch of the library that will eventually be replacing the current ruuvitag_sensor codebase. The issue could also be the Pi3 BT. I haven't tested with the Pi4, since I don't have one.

eap904, 2020-02-14:
Just ordered 3 ruuvitags, any update on using them together with the SHT31-SG? I am looking for a better bluetooth USB adapter for the pi that will have better range. I really like the smart gadgets but they seem to eat batteries and generally be problematic connecting to the pi, so I'm hoping the ruuvi's will be a bit more stable.

kizniche, 2020-05-01:
Just a quick comment to link issue #731 where using a Bluetooth dongle significantly improves reliability with connecting to SHT31-SGs.

closed 2021-09-20 by kizniche

spilsbury4, 2025-03-07:
Hi Kyle I am not able to configure the input (gear icon non-responsive) for the RuuviTag. Trying to limit the measurement selection to temperature (converted to F), humidity, and VPD. Any ideas?