kizniche/Mycodo #640: Failed update

jacopienaar, 2019-03-17:
Ran the update in the web interface and it all bombed out. Restarting doesn't work like it did for others. Tried reinstalling but doesn't work since existing 

  - Specific Mycodo Version:
I think it was 7.1.3 or something like that

 
### Errors
  Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 507, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: input.ftdi_location

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1058, in page_live
    input_dev = Input.query.all()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2773, in all
    return list(self)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2925, in __iter__
    return self._execute_and_instances(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2948, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 948, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/sql/elements.py", line 269, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1060, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1200, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1413, in _handle_dbapi_exception
    exc_info
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 203, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 186, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 507, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: input.ftdi_location [SQL: 'SELECT input.id AS input_id, input.unique_id AS input_unique_id, input.name AS input_name, input.is_activated AS input_is_activated, input.is_preset AS input_is_preset, input.preset_name AS input_preset_name, input.device AS input_device, input.interface AS input_interface, input.period AS input_period, input.power_output_id AS input_power_output_id, input.resolution AS input_resolution, input.resolution_2 AS input_resolution_2, input.sensitivity AS input_sensitivity, input.thermocouple_type AS input_thermocouple_type, input.ref_ohm AS input_ref_ohm, input.calibrate_sensor_measure AS input_calibrate_sensor_measure, input.location AS input_location, input.gpio_location AS input_gpio_location, input.i2c_location AS input_i2c_location, input.i2c_bus AS input_i2c_bus, input.ftdi_location AS input_ftdi_location, input.uart_location AS input_uart_location, input.baud_rate AS input_baud_rate, input.pin_clock AS input_pin_clock, input.pin_cs AS input_pin_cs, input.pin_mosi AS input_pin_mosi, input.pin_miso AS input_pin_miso, input.bt_adapter AS input_bt_adapter, input.switch_edge AS input_switch_edge, input.switch_bouncetime AS input_switch_bouncetime, input.switch_reset_period AS input_switch_reset_period, input.pre_output_id AS input_pre_output_id, input.pre_output_duration AS input_pre_output_duration, input.pre_output_during_measure AS input_pre_output_during_measure, input.sht_voltage AS input_sht_voltage, input.adc_gain AS input_adc_gain, input.adc_resolution AS input_adc_resolution, input.adc_sample_speed AS input_adc_sample_speed, input.cmd_command AS input_cmd_command, input.weighting AS input_weighting, input.rpm_pulses_per_rev AS input_rpm_pulses_per_rev, input.sample_time AS input_sample_time, input.port AS input_port, input.times_check AS input_times_check, input.deadline AS input_deadline, input.custom_options AS input_custom_options \nFROM input'] (Background on this error at: http://sqlalche.me/e/e3q8)

jacopienaar, 2019-03-17:
So I tried upgrading with the command line using the command the user manual provides - that failed and I got this at the end:
###
There was an error during the upgrade.
Initial steps to try to fix:
1. Reboot
2. If that doesn't fix the issue, run the following command:
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh upgrade
3. If that command returns that you are running the latest version, run the following command:
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh

###
Tried all of em and the third one kinda worked - it completed but I still cannot connect via the web portal.
Installation finished with this : 
###
#### Updating Mycodo restart monitor crontab entry
Removing crontab entry ...
Appending to crontab:
-----------------------------------------------
@reboot /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/scripts/restart_daemon.py --continuouscheck &
-----------------------------------------------
To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh restart_daemon --remove
#### Setting permissions
#### Restarting the Mycodo daemon
Mycodo previously shut down properly
#### Restarting nginx
#### Reloading mycodoflask
#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Successfully connected to http://localhost

jacopienaar, 2019-03-17:
HI Kyle - what is the command to completely uninstall mycodo so I can start fresh?

kizniche, 2019-03-17:
Follow the steps in the linked comment to delete your database and generate a new one: https://github.com/kizniche/Mycodo/issues/340#issuecomment-367819796

kizniche, 2019-03-17:
Once you have a new database and can log in, you can attempt a restore of your previous working system, then attempt the upgrade again. If you continue to have issues upgrading your database (as was the case here), then you may have something wrong with your database (opposed to it being an issue with the database upgrade process).

jacopienaar, 2019-03-17:
Thanks Kyle - the steps in the comment worked, however, when I attempted to upgrade from 7.0.3 the same happened again.  I'm not too concerned with with retrieving my old database and actually prefer starting with a vanilla Mycodo.

I followed [https://github.com/kizniche/Mycodo/issues/452#issuecomment-383296936](url) to remove and install the latest version.

kizniche, 2019-03-17:
What version are you currently at, btw? You may be at a particular version that had a database schema issue.

Here's how to reinstall:

```
sudo service mycodo stop
sudo service mycodoflask stop
mv ~/Mycodo ~/Mycodo-old
```

Then follow the install instructions in the README here in this repository.

jacopienaar, 2019-03-17:
I think I was on a old version  - 3.something and upgraded to 7.0.3 and that's maybe where the problem came in as I couldn't upgrade from 7.0.3 to the latest version.

All good now - have a new Mycodo running 

Thanks for the support!

kizniche, 2019-03-17:
3? 😬

I don't think 3.x even had a working upgrade system for major revisions. Reinstalling was the proper course of action.

closed 2019-03-24 by kizniche