kizniche/Mycodo #641: Mycodo CCS811 Install Does Not Record Data

justinjaythomas, 2019-03-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.3.1
 
#### Problem Description

Fresh Pi Zero W, 9000C1 

Fresh and updated Raspbian

Raspbian GNU/Linux 9 (stretch)
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"

Fresh mycodo 7.3.1

New Adafruit CCS811 Air Quality Sensor

Physically connect CCS811 sensor to Pi Zero W as per:

https://learn.adafruit.com/adafruit-ccs811-air-quality-sensor/raspberry-pi-wiring-test

Edit /boot/config.txt  and add line to slow down baud rate for Pi clock: 

dtparam=i2c_baudrate=10000

Once I set the baud rate I can run:

sudo i2cdetect -y 1

Sensor lists properly as 5a

  0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- 5a -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- — --                       

When adding  CCS811 in fresh install of Mycodo 7.3.1, I am asked to install dependencies, and this process completes without error. Log lists successful install of all dependencies.

CCS811 input adds in mycodo, then activates without error, but sensor does not collect or display data in Live or Dashboard. 

Live page displays only this, only the CO2 value even lists an update time. VOC and Temp always list no data.

0 ppm (CO2) CH1 | 2019/3/17 15:19:53
0 ppb (VOC) CH2 | No Data Last 16 sec
0 °C (Temperature) CH3 | No Data Last 16 sec

A Dashboard graph or gauge displays no data.

When I test same sensor on Pi using the default Adafruit libraries and scripts, sensor DOES properly display data.

CO2:  611 ppm, TVOC:  32  temp:  25.0
CO2:  0 ppm, TVOC:  0  temp:  25.0
CO2:  0 ppm, TVOC:  0  temp:  24.5426214168
CO2:  733 ppm, TVOC:  50  temp:  24.5426214168
CO2:  400 ppm, TVOC:  0  temp:  24.5426214168
CO2:  400 ppm, TVOC:  0  temp:  25.0
CO2:  400 ppm, TVOC:  0  temp:  25.0
CO2:  1544 ppm, TVOC:  174  temp:  25.0
CO2:  575 ppm, TVOC:  26  temp:  24.5426214168
CO2:  423 ppm, TVOC:  3  temp:  25.0

(***variations in CO2 ppm caused by my breathing on the sensor.)

 
### Errors

There are no errors in the mycodo log, or the dependency log.

Sensor manually activates and deactivates without error.

I have tried this same test with two different CCS811, both new. Both test results are the same.
No result in Live or Dashboard graph or gauge.

Proper results using default adafruit libraries and scripts.

### Steps to Reproduce the issue:
How can this issue be reproduced?
1. Fresh Pi Zero W, 9000C1 
  2. Fresh and updated Raspbian 9 (stretch)
 3. Fresh Mycodo 7.3.1
4. Connect sensor
5. Edit /boot/config.txt dtparam=i2c_baudrate=10000
6. sudo i2cdetect -y 1 “5a”
7. Add mycodo input (Install dependencies)
8. Add mycodo input CCS811 and activate.
9. Live and Dashboard blank.
 
### Additional Notes

Until dtparam=i2c_baudrate=10000 is added to the config.txt, both methods to access this sensor fail.

kizniche, 2019-03-18:
Can you check the Daemon Log for errors?

kizniche, 2019-03-18:
I think I found the issue. Thank you for the detailed report- it definitely helped. I just pushed a potential fix. Can you manually edit your ~/Mycodo/mycodo/inputs/ccs811.py to match the edits I made (see commit, above), save, restart the daemon with ```sudo service mycodo restart```, and test the Input?

justinjaythomas, 2019-03-18:
Made those changes to ccs811.py

Nothing changes in the original problem scenario, with the exception of this single log line, which occurs three different times during the start of the daemon:

2019-03-18 12:55:44,567 - mycodo.utils.inputs - ERROR - Error: Cannot add input modules because it does not have a unique name: CCS811

justinjaythomas, 2019-03-18:
Thank you for responding so quickly.

I made those changes to ccs811.py

Nothing changes in the original problem scenario, with the exception of this single log line, which occurs three different times during the start of the daemon:

2019-03-18 12:55:44,567 - mycodo.utils.inputs - ERROR - Error: Cannot add input modules because it does not have a unique name: CCS811

Sounds like you’re almost there…

J.

> On Mar 18, 2019, at 10:33 AM, Kyle Gabriel <notifications@github.com> wrote:
> 
> sudo service mycodo restart

kizniche, 2019-03-18:
> 2019-03-18 12:55:44,567 - mycodo.utils.inputs - ERROR - Error: Cannot add input modules because it does not have a unique name: CCS811

It appears you may have created a custom input module with the same name as the built-in one. That, or you saved a new file in the inputs directory instead of overwriting the file you edited.

I did just find another issue that may or may not affect the function of the module. You'll have to test to determine that. The commit, above, has the changes you should apply, restart the daemon, and test.

The input module should now be the exact same as the Adafruit example code, so if it doesn't work after this, I'm not sure why it's not obtaining measurements, if the example code you executed does return measurements.

kizniche, 2019-03-18:
Just pushed another commit that fixes the numbering... I had them incorrect. Try that and see if ti works.

kizniche, 2019-03-18:
I accidentally pushed a bunch of other files I didn't mean to with that last commit. Here's a direct link to the latest file: https://github.com/kizniche/Mycodo/blob/d961aab484fb805a01e394533510346e8b2dc26e/mycodo/inputs/ccs811.py

justinjaythomas, 2019-03-18:
Very sorry about that. 

I did rename the original script ccs811BAK.py, and made a new script using the name ccs811.py. 

I guess I didn't rename the original script well enough? 

I edited the original ccs811.py and the error went away.

The CCS811 sensor collects CO2 data now. One up.

The other two functions of the sensor, Temp and Voc are still not populating in Live. Two down.

0 ppb (VOC) CH2 | No Data Last 16 sec
0 °C (Temperature) CH3 | No Data Last 16 sec

Data also not displaying on the Dashboard.

I have only an interest is the CO2 data collection, so this issue is closed, from my narrow perspective?

justinjaythomas, 2019-03-18:
Very sorry about that. 

I did rename the original script ccs811BAK.py, and made a new script using the name ccs811.py. 

I guess I didn't rename the original script well enough? 

I edited the original ccs811.py and the error went away.

The CCS811 sensor collects CO2 data now. One up.

The other two functions of the sensor, Temp and Voc are still not populating in Live. Two down.

0 ppb (VOC) CH2 | No Data Last 16 sec
0 °C (Temperature) CH3 | No Data Last 16 sec

Data also not displaying on the Dashboard.

I have only an interest is the CO2 data collection, so this issue is closed, from my narrow perspective?

You may want to patch over the rest when you are ready.

BTW: Your software rocks. Thank you very much for creating and maintaining it.

J.

> On Mar 18, 2019, at 10:33 AM, Kyle Gabriel <notifications@github.com> wrote:
> 
> sudo service mycodo restart

kizniche, 2019-03-18:
Could you please test the latest version of the input file (commit, above)? I think I finally fixed it.

justinjaythomas, 2019-03-19:
Yes, those final changes worked. Thank you.

439 ppm (CO2) CH1 | 2019/3/18 19:09:49
8 ppb (VOC) CH2 | 2019/3/18 19:10:19
24.627 °C (Temperature) CH3 | 2019/3/18 19:10:19

justinjaythomas, 2019-03-19:
Yes, those final changes worked. Thank you.

439 ppm (CO2) CH1 | 2019/3/18 19:09:49
8 ppb (VOC) CH2 | 2019/3/18 19:10:19
24.627 °C (Temperature) CH3 | 2019/3/18 19:10:19

J.

kizniche, 2019-03-19:
Great! Thanks for testing. There are a few sensors Mycodo supports that I don't have because I added them at a user's request. This sensor module has had issuses in the past, so I'm glad we got it straightened out.

closed 2019-03-19 by kizniche