kizniche/Mycodo #643: Add feature: Add the Atlas inputs of ORP and DO for Atlas sensor chip

alonrab, 2019-03-29:
Hi Kyle et. al.
Can you please help me with adding the atlas sensor inputs for O.R.P and D.O.?
https://www.atlas-scientific.com/product_pages/circuits/ezo_orp.html
https://www.atlas-scientific.com/product_pages/circuits/ezo_do.html
Best,

kizniche, 2019-04-01:
Sure. What is your interface, UART or I2C?

alonrab, 2019-04-02:
I'm using i2c,
Thanks!

kizniche, 2019-04-18:
Added both sensors with FTDI, UART, and I2C support for each. I'm going to try to get this out in a release tonight.

kizniche, 2019-04-18:
I'll keep this issue open until you get a chance to test both Input modules.

alonrab, 2019-04-19:
Hi, tested ORP today, works great! Thanks, i am waiting to get the DO probe but its very similar code so i assume it works as well
Many thanks for your help

kizniche, 2019-05-17:
Did you ever get the DO probe?

alonrab, 2019-05-22:
No - was not able to afford it yet. Will update when I get it if there are issues,Many thanks!

closed 2019-06-07 by kizniche