kizniche/Mycodo #644: Backup transfer to another S Card

RynFlutsch, 2019-03-31:
How it is possible to transfer an backup from Card A to another SD Card B and restore it there?

I tried to copy the Backup but 1 file cant be copied.
[image]

RynFlutsch, 2019-03-31:
Someone know why the thumbnail is twistet, but the origina pic.l is correct?

kizniche, 2019-04-01:
Yes, but you will need root privileges. Using the command line, you would use ```sudo cp - R source-directory destination```

RynFlutsch, 2019-04-01:
Ok thanks, i tried with Thunar root (+ i changed the permissions to write and read, to everyone), but i will try again via console.

kizniche, 2019-04-01:
You can also change the permissions of the whole backup directory with:

```sudo chown -R pi.pi /var/Mycodo-Backups```

RynFlutsch, 2019-04-02:
Console roxx ;-D

closed 2019-04-02 by RynFlutsch