kizniche/Mycodo #647: 128x64 OLED Address (I2C) cannot be changed

stminm, 2019-04-10:
## Mycodo Issue Report:
  - Specific Mycodo Version:  7.4.2
#### Problem Description
On a fresh installed Mycodo I added a 128x64 OLED LCD but the Address (I2C) cannot be changed. The supposed drop-down menu is gray out with the preconfigured 0x3c.
 
[image]

kizniche, 2019-04-10:
Can you provide information about the hardware you're using?

stminm, 2019-04-11:
First of all, I want to say that I was able to make the display work.
Below are the hardware specs and what I did.
Raspberry Pi 3 Model B
Monochrome 1.3" 128x64 OLED graphic display: https://www.adafruit.com/product/938
I forced the display to work in I2C mode by closing SJ1 and SJ2 on display's back.
Wiring as follow:
LCD           Pi
Gnd  -> 9 (GND)
Vin    -> 1 (3.3V PWR)
Rst    -> 7 (GPIO4 (GPIO_GCLK))
Clk   -> 5 (GPIO3 (SCL1, I2C))
Data -> 3 (GPIO2 (SDA1, I2C))

I had to use Rst because without it I was getting random white pixels on display.
Also, I specified the GPIO4, where the Rst is connected to, by changing https://github.com/kizniche/Mycodo/blob/82c19420a23d63016a7c32cbb5a5aa7abd0ad889/mycodo/devices/lcd_pioled.py#L32 into "rst=4,"

Because of this:
$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- 3d -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
I changed https://github.com/kizniche/Mycodo/blob/ecc6bbfdf24c2d98b62f7bb3f315db683a5f7b66/mycodo/mycodo_flask/utils/utils_lcd.py#L68 into "new_lcd.location = '0x3d'"
Now, the Address (I2C) in webinterface is still grayed out but it shows the 0x3d address and the display is working.

Since all my manual changes will be wiped out with the next upgrade It will be awesome if you could put them in the code or fix the Address (I2C) grayed out drop-down menu issue and for displays with Rst pin offer the possibility to choose the GPIO where the Rst is connected.

kizniche, 2019-04-11:
The SD1306 driver is used in a lot of 128x64 OLED products. When I wrote this driver, I was unfamiliar with it, and did it at the request of a user who had a display. Since then, I've obtained several different boards. I'll add the Reset Pin option and make the location editable, to account for device variability.

stminm, 2019-04-11:
Sorry for accidentally closing the issue. I thought I'm closing the editor and sending the comment.
In no way, I'm blaming or criticizing your work. It's understandable that you can't keep track of all possible attachable devices to Mycodo. Plus, "WITHOUT ANY WARRANTY" means exactly what it says.
IMHO Mycodo is one of the best projects for Raspberry Pi in terms of usefulness, interface usability, versatility by supporting so many sensors and last but not least by your prompt support in addressing issues and adding users' feature request.
Thank you for keeping the Mycodo alive :)

kizniche, 2019-04-18:
Just pushed a commit that adds Reset pin and editable location support. I'm going to try to get a release out this evening.

closed 2019-04-18 by kizniche