kizniche/Mycodo #649: PID controller

hpl002, 2019-04-19:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.4.3
 
#### Problem Description
Cannot seem to get PID controller to work. 'Last out' does not produce any output, thus 'output' is never triggered.(See screenshot) 

Trying to set up simple PID controller using Relay and DS18B20 sensor. Cannot get Relay / output to fire even though it should. (See screenshot)
  
### Errors
Did encounter this error, do not know if it is related. 
`Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1254, in page_function
    form_mod_pid_base.function_id.data)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_pid.py", line 243, in pid_activate
    if has_required_pid_values(pid_id):
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_pid.py", line 226, in has_required_pid_values
    device_unique_id = pid.measurement.split(',')[0]
AttributeError: 'NoneType' object has no attribute 'measurement'
[image]
[image]

`

### Steps to Reproduce the issue:
No particular steps
 
### Additional Notes
First deleted the database and reinstated it, still got the same result.
Then wiped mycodo from my system and installed it fresh, still same result..

kizniche, 2019-04-19:
Are there any errors in the daemon log? Are you acquiring measurements with the Input? Is your Input Period less than 120 seconds?

kizniche, 2019-04-19:
Is the output properly configured and does it work when you press ON and OFF on the Output page?

kizniche, 2019-04-19:
It looks like that error is from you attempting to activate the PID controller when a Measurement hadn't been saved yet. Does this sound accurate?

kizniche, 2019-04-19:
Have you added all the PID measurements to a new graph (on Dash page) to see what data is being recorded? Also add the output and input to the same graph to get the whole picture.

hpl002, 2019-04-19:
Thanks for getting back to me so quickly!

Yes, i'm getting clear readings from the sensor.
Output is properly configured, currently switching it on and off to regulate the environment myself.

I do get an error if i try saving a change before deactivating the controller, this has however never been an issue before.

I have a graph setup as i've seen you recommend in other related posts. I do not get any calculated output at all, oven visualised as bars in the graph.

After playing around, i see that ive gotten an new error from the db, which i thought would be reset as i reinstalled mycodo, but i may have missed something? Just to be clear, i still had the same issue, even before reinstalling.

From Daemon
`(Background on this error at: http://sqlalche.me/e/e3q8)
2019-04-19 22:06:27,332 - mycodo.daemon - INFO - Received command to terminate daemon
2019-04-19 22:06:28,295 - mycodo.daemon - INFO - All LCD controllers stopped
2019-04-19 22:06:28,296 - mycodo.daemon - INFO - All PID controllers stopped
2019-04-19 22:06:28,297 - mycodo.daemon - INFO - All Math controllers stopped
2019-04-19 22:06:28,297 - mycodo.daemon - INFO - All Input controllers stopped
2019-04-19 22:06:28,298 - mycodo.daemon - INFO - All Trigger controllers stopped
2019-04-19 22:06:28,298 - mycodo.daemon - INFO - All Conditional controllers stopped
2019-04-19 22:06:28,330 - mycodo.daemon - INFO - Mycodo terminated in 0.997 seconds

2019-04-19 22:06:38,863 - mycodo.daemon - INFO - Mycodo daemon v7.4.3 starting
2019-04-19 22:06:39,250 - mycodo.daemon - INFO - Anonymous statistics enabled
2019-04-19 22:06:39,275 - mycodo.daemon - INFO - Starting rpyc server
2019-04-19 22:06:39,883 - mycodo.output - INFO - Output controller activated in 300.6 ms
2019-04-19 22:06:40,384 - mycodo.daemon - INFO - All activated Conditional controllers started
2019-04-19 22:06:40,385 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-04-19 22:06:40,386 - mycodo.daemon - INFO - All activated Input controllers started
2019-04-19 22:06:40,386 - mycodo.daemon - INFO - All activated Math controllers started
2019-04-19 22:06:40,386 - mycodo.daemon - INFO - All activated PID controllers started
2019-04-19 22:06:40,386 - mycodo.daemon - INFO - All activated LCD controllers started
2019-04-19 22:06:40,887 - mycodo.daemon - INFO - Mycodo daemon started in 2.024 seconds
2019-04-19 22:06:40,892 - mycodo.daemon - INFO - 39.37 MB RAM in use
2019-04-19 22:09:15,557 - mycodo.daemon - INFO - Received command to terminate daemon
2019-04-19 22:09:15,620 - mycodo.daemon - INFO - All LCD controllers stopped
2019-04-19 22:09:15,620 - mycodo.daemon - INFO - All PID controllers stopped
2019-04-19 22:09:15,621 - mycodo.daemon - INFO - All Math controllers stopped
2019-04-19 22:09:15,621 - mycodo.daemon - INFO - All Input controllers stopped
2019-04-19 22:09:15,621 - mycodo.daemon - INFO - All Trigger controllers stopped
2019-04-19 22:09:15,621 - mycodo.daemon - INFO - All Conditional controllers stopped
2019-04-19 22:09:15,644 - mycodo.output - INFO - Output controller deactivated in 22.1 ms
2019-04-19 22:09:15,645 - mycodo.daemon - INFO - Mycodo terminated in 0.088 seconds

2019-04-19 22:09:21,402 - mycodo.daemon - INFO - Mycodo daemon v7.4.3 starting
2019-04-19 22:09:21,791 - mycodo.daemon - INFO - Anonymous statistics enabled
2019-04-19 22:09:21,818 - mycodo.daemon - INFO - Starting rpyc server
2019-04-19 22:09:22,365 - mycodo.output - INFO - Output controller activated in 267.9 ms
2019-04-19 22:09:22,867 - mycodo.daemon - INFO - All activated Conditional controllers started
2019-04-19 22:09:22,868 - mycodo.daemon - INFO - All activated Trigger controllers started
2019-04-19 22:09:22,868 - mycodo.daemon - INFO - All activated Input controllers started
2019-04-19 22:09:22,868 - mycodo.daemon - INFO - All activated Math controllers started
2019-04-19 22:09:22,869 - mycodo.daemon - INFO - All activated PID controllers started
2019-04-19 22:09:22,869 - mycodo.daemon - INFO - All activated LCD controllers started
2019-04-19 22:09:23,370 - mycodo.daemon - INFO - Mycodo daemon started in 1.967 seconds
2019-04-19 22:09:23,373 - mycodo.daemon - INFO - 39.41 MB RAM in use
2019-04-19 22:10:13,508 - mycodo.input_ef09105e - INFO - Activated in 240.6 ms
2019-04-19 22:11:23,532 - mycodo.output - INFO - Output 9298a6ab setup on pin 17 and turned OFF (OFF=LOW)
2019-04-19 22:12:19,451 - mycodo.output - INFO - Output 9298a6ab setup on pin 7 and turned OFF (OFF=LOW)
2019-04-19 22:14:31,466 - mycodo.output - INFO - Output 9298a6ab setup on pin 18 and turned OFF (OFF=LOW)
2019-04-19 22:17:11,069 - mycodo.pid_69142450 - INFO - PID Settings: Device ID: ef09105e-fa7e-4b73-aa7a-aa851c7e12f7, Measurement ID: e351911f-a538-450a-947c-1041ecb388c1, Direction: lower, Period: 30.0, Setpoint: 3.0, Band: 0.0, Kp: 1.0, Ki: 0.0, Kd: 0.0, Integrator Min: -100.0, Integrator Max 100.0, Output Raise: None, Output Raise Min On: 0.0, Output Raise Max On: 0.0, Output Raise Min Off: 0.0, Output Lower: 9298a6ab-7e23-46ff-9150-91d7af9f08f6, Output Lower Min On: 0.0, Output Lower Max On: 600.0, Output Lower Min Off: 0.0, Setpoint Tracking: 
2019-04-19 22:17:11,071 - mycodo.pid_69142450 - INFO - Activated in 210.3 ms
2019-04-19 22:17:11,071 - mycodo.pid_autotune - INFO - PID Autotune started
2019-04-19 22:17:41,236 - mycodo.pid_autotune - INFO - 
2019-04-19 22:17:41,236 - mycodo.pid_autotune - INFO - Cycle: 1
2019-04-19 22:17:41,237 - mycodo.pid_autotune - INFO - switched state: relay step down
[... cut ...]
2019-04-19 23:49:12,030 - mycodo.pid_69142450 - WARNING - No data returned from influxdb
2019-04-19 23:49:42,102 - mycodo.pid_69142450 - WARNING - No data returned from influxdb
2019-04-19 23:50:12,044 - mycodo.pid_69142450 - WARNING - No data returned from influxdb
2019-04-19 23:50:42,089 - mycodo.pid_69142450 - WARNING - No data returned from influxdb
2019-04-19 23:51:12,027 - mycodo.pid_69142450 - WARNING - No data returned from influxdb
2019-04-19 23:51:42,055 - mycodo.pid_69142450 - WARNING - No data returned from influxdb`

kizniche, 2019-04-19:
> mycodo.pid_69142450 - WARNING - No data returned from influxdb

This indicates the PID is not able to obtain measurement data. Without measurement data, it will not operate.

You didn't write what your Input Period is.

kizniche, 2019-04-19:
I would also not recommend attempting to use the auto tune feature until you can verify the PID is working.

kizniche, 2019-04-19:
In the log excerpt, there is no point where you activated the PID. It appears you are starting the auto tune and not actually atvitating the PID with the Activate button.

hpl002, 2019-04-24:
You are absolutely correct! I've now gone through again and initiated it via 'Activate'. 

Everything is working flawlessly.

I appreciate your help, and apologize for the clumsy mistake.

closed 2019-04-24 by hpl002