kizniche/Mycodo #650: DHT22 unable to activate with daemon

jlb-qi, 2019-04-21:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.4.3
 
#### Problem Description
Please list:

  - trying to add and activate DHT22 unit 
  - tested the unit seperately and works but seems to run into an error with the mycodo daemon
 
### Errors
Error: Could not activate Input controller with ID 7829a1e3-f8bf-4430-a1e4-04ff0ed4c403: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

DAEMON LOG

2019-04-21 13:57:00,148 - mycodo.daemon - WARNING - Input controller with ID 7829a1e3-f8bf-4430-a1e4-04ff0ed4c403 not found
2019-04-21 13:57:23,466 - mycodo.daemon - ERROR - Could not activate Input controller with ID 495f207f-ed95-4f2c-b9d6-3635ea7c4f0f: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 556, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 173, in __init__
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/dht22.py", line 117, in __init__
    self.gpio = int(input_dev.gpio_location)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

from line; 

https://github.com/kizniche/Mycodo/blob/b220ed6f04b9718653e6b8a77ca7f9a4a2e70811/mycodo/mycodo_daemon.py#L564-L569

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Add DHT to Mycodo using Setup > Data > Input dropdown > DHT22
 2. Activate DHT using the Activate button
 3. Error thrown up saying;
> "Error: Could not activate Input controller with ID 495f207f-ed95-4f2c-b9d6-3635ea7c4f0f: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

closed 2019-04-21 by jlb-qi

kizniche, 2019-04-21:
It looks like the pin wasn't set. It should have returned a warning about the missing pin. I'll check why the error-checking didn't work. I assume you closing TV e issue means you got it working?

jlb-qi, 2019-04-30:
yeah I got it working. my fault for not realising the save button was part of the workflow. school boy error

kizniche, 2019-04-30:
You found a bug in the warning system. I pushed a fix that will prevent activation if the pin isn't set. Thanks for the issue report.