kizniche/Mycodo #651: PID Stuck in calculation parameters calculated as 'NULL'

pigelb, 2019-04-24:
## Mycodo Issue Report:
  - Specific Mycodo Version: Mycodo Version: 7.4.2 
Python Version: 3.5.3 (default, Sep 27 2018, 17:25:39) [GCC 6.3.0 20170516] 
 
#### Problem Description
Please list: PID calculation parameters sometimes displays NULL (see pictures) and does not regulate output. Output seems to get stuck to a specific value. It is stuck in this state for hours.

Resetting (backend, RPi soft or hard) doesn't seem to fix it. 

PID HTR1 and PID HTR2 uses the same input and when not stuck in 'NULL mode'.

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
Deamon log has repetitive: 
2019-04-24 13:54:22,562 - mycodo.math_c3f71405 - ERROR - Unknown math type: average_single 

PID log has no log of value: 

### Steps to Reproduce the issue:
Cannot reproduce, occurs after some time. 
 
### Additional Notes

[image]

kizniche, 2019-04-24:
Can you please screenshot the expanded options of your PIDs as well as screenshot the expanded options of the Measurements used in each PID.

pigelb, 2019-04-25:
Kyle,

Here's a screenshot and a little more.

Let me how I can help.

Btw, it's a really great project.

I'd like to do some automatic reporting but I'm not very familiar with
influxdb and the datastructure generated by mycodo. Any pointers would be
greatly appreciated.

Best regards,

Pierre-Luc Gélineau
cell: 518-570-4000
''You will have to brace yourselves for this — not because it is difficult
to understand, but because it is absolutely ridiculous: All we do is draw
little arrows on a piece of paper'' Richard Feynman

kizniche, 2019-04-25:
Unfortunately, images don't get posted to the github issue from an email attachment when replying to the issue notification. To attach images, you'll have to make a comment with a web browser. I'll make a longer reply soon with information about how to query Mycodo data from influxdb.

pigelb, 2019-04-25:
[HVTA2TC - Function - Mycodo 7.4.2.pdf](https://github.com/kizniche/Mycodo/files/3117729/HVTA2TC.-.Function.-.Mycodo.7.4.2.pdf)
[Mycodo_7.4.2_Influxdb_v1.7.2_HVTA2TC_2019-04-24_15-52-56.zip](https://github.com/kizniche/Mycodo/files/3117730/Mycodo_7.4.2_Influxdb_v1.7.2_HVTA2TC_2019-04-24_15-52-56.zip)

kizniche, 2019-04-25:
I don't see any issues with the PID configuration. I did notice the autotune variables are changed from the default. Autotune is experimental and I wouldn't recommend using it in the current state, as it has caused PID issues in the past.

Can you create a graph with the PID output and determine when the PID stopped operating. Then, find if there are any PID errors in the daemon log at that time?

pigelb, 2019-04-25:
Here's the log from one instance where PID HTR2 was stuck with a command of 20sec output.
[Debug Chart.csv.xlsx](https://github.com/kizniche/Mycodo/files/3117817/Debug.Chart.csv.xlsx)
[debug Chart.log](https://github.com/kizniche/Mycodo/files/3117818/debug.Chart.log)

kizniche, 2019-04-25:
> PID HTR2 was stuck with a command of 20sec output

Can you elaborate what you mean by this?

Also, the long you attached shows PID with ID starting with "f1a04f0d" conducting an autotune (not activated and running).

kizniche, 2019-04-25:
Also, I would recommend using the Start Offset to prevent the "do data" type errors when the daemon starts. This option holds off starting controllers, giving influxdb time to start and data to start coming in.

pigelb, 2019-04-25:
The output calculated stays the same in this case 20 secs.

It is probably the auto tune setup that messed it up. I did play with it at
some point. Unfortunately, there is no auto tune stop.

kizniche, 2019-04-25:
To stop the autotune, you must click Deactivate. Then you can click Activate to start it normally. Restarting the Mycodo daemon does not stop the autotune.

pigelb, 2019-04-25:
Do you have any sense of how long it takes for the db to be ready?

kizniche, 2019-04-25:
On a Pi 3, I set Inputs/Maths with an Offset of ~10 seconds, and PIDs with an offset of 30 seconds. This allows Inputs/Maths time to wait fro influxdb to start, then PIDs a bit longer to allow Inputs/Maths to get at least one measurement in to be queried.

pigelb, 2019-04-25:
In this case, I did try to deactivate/reactivate the PID controler. But it
would still be stuck to a specific command value with all the NULL values.

kizniche, 2019-04-25:
In that case, you will have to delete that PID and add another PID.

I fixed one bug preventing stopping autotune a few months ago and successfully tested that fix. I'll see if I can reproduce your autotune issue.

kizniche, 2019-04-25:
I was able to replicate the PID autotune issue and just pushed a fix that will be incorporated into the next release.

kizniche, 2019-05-06:
Just released 7.5.0. I'm closing this issue. If you find the issue still persists, please reopen it. Thanks.

closed 2019-05-06 by kizniche