kizniche/Mycodo #652: Internet not detect

pigelb, 2019-04-25:
## Mycodo Issue Report:
  - Specific Mycodo Version: 7.4.2
 
#### Problem Description
Please list: Internet is not detected even if internet is accessible.

[image]

Eth0 connection is straight to a laptop to interface with Mycodo. 
Wlan0 connection is the one with access to the internet. 

### Errors

[image]
[image]

### Additional Notes
Routing table has Wlan0 as the main gateway.
[image]

kizniche, 2019-04-25:
Here is the code to check for internet:

https://github.com/kizniche/Mycodo/blob/b94fe06184e96320a8ad8a165009e666d319324a/mycodo/utils/system_pi.py#L202-L217

When logged into the terminal of your Pi, what is the output of the following command?

```
sudo apt install -y nmap
nmap -p 52 8.8.8.8
```

kizniche, 2019-04-25:
Additionally, you can run the following command from the terminal to see if a traceback is yielded from the error:

```bash
cd ~/Mycodo
~/Mycodo/env/bin/python -c 'from mycodo.utils.system_pi import internet; print(internet())'
```

kizniche, 2019-04-25:
Though, you may have to edit system_pi.py and change "error" to "exception" on the following line, save, then rerun that command, above.

https://github.com/kizniche/Mycodo/blob/b94fe06184e96320a8ad8a165009e666d319324a/mycodo/utils/system_pi.py#L215

closed 2019-06-07 by kizniche