kizniche/Mycodo #653: Display Addition: Generic 20x4 LCD

Magnum-Pl, 2019-04-29:
Hi Kyle, 

Can you add support for generic 20x4 LCD displays?

I was able to set one up and get it to display data using the 16x4 option, but it’s limited to 16 characters wide. 

[image]

kizniche, 2019-04-29:
I was I formed of this issue in #627 but it appears the issue was deeper than I thought. I'll check out the code where it cuts the character limit and get a fix for the next release.

Magnum-Pl, 2019-04-29:
Thanks.  

I was hoping there was a line in the code I could change from 16 to 20 but there is obviously more to it than that. 

The line I found about the width doesn’t have any number values in it 

`self.LCD_WIDTH = self.lcd_x_characters  # Max characters per line`

Magnum-Pl, 2019-04-29:
This is totally unrelated to this but I didn’t want to open a new issue for it. I was trying to checkout your forum for non-issue discussion but was unable to because it appears your website certificate is revoked. 

[image]

kizniche, 2019-04-29:
> it appears your website certificate is revoked

This happened to me with firefox, but it was due to a stale certificate stored with the browser. I opened Chrome which had never been to the site, and it worked fine, valid from 4/5 to 7/4. You may have to look up how to remove any stored certificates for the site.

kizniche, 2019-04-29:
Okay, I see there are issues, https://www.ssllabs.com/ssltest/analyze.html?d=kylegabriel.com

I'll see if I can fix that.

kizniche, 2019-04-29:
The SSL issue should be fixed now. Thanks for the heads-up! I'll try to get to the LCD issue soon.

kizniche, 2019-05-05:
What version Mycodo are you using, and what version Mycodo were you using when you added this specific LCD controller? I suspect this issue has been fixed already. Note that you will have to delete any 16x4 LCDs and add the 20x4 LCDs for the fix to take effect. Can you try to add a new 20x4 LCD and see if it works as expected?

Magnum-Pl, 2019-05-07:
That’s what the problem was. I didn’t realize the version of Mycodo was so old. The lcd worked fine after an upgrade. I didn’t even have to delete it the lcd. I was also able to add a new lcd with no issues. Thanks for the help

kizniche, 2019-05-07:
You're welcome. Glad to help.

closed 2019-05-07 by kizniche